MarlinFirmware/Marlin #10862: [1.1.x-bugfix] bug with sdcard support

mtmello2003, 2018-05-26:
Hello,

Using the tag:
#define SDSUPPORT 
with REPRAP DISCOUNT (non graphical display) got compile error:

In file included from sketch\MarlinConfig.h:26:0,

                 from sketch\Sd2Card.cpp:29:

Sd2Card.h:100:39: error: pasting "/* SPI Master In Slave Out pin*/" and "_DDR" does not give a valid preprocessing token

   #define SPI_MISO_PIN MISO_PIN       // SPI Master In Slave Out pin

config files:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2041829/Marlin.zip)

ejtagle, 2018-05-26:
@mtmello2003 : This a bug of the C preprocessor that some Arduino and some PlatformIO environments are using. Just remove the comments (```//``` onwards) and it will compile and work

mtmello2003, 2018-05-26:
That´s weird! I did as you recomended and it worked!

Tks

closed 2018-05-27 by thinkyhead

jhoblitt, 2018-09-14:
I saw this error with 1.8.6 but it went away after upgrading to `1.8.7`.

paul29163, 2018-11-22:
> @mtmello2003 : This a bug of the C preprocessor that some Arduino and some PlatformIO environments are using. Just remove the comments (`//` onwards) and it will compile and work

Great, completed as stated and compiled with no issues, Thanks.

disappear123df, 2019-10-24:
> @mtmello2003 : This a bug of the C preprocessor that some Arduino and some PlatformIO environments are using. Just remove the comments (`//` onwards) and it will compile and work

where do i find the (`//` onwards)