MarlinFirmware/Marlin #10863: [2.0.x-bugfix] bug with Reprapdiscount display

mtmello2003, 2018-05-26:
Hello,

After compile latest bugfix, using non graphical reprapdiscount display, it is impossible to see anything:

[image]

config files:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2041853/Marlin.zip)

ejtagle, 2018-05-26:
@mtmello2003: Should be fixed by #10864 , once pushed. If you want to try it, please do and report if it fixes the problem for you  👍

MasterPIC, 2018-05-26:
My Anet A8 was totally bricked too.
To be able to revert to a previous working firmware release I had to upload the firmware before the screen became corrupted (just within a few seconds from a power cycle). In a few words a few seconds after power up I had "not in sync" errors while trying to flash the working firmware and printer communication not working...
I'm going to try #10864 ...

mtmello2003, 2018-05-26:
Yes, solved the issue!

Tks

Roxy-3D, 2018-05-26:
I merged it so we don't get people confused...   

Thank You @ejtagle

boelle, 2019-02-20:
@mtmello2003 if issue is solved then maybe hit the green close button below

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead