MarlinFirmware/Marlin #10865: extruder stepping locked to axis stepping

leefla, 2018-05-26:
While investigating the Prusa inconsistent extrusion issue
https://github.com/prusa3d/Prusa-Firmware/issues/602
which I believe you're already aware of (though that thread has gotten out of hand)
I noticed an issue in the core marlin firmware, which I suspect may be responsible (though the issue is exacerbated by the Prusa Mk3 direct-drive extruder)

From looking at the stepper outputs with a logic analyzer it's obvious that the extruder steps are linked to be a multiple of the axis steps.
In specific example, if the X axis stepper interval is 0.4ms and the required extruder stepper interval is 4.75ms then the stepper will be processed every 0.4ms and the extruder stepper will be pulsed in the same process, resulting on the extruder interval being 4.8ms, with occasional catchup steps at 4.4ms, to maintain the required average of 4.75ms. This results in a brief ~10% over-extrusion.
The problem is lessened by an extruder with a higher gear ratio (thus requiring more frequent steps), but this just lessens the issue, rather than eliminating it.

The problem appears to be in planner.cpp where it sets block->step_event_count.wide to be the max of the step count on any axis.
In order to be able to maintain a steady interval on all steppers, this really needs to be the lowest-common-multiple of the number of steps on each axis (inc extruder), however this may result in a very high event count (and obviously some events will not result in any stepper output pins being pulsed).

I'm working on a solution, but I'm still ramping up on the Marlin firmware.

leefla, 2018-05-26:
For live discussionm feel free to join the discord discussion on this issue here: -
https://discord.gg/rchZGUg
#inconsistent_extrusion channel

ejtagle, 2018-05-26:
@leefla : There was a discussion on the limitations of running the steppers and the extruders at the same time. There is a physical limitation on scheduling of pulses on AVR (on ARM the problem is less severe). Please, read https://github.com/MarlinFirmware/Marlin/pull/10688#issuecomment-392197513 ... Hope you are trying an older version than yesterdays.... 👍

ejtagle, 2018-05-26:
@leefla : BTW: Increasing the number of events is not the answer. I think your idea of scheduling events asynchronously is possible. Since yesterday, we have a interrupt scheduler that can easily do that (Linear advance is using that feature!)

leefla, 2018-05-26:
@ejtagle, thanks for the quick reply. To handle this you would effectively need to either have non-linear interrupt timing, or have enough resources to schedule at the LCM of the event count (effectively the gcd of the step intervals) and quickly bail if no step operations are required for the current interval.

Assuming you don't have dynamic scheduling, and don't have resources to schedule at the interval GCD then at least scheduling at a higher frequency will reduce the effect somewhat.

Ideally the planner/scheduler would keep track of when the next pulse is desired for any stepper, schedule the ISR accordingly, and repeat, so it would fill out an array of desired step pulse times for each stepper, and just schedule the ISR for the next occurring step (taking into account the actual elapsed time thus far)

And you're right, this isn't on latest Marlin code, it's on Prusa code which is very out of date, but it is known to still be as issue with (reasonably recent) vanilla-ish Marlin running on Prusa hardware :)

leefla, 2018-05-26:
seems like a planner and stepper re-write would largely be needed to handle independent arrays of steps with desired times for each
I'm not familiar enough with the marlin code yet to attempt this, but it seems like this is the only real fix (or migrating to hardware with more 'oomph')

thinkyhead, 2018-05-26:
The way the stepper ISR works (non-linear-advance) is that it runs at the interval required for the stepper that needs to step the most while using the Bresenham algorithm to pulse all slower steppers at the intervals they require.

So, for example, let's say (for a `G1 X E` move) that at peak speed the X stepper needs to generate 100 pulses per second while the E stepper needs to generate 42 pulses per second…. The stepper ISR will be called 100 times per second and will generate a pulse for X on every call. However, it will only generate a pulse for E on 42% of those calls, staggered Bresenham-style at semi-regular intervals.

When acceleration or deceleration is in play, the ISR call interval will vary, but the pulse distribution will remain the same, proportional to the call interval.

The Prusa firmware does an interesting trick, which is to run the Stepper ISR faster when there's room to do so. It will double or quadruple the rate of the ISR while cutting in half the number of pulses generated at each call. This makes the steps distribution more regular, and has the most positive effect at lower feedrates.

ejtagle, 2018-05-26:
@thinkyhead : Grbl does exactly the same: It is not complex to do: Double rate, divide by 2 bresenham steps. The only thing that must be taken into account is rounding (as dividing by 2 could lose the last step)
I didn´t do it because "that" PR was huge enough. But, as you imagine, there is no change to the stepper code, besides taking into account that "last bit"

thinkyhead, 2018-05-26:
I added a paragraph above, mentioning that we'd have to do our doubling earlier in the flow.

ejtagle, 2018-05-26:
Maybe it is even easier than expected... ;)
```
 block->step_event_count = MAX4(block->steps[A_AXIS], block->steps[B_AXIS], block->steps[C_AXIS], esteps);
```

Just multiply by 2, 3, 4... and divide steps[] by the same amount, do proper rounding and all should be adjusted ..

thinkyhead, 2018-05-26:
Doubling integers will only result in half of the ISRs doing no stepping.

ejtagle, 2018-05-26:
Not right. Let me show you why:

Assume d step count, And movement in X is 10, Y is 6

Basically, bresenham will estimate Y/X = 6/10 = 0.6
So, for each step, X advances 1, Y advances 0.6. But, Bresenham "rounds" the error 

The code as it is right now, assumes going by the fastest axis
```
   X     Y      dX  dY 
 0.0 | 0.0 = 0 | 0 | 0
 1.0 | 0.6 = 1 | 1 | 1
 2.0 | 1,2 = 1 | 1 | 0
 3.0 | 1.8 = 2 | 1 | 1
 4.0 | 2.4 = 2 | 1 | 0
 5.0 | 3.0 = 3 | 1 | 1
 6.0 | 3.6 = 4 | 1 | 1
 7.0 | 4.2 = 4 | 1 | 0 
 8.0 | 4.8 = 5 | 1 | 1
 9.0 | 5.4 = 5 | 1 | 0
10.0 | 6.0 = 6 | 1 | 1
```

Lets double the rate. I intentionally divided advance steps by 2 (but it is better
to recompute them as total_steps/advance in that axis
```
   X         Y      dX  dY 
 0.0 = 0 | 0.0 = 0 | 0 | 0
 0.5 = 1 | 0.3 = 0 | 1 | 0
 1.0 = 1 | 0.6 = 1 | 0 | 1
 1.5 = 2 | 0.9 = 1 | 1 | 0
 2.0 = 2 | 1,2 = 1 | 0 | 0
 2.5 = 3 | 1,5 = 2 | 1 | 1
 3.0 = 3 | 1.8 = 2 | 0 | 0
 3.5 = 4 | 2.1 = 2 | 1 | 0
 4.0 = 4 | 2.4 = 2 | 0 | 0
 4.5 = 5 | 2.7 = 3 | 1 | 1
 5.0 = 5 | 3.0 = 3 | 0 | 0
 5.5 = 6 | 3.3 = 3 | 1 | 0
 6.0 = 6 | 3.6 = 4 | 0 | 1
 6.5 = 7 | 3.9 = 4 | 1 | 0
 7.0 = 7 | 4.2 = 4 | 0 | 0 
 7.5 = 8 | 4.5 = 5 | 1 | 1 
 8.0 = 8 | 4.8 = 5 | 0 | 0
 8.5 = 9 | 5.1 = 5 | 1 | 0
 9.0 = 9 | 5.4 = 5 | 0 | 0
 9.5 =10 | 5.7 = 6 | 1 | 1
10.0 =10 | 6.0 = 6 | 0 | 0
```

You get the same result, but the stepper pulse distribution is way better

ejtagle, 2018-05-26:
The rounding is the thing to be taken in to account.

In stepper.cpp:

```
// Initialize Bresenham counters to 1/2 the ceiling
      counter_X = counter_Y = counter_Z = counter_E = -((int32_t)(current_block->step_event_count >> 1));
```

That is the line that rounds. We start on purpose the counters to 0.5 ;)

leefla, 2018-05-26:
ideally the rate would be the LCM of the rates (GCD of the interval), but that could end up with a big event count
so calculating the LCM of the individual step counts, and then clamping to some multiple of the axis step count if the LCM is big, sounds good

BTW, I didn't see where the prusa code was running the ISR faster. It doesn't seem to be having any effect if so, unless it's not got the time in most cases
separate ISR for extruder with a completely separate interval would definitely be much better

leefla, 2018-05-26:
@ejtagle I'd love to see any changes that you make to improve this, since I'd like to migrate the same fix (if applicable) to Prusa (though their firmware is based on old Marlin)

leefla, 2018-05-26:
just increasing step_event_count doesn't seem to resolve, but I'm not clear yet where the other end of that needs to compensate

ejtagle, 2018-05-26:
As far as I understand it, that is what ADAPTIVE_MULTI_AXIS_STEP_SMOOTHING in Grbl does. It shouldn´t be hard to "transplant" to Marlin...

thinkyhead, 2018-05-26:
Seems to be [disabled in Prusa](https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/planner.cpp#L1053-L1067) firmware:
```cpp
#if 0
  // Oversample diagonal movements by a power of 2 up to 8x
  // to achieve more accurate diagonal movements.
  uint8_t bresenham_oversample = 1;
  for (uint8_t i = 0; i < 3; ++ i) {
    if (block->nominal_rate >= 5000) // 5kHz
      break;
    block->nominal_rate << 1;
    bresenham_oversample << 1;
    block->step_event_count << 1;
  }
  if (bresenham_oversample > 1)
    // Lower the acceleration steps/sec^2 to account for the oversampling.
    block->acceleration_st = (block->acceleration_st + (bresenham_oversample >> 1)) / bresenham_oversample;
#endif
```

> separate ISR for extruder with a completely separate interval would definitely be much better

Not really. Bresenham (especially with a high rate) accomplishes the same thing.

leefla, 2018-05-27:
Thanks, we're testing this now
obviously it's limited by the print speed, so slower prints should exhibit more difference

leefla, 2018-05-27:
one other idea would be using hardware PWM, so that it's decoupled from the ISR (just change the PWM rate when needed). Assuming the atmega has independent hardware PWM (I don't actually recall the specs, and haven't checked)

ejtagle, 2018-05-27:
It is also hard to use hardware PWM. PWM is assumed to have constant period, and a variable duty cycle. Even if you can change the period, it is not double buffered, so ISR timing could become critical.
Maybe a "compare" mode of the timer could work.. but it is also complex (but not impossible)

thinkyhead, 2018-05-27:
Also, the pins connected to `En_STEP_PIN` are not PWM pins on most (if any) boards.

leefla, 2018-05-27:
yeah, true, enough.
Just a quick update for you, the oversampling definitely improved movement overall. Linear axis accel/decel is now much smoother, but it hasn't made much improvement to actual print quality, so I'm waiting for someone to be able to get the logic analyzer hooked up again, and make sure that it's actually making the extruder steps more even as expected.
Either way it seems like a good improvement, but perhaps not making as much of a print quality improvement as we'd hoped.

leefla, 2018-05-28:
FYI, if you take a look at the photos added by mcm001 in this thread 
https://github.com/prusa3d/Prusa-Firmware/issues/602#issuecomment-392394716
You can see what a difference having the extruder steps evenly spaced makes !
It's a major difference
(this was accomplished by scripting g-code that made the extruder step interval be an exact multiple of the axis step interval)

bubnikv, 2018-05-28:
> Seems to be disabled in Prusa firmware (and has "<<=" typos):

I spend most time on the Slic3r Prusa Edition now, so I am not fully in the firmware development and I do not posses a up to day knowledge of the grbl state. I wrote that oversampling prototype for the Prusa3D fork, and IMHO the GRBL guys (I believe it was Sonny @chamnit, we exchanged couple of e-mails on the topic) followed up on the idea. Embarrassingly, my code in the Prusa3D was defunct, but at that time the development at Prusa3D was quite rapid, I did not have time and hardware to verify the oversampling with a logical analyzer, and due to the placebo effect I thought it was actually working. AFAIK when we found out the typo that made the code inactive, and we enabled it, the stepper routine was taking too much clock cycles, so the firmware was stuttering on a complex G-code. Anyway, I believe that the oversampling of the Bresenham algorithm is a good idea worth following up, but not really on the 8 bit Arduino.

thinkyhead, 2018-05-28:
> I believe that the oversampling of the Bresenham algorithm is a good idea worth following up, but not really on the 8 bit Arduino.

@bubnikv — I agree that we wouldn't want to double up when doing high-speed moves, but it should be safe to double up when moving at slower feedrates, maybe putting the ceiling at 30mm/s.

bubnikv, 2018-05-28:
@bubnikv — I agree that we wouldn't want to double up when doing high-speed moves, but it should be safe to double up when moving at slower feedrates, maybe putting the ceiling at 30mm/s for AVR.

The firmware (based on the ancient version of the GRBL planner) groups the stepper pulses by 2 or four to tick the interrupt at less than 10kHz, that is 50mm/sec maximum at the Prusa3D machines with no stepper pulse grouping. At 30mm/s you should be safe to do 2x oversampling. BTW, I ported our older MK2 firmware (to be sure it is well tested) to the MK3 / Einsy board with a minimum amount of changes, and I have also increased the micro stepping to 32x on XYZ, 64x on E axis, and IMHO it had no visible effect on the print quality. The problem here is, the print quality measure is very subjective.

thinkyhead, 2018-05-28:
> The problem here is, the print quality measure is very subjective.

Quite true, and even the best machine components have some amount of flex. And the dramatic improvements that can come from simply taking the time to do careful slicer tuning and doing much slower printing with smaller layers can't be overstated.

bubnikv, 2018-05-28:
> I can confirm that when getting into the 5-10KHz range it does slow things
> down, since it can't keep up.
> And oversampling at only 2x or 4x does not make a good enough improvement
> to E-step linearity.
>
> The only changes that actually make a VERY significant difference are
> either hand-scripting g-code so that the E step interval is an exact
> multiple of the axis step interval (so bresenham isn't really involved at
> all) and the E-steps are perfectly linear (which makes a HUGE difference to
> quality), OR using a geared extruder in combination with over-sampling such
> that the effect of e-step non-linearity is greatly reduced (not as perfect).
>
> In theory having variable-rate ISRs working with a time-stamp for each
> planned step (rather than just an interval), and each time scheduling the
> ISR to fire whenever the next step is required (whether that be an E-step
> or an axis step) should also make significant improvements in print quality
> (tho' I know this is a big change to the way the planner and stepper/ISR
> control code behaves).
>
> On different/more-powerful (CPU/controller) hardware there are multiple
> options available, such as scheduling multiple real-time jobs per gcode
> line, for processing steps, or if available using fully programmable PWM to
> drive the steppers (more practical for objects with straight sides than
> curvy objects).
> The absolute ideal would be to just calculate the ideal steps timings for
> every step ahead of time, and hand it off to a dedicated CPU or FPGA and
> let it run with it. (the main cpu then just handles error conditions,
> screen, serial/usb etc)

>  And oversampling at only 2x or 4x does not make a good enough
improvement to E-step linearity.

How do you know?

> The absolute ideal would be to just calculate the ideal steps timings for
every step ahead of time

I am looking into it as an experiment, reading the ticks from the SD card,
though the files will be huge.

chamnit, 2018-05-28:
Oversampling primarily reduces dynamic disturbances excited by the stepper motors. In other words, it reduces noise. It also smooths out the off-axis steps in particular scenarios. 

In my study of this, just a 2x oversampling significantly improves the consistency of the output but is asymptotic as you increase the oversampling. 4x provide more improvement but not as much. 8x even less so but still a small improvement. 

Grbl does dynamic oversampling. For low step frequencies, it oversamples by 4x. For mid frequencies, it oversamples by 2x. For the higher frequencies, it just runs it straight. The frequency bands are tuned to keep the AVR processor consistently loaded throughout it’s supported performance range.

ejtagle, 2018-05-28:
I already have a Marlin patch that implements oversampling. But i also have found that AVR is not fast enough...8khz ISR seems to be the maximum that is can handle... The patch is not ready to be merged yet...

leefla, 2018-05-28:
> > And oversampling at only 2x or 4x does not make a good enough
> > improvement to E-step linearity.
> How do you know?

By actually testing it, and comparing the results both against prints without over-sampling, and against prints with perfect E/X stepper ratio (using manual g-code)

thinkyhead, 2018-05-28:
> 8khz ISR seems to be the maximum that is can handle.

No doubt. I believe we limit it to 5KHz currently.

leefla, 2018-05-29:
With what's in Prusa firmware at least (haven't checked latest marlin) it's limited to not increase once it's over 5KHz, so that means it can be anywhere between 5KHz and 9.999KHz at max.
I've reduced that locally to keep it in the 3.75KHz to 7.5KHz range (ensuring that the CPU isn't overloaded)
Of course that limits what print-speeds it kicks in at.

bubnikv, 2018-05-29:
>  No doubt. I believe we limit it to 5KHz currently.

The Prusa3D fork limits the stepper to 10kHz.

I am experimenting with a raw G-code line, something like:
byte(128)
byte(length)
byte(4 directions + 4 steps)
byte(timer to the next ISR tick)
byte(4 directions + 4 steps)
byte(timer to the next ISR tick)
....

where this raw G-code is then executed by the stepper routine as is.

This way I can run the planner on the PC and get the exact timing, though I
cannot easily do the mesh bed leveling. This simple trick allows me to
experiment with exact stepper timing, with S-curves etc. Let's see what
will come out of this.

metacollin, 2018-05-29:
Trinamic has a [nice little PDF](https://www.trinamic.com/fileadmin/assets/Solutions/White_papers/Reducing_timer_interrupt_load_in_3D_printer_applications.pdfl) that explains this issue quite well, in case anyone was still a little shaky on this.  It also uses equations to generalize the timing error, which is nice.  

Unfortunately, it doesn't offer much in the way of solutions that aren't already being used in Marlin or don't involve buying a fancy new Trinamic chip to solve the problem for you ;).  The part that isn't marking is quite good though.

thinkyhead, 2018-05-29:
> The Prusa3D fork limits the stepper to 10kHz.

Thanks, I was about to correct myself!

thinkyhead, 2018-05-29:
> I am experimenting with a raw G-code line, something like:

You might want to take a look at #7047.

bubnikv, 2018-05-29:
>  You might want to take a look at #7047
<https://github.com/MarlinFirmware/Marlin/pull/7047>.

Thanks for the hint. Funny thing, one is usually not the only one having
the same idea.

leefla, 2018-05-29:
Heh, I'd suggested embedding the time between extruder steps (and potentially all steps) in the planner output in the discord discussing this.
Funny how, as they say, "great minds think alike" :)

thinkyhead, 2018-05-29:
Since time in a given block scales with acceleration and deceleration I'm not sure what more we'd do with that value. We already have the proportion of E to the most-stepping axis.

leefla, 2018-05-29:
having an actual timestep rather than just interval, and rather than just say 11.45 E steps per X axis step, means that you can schedule the ISR specifically to whichever step is due next, AND if marginally late can still maintain the average by delaying from current time to the next step, rather than by the pre-calculated interval.
I've been using this kind of setup for decades in realtime embedded control systems ;)

thinkyhead, 2018-05-29:
Surely we have not all been getting lousy extrusion for all this time. I get beautiful surface quality in my prints. I honestly don’t see the issue with having E micro-steps regularly vary by one ISR interval, which is the most that they `can` vary under Bresenham.

metacollin, 2018-05-30:
Ok, this might be a dumb or impractical idea, but is there anything stopping us from using a timer in 'one-shot' mode (more on that in a second) to selectively 'nudge' the step pulse such that it occurs slightly delayed from the ISR interval, but only that one time?  It would almost be like nudging the phase of the pulse a bit. 

Lets say you put a timer in fast PWM mode.  You have two registers you can play with, TOP and MATCH.  TOP is, of course, how high the timer will count before looping back around, and MATCH is the value that causes an interrupt or changes the value of a pin if connected to the PWM peripheral, etc.  

So, take an 8-bit timer, and set it's top value to 1, or 2, I dunno.  Something smaller than MATCH. It'll sit there buzzing away, madly looping from 0 to 1 or whatever value you picked, and will never actually trigger anything because it will never actually count up to MATCH.   

But, lets say you have a certain amount of time you'd like to send a STEP pulse to the extruder, and instead of being forced shoehorn the timing into the ISR interval is best you can, you set MATCH to a value that will roughly give you a correctly timed E step (that can now occur independently of the ISR interval).  Then you manually write to the timer's COUNT register to the value just above top.  It will count up to match, then trigger on that to send a STEP pin pulse.  It will eventually overflow, start counting up to TOP again, and will not trigger again.  A one shot =P.   

This would also most likely make that timer/counter unit useless for anything else, and I'm not sure what the timer situation is, but it might require shuffling something and interleaving it onto another timer or worse case, simply making a trade-off between some optional feature and better extrusion consistency.  

Is there any merit to this idea?  I can't actually tell myself.  I'm an engineer, not an embedded programmer, so I apologize if this is silly and I just wasted everyone's time.

metacollin, 2018-05-30:
> I get beautiful surface quality in my prints. 

So, I think one of the biggest issues surrounding this problem, as well as the difficulty in determining the scope and degree of the issue is that, well, people keep saying stuff like this.  

I'm glad you think your prints are beautiful.  I am in the the same boat - I never noticed anything was wrong until I happened across the github issue on the Prusa firmware by accident.  And even though I've been able to confirm my printer has the issue, I still don't really care.  The print quality is perfectly acceptable to me, and I'm mostly here because it was an interesting problem to track down.  I am not particularly concerned with 'solving' this (if a solution even exists).  

But my subjective opinion on the quality of my prints is utterly irrelevant to any of this.  It doesn't matter what my opinion is of the quality of my prints is because that is not in anyway a useful metric in determining if the issue exists or not.  Indeed - I KNOW I have the issue with my MK3, yet I also still think my print quality is awesome.  

And you also think your print surface quality now and in the past has been great.  Awesome.  What does that say about the presence or severity of this timing issue?  Absolutely nothing.  All it says is that you think the surface quality of your prints is beautiful - something that is not necessarily mutually exclusive with the presence of this issue.   

Multiply this by hundreds of people chiming in to assert various things about if a printer does or doesn't have it or if there is or isn't an issue at all, all based on nothing except their utterly subjective personal opinion, and I think it is easy to see how problematic it has been to even narrow down what is really going on.  Honestly, I STILL don't think we have a very satisfying answer as to why the problem is so much more noticeable on the MK3 and MK2.5.  The current hypothesis is the bond tech extruder gears, the only other commonality between the MK3 and MK2.5 besides the magnetic heat bed.  However, the MK2 is a direct-drive extruder as well, albeit one with a more traditional hobbed gear + spring-loaded lever with a  free-spinning bearing to exert pressure on the filament.  

Just humor me for a moment, and consider this timeline of events:

1.  This issue really has always been present, but as you've been saying, the error is too small to have any easily noticeable impact on surface quality.  All is well, everyone is happily printing their benchys etc.  

2.  For some reason (that, at least for me, still doesn't have a fully satisfactory explanation), a new printer is impacted by the timing issue much more severely.  The actual timing error is no better or worse of course, but for whatever reason, the impact on relative extrusion width is increased, or perhaps more likely, is no longer being removed by some sort of mechanical low-pass filter effect, something like that. This makes the issue noticeable for the first time, and for the first time, it IS noticed. 

3.  People start, well, bitching about it, which makes yet more people start to look especially hard and close at their own print quality, causing even more people to notice the issue.  Then people with printers that have always printed, in their eyes, utterly immaculate prints with perfect surface finish, out of morbid curiosity, begin to wonder if the problem is present on their printer as well, just much less noticeable.  They begin using a flash light or whatever to illuminate their prints edge-on, making any line width variation that would not be visible/noticeable under ordinary lighting conditions become visible quite readily.  And too their dismay, the same tell-tale is present on their prints, just much more difficult to notice.  Now that the issue was made noticeable on one printer, people begin putting their prints under hyperscrutiny, a level and in a circumstance that no one would have done otherwise.  

And there are several reactions:  There is honest acceptance. These people will say that the issue exists and on their printer, though not nearly as bad.  And there will be denial, because no one wants to believe there is this flaw (however insignificant for some or most printers it may be) with their fairly expensive machine.  And there will be the people who will not really check, and just use their own subjective opinion about the print surface quality under normal lighting/viewing conditions to determine if their printer has the problem or not, and will almost certainly say it doesn't.  

I'm not saying you don't have a good point here - if the print looks great upon normal lighting/viewing conditions and reasonable level of scrutiny, then isn't that what matters?  Doesn't that mean the problem isn't a problem, and the effect it has is insignificant?  

Yes and no.   It is insignificant on that printer setup, and to you personally.  But that means little to whether it is there or not, and it isn't like a problem can be masked/hidden for years by some other mechanism(s), until circumstances occur that remove these mitigating factors and the real severity of the problem suddenly comes out of the shadows.   There are countless examples of exactly such a series of events happening.  

I'm just saying, it would be wise to keep any sort of subjective evaluations out of this.  Comparisons are the only thing here that is worth anything.  We can compare the surface quality of two different printers.  We can  [... cut ...] ning or what have you.  

And to tell if your printer(s) that seem perfectly fine have it or not will require, at a minimum, photos taken in such a way to make even tiny extrusion width variation much more visible, and then performing a print with the timing error present, and one of the same object, but with artificially 'chopped' g-codes that gets rid of the remainder/error counter, then comparing the surface quality between the two.  That is really the extent you'd need to go to say for sure.

hoffbaked, 2018-05-30:
@metacollin I've been trying to, mostly for fun, wrap my head around how all this works, and I came around to see what I think is @thinkyhead 's point. It seems like if a single extruder step is that noticeable by itself, either because of low extruder resolution, some micro stepping wonkiness, or a mechanical issue, the very slight error in the timing of when that step happens isn't going to mitigate it.

thinkyhead, 2018-05-30:
@jonnieZG — Welcome to the Marlin project. If you have any questions about our policies and procedures feel free to ask. We’re here to help.

thinkyhead, 2018-05-30:
As I have stated, and as I can certainly demonstrate, there is no unusual aliasing produced by the use of the Bresenham algorithm. A single extruder step is not going to be noticeable in extrusion, which is delayed by the “springiness” of pressure in the hot-end. There is no accumulation of error, and the timing of pulses to the E stepper is highly regular, at most varying by one single ISR interval at any given step. You can see how it behaves simply by examining a large-pixel line drawn by Bresenham on an old Atari 400 computer. Take the X axis as the X axis of a 3D printer, and the Y axis as the Extruder. The diagonal is imperfect, but the variation of E to X is minimal, limited to a variance of a single step more or less compared to a perfect multiple of the X axis. This variance is so small as to be negligible, especially given the the pressurized and delayed nature of extrusion.

leefla, 2018-06-01:
The problem, based on my understanding is based on a mixture of things
1. The exact ratio of E to axis steps (or more specifically to bresenham resolution) can make this more or less severe (ie: whether it's one E step in 10 that's irregular or one E-step in 50).
That in itself is a result of stepper ratios for all axis.
2. The amount of extrusion per extruder-step. This is something that not only affects how noticeable a single step variation (of one unit of bresenham resolution) is, but also how frequently it occurs.
If one step is 1/4 of the amount of filament on one printer than it is on another, as a result of stepper ratios and gearing, then it may make less difference.
Micro-stepping also plays a factor here
3. Fluid dynamics of a viscous material (molten filament) any momentary variation in speed of a viscous fluid produces bow-wake like affects that can cause longer lasting ripples as the material is compressed and expanded, this is compounded further by the viscosity changing as the material cools. (A friend of mine has a PhD in fluid dynamics and has published many articles on this)

For me, thus far, I'd consider the effect relatively minor, and not a huge detractor, BUT that doesn't mean that it doesn't exist.

I've seen categorical proof that by making the E step timing 100% linear (by generating Gcode that ensures the E-step interval is a direct multiple of the axis step internal) it eliminates the issue and makes a huge improvement to surface quality (comparing two prints, side by side, under the same lighting, from the same printer).

The same result (near enough) is achieved by using an early version on Klipper, which effectively has a much finer resolution for step intervals. While there are other problems with those prints (due to the lack of mesh bed levelling, and other features not yet implemented) this specific issue is absent from prints (where it's visible from prints on exactly the same printer, using stock firmware).

Finally using a higher geared extruder (3.5:1) reduces, but does not fully eliminate, the issue since it reduces the amount of variation in extrusion from a single step variation (because the variation, while it may end up being more frequent, is less filament per-step, and thus per variation). This latter isn't quite as concrete proof, because the printer has been physically modified to achieve the result.

Just to be clear, none of this is ever intended to be any kind of negativity at the Marlin developers, the firmware implementation, or of the Prusa developers/firmware. I'm just making clear that an issue (however minor it may be, at least for some) does exist for many users, and that extruder stepper timing seems to be a significant contributor (combined with the extruder gearing ratio and direct-drive on the Prusa Mk3 - likely because any timing variation thus has a bigger, and more immediate effect).

It is also worth noting that the viscosity of any specific filament may also have some small effect, since I have noticed variations between different filaments. At first I thought this was primarily due to the visual surface finish of the filament, but having measured the thickness of a single wall cube, printed with exactly the same gcode running on exactly the same printer, one is exactly 0.45mm, where the other is 0.47-0.48mm on average, and i suspect this is due to the surface imperfections which are more noticable using that filament

ejtagle, 2018-06-01:
@leefla , @thinkyhead : I share the ideas with you, but even if bresenham does NOT lose steps at all, and also does NOT accumulate error, there is a concept i would call "time resolution" - If the quotient between major axis and minor axis (major axis means, in this context, the axis that must create more step pulses compared with the other ones, including the extruder)
Well, if the quotient result is not an integer, then bresenham, at some points in the movement of the major axis, must decide that it has to move the minor axis. It is done in such way that after the full major axis movement has executed, it also has executed the full movements of the minor axis. And the minor axis steps were properly distributed evenly along the major axis movement. So good so far.
But, as said, Bresenham has "discrete" decision points: It can only decide to move (or not to move) minor axis exactly at the moment the major axis moves. And that is not the ideal point (in time) usually.
With slow movements that are composed of a similar, but not equal number of steps in all axis, the problem worsens, as the decision points are distributed very scarcely, and there are large delays between those decision points.
It is nearly trivial to extend bresenham to "oversample" in that situation. I finally was able to get enough time and already did the math. The good news is that we can retain the exactness and simplicity of bresenham, and just slightly changing the initialization of it, do oversampling and increase the decision points of slow moves, and thus allow more precise calculated pulses.
I will port the math to marlin, and open a PR with this idea, for all us to test 👍

thinkyhead, 2018-06-02:
> none of this is ever intended to be any kind of negativity at the Marlin developers, the firmware implementation, or of the Prusa developers/firmware

No worries. I am personally just interested in the problem-solving aspect, and I think a lively debate can be healthy and fun. We wouldn't get far in our explorations without some occasional disagreements!

> an issue (however minor it may be, at least for some) does exist for many users, and that extruder stepper timing seems to be a significant contributor

One simple experiment we can try is to use the smallest multiple of the E steps that exceeds the major axis's steps as the "ceiling" for the Bresenham algorithm and then tie all stepping to that value. The ISR would run slightly more often (potentially up to twice as often, but we could fall back in such cases), but it would produce totally regular E stepping.

ejtagle, 2018-06-02:
There is a better way. I am currently testing it. I am sure you'll be delighted by how can the Bresenham algorithm be extended without computing costs to handle oversampling... ;) -- I'll supply also the demostration and a simple program so you can analyze it 👍

ejtagle, 2018-06-03:
No, blocks are properly merged- This is the dump of the created blocks:

get_current_block: Tail:0
Head:2

flag:2
extrude:0
step_event_count:8000
accelerate_until:167
decelerate_after:8000
cruise_rate:8000
acceleration_time:1723750
deceleration_time:0
acceleration_time_inverse:2491
deceleration_time_inverse:0
direction_bits:0
nominal_speed_sqr:10000.00
entry_speed_sqr:0.00
max_entry_speed_sqr:0.00
millimeters:100.00
acceleration:2400.00
nominal_rate:8000
initial_rate:120
final_rate:8000
acceleration_steps_per_s2:192000
segment_time_us:192000

get_current_block: Tail:1
Head:2

flag:2
extrude:0
step_event_count:8000
accelerate_until:0
decelerate_after:7834
cruise_rate:8000
acceleration_time:0
deceleration_time:1723750
acceleration_time_inverse:0
deceleration_time_inverse:2491
direction_bits:0
nominal_speed_sqr:10000.00
entry_speed_sqr:10000.00
max_entry_speed_sqr:10000.00
millimeters:100.00
acceleration:2400.00
nominal_rate:8000
initial_rate:8000
final_rate:120
acceleration_steps_per_s2:192000
segment_time_us:192000

dirkvdb2, 2018-06-06:
Very impressive work that you guys are doing! Just a thought: the improvements that you are proposing involve a higher rate of interrupts anyways. I was wondering wether completely desynchronising the interrupt from any specific timing for any of the steppers, simplifying the ISR (to make it faster) and serving it more frequently would not be a option. That would create jitter on all the stepper timings (now it is on all but one) , but probably more random and lower in amplitude, and it could make the ISR code simpler and faster:

ejtagle, 2018-06-13:
Basically, Marlin has now an Adaptive Multi axis stepping feature that should be solving the problem presented here...

Skiidlive, 2018-06-19:
Anyone know the status of this?

ejtagle, 2018-06-19:
AMASS algoritm is already implemented in bugfix?.x.x branches

closed 2018-06-19 by thinkyhead

triplepoint, 2018-06-19:
Can we get a link to the work that addressed this issue?

ejtagle, 2018-06-20:
We did implement a bresenham oversampling. It is a huge commit. We did not implement separate ISR handlers for each motor, as that would mean more positional error on pulses (AVR is just not fast enough to handle multiple motors using separate ISRs: the ISR processing time would interfere the timing of the other motor pulses.
AMASS is disabled by default: You must enable it on marlin;
``` ADAPTIVE_STEP_SMOOTHING ``` is what you are looking for