MarlinFirmware/Marlin #10866: [1.1.x-bugfix] strange end stop behaviour on y-axis

agoulah, 2018-05-26:
Hi,

I recently updated to a 1.1.x-bugfix build from a pull on May-24 and have been noticing some strange behaviour with my y min endstop.  Some of the firmware settings that I have enabled that may be a culprit are QUICK_HOME and Z_SAFE_HOMING.  I recently just installed a BLtouch and this was when I upgraded the firmware and noticed the issue.  What is happening is during homing the y axis, it would randomly stop as if it hit the end stop, and re-home bump and it would either pause again as if it hit the endstop or continue homing until it actually hits the end stop.  

The issue seems to occur randomly, so thinking it was noise I installed a 100nF ceramic capacitor in parallel but that did not fix the issue.  Tried another endstop and still there.  Hooked up an oscilloscope to the switch and set triggers and there would be nothing when it triggers in marlin, so I don't think it's noise or a faulty endstop.

With M43 E1 enabled, I issue a G28 X Y followed by G1 F3600 X150 Y150, and repeat this several times to monitor and it would randomly output y-min triggered, so makes me think it's a firmware bug with some recent changes.  I reverted back to a build that was pulled from 1.1.x-bugfix on May-11, enabling BLtouch for that build and went through the same process as above testing the homing, and everything worked as expected.

Something I noticed between the two builds is the M43 E1 endstop debugging output is very different, and looking at the commits between May-11 and May-26 (just updated to latest build and the issue is still there) there was a big change with the endstop core, so maybe there's an issue in there.   I'm taking a look through to see if I can find anything obvious, but it's probably beyond what I can fix.

If there's any other details that may be required that I missed, please let me know.

thinkyhead, 2018-05-26:
We've just done some changes related to endstops in the last day or so, so please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands.

I'm not sure how well the `M43` endstop testing aligns with the recent changes. I don't think it was paid much attention during the recent update.

agoulah, 2018-05-27:
Just tested the latest branch of bugfix-1.1.x and bugfix-2.0.x and issue still present on both.

agoulah, 2018-05-27:
The issue isn't present when using the May-24 compile of bugfix-1.1.x on another printer, so I think it's a signal integrity issue with my cabling since with ENDSTOP_NOISE_FILTER on it fixes the problem.  I ended up swapping out the y-axis end stop cable with a known working one and disabled ENDSTOP_NOISE_FILTER and it's back to perfect homing.  Sorry!

closed 2018-05-27 by agoulah

thinkyhead, 2018-05-27:
Thanks for the update!

We'll keep an eye on whether that `ENDSTOP_NOISE_FILTER` side-effect (_"not stopping back on the endstop on the second slower bump"_) amounts to a bug.