MarlinFirmware/Marlin #10869: No SD-Card support for Red RepRapDiscount Full Graphic Smart Controller [2.0.x-bugfix]

madmax2, 2018-05-27:
### Description

I an issue with my Reprap Discount Full Graphics Smart Controller 
When I insert SD card, no SD card is detected.

I have tried the code - safe_delay(1000) but still not detecting any external SD card.
as suggested in this issue
https://github.com/MarlinFirmware/Marlin/issues/2628#issuecomment-392239693

I am using marlin Marlin-bugfix-2.0.x version
Board = MKS Sbase 1.3

SD card support has already been defined/enabled in configuration.h.

### Steps to Reproduce

1. Turn on the 3D printer
2. Insert SD card
3. Click knob on LCD to see if any SD Card is detected.

**Expected behavior:** [What you expect to happen]
SD card should be detected.

**Actual behavior:** [What actually happens]
No SD card detected.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

Roxy-3D, 2018-05-27:
On email you show up as Tyler Durden.   I thought we were going to get some conspiracy theory stuff and have to scramble to patch something.

madmax2, 2018-05-27:
@Roxy-3D Haha, thanks for your input.
But not really useful to solve my problem.

Nope, no conspiracy here.
Just reporting a problem/bug.

Roxy-3D, 2018-05-27:
As you know...  SD Card support has been broken out from the particular LCD Panel and needs to be enabled separately.  But you obviously know that and have it turned on.   We have had a couple of recent patches to the LCD firmware so it is possible something there is factoring in.

Let's just sit tight for a day...  Somebody with more expertise on the subject will probably jump in.

Now...  Back to the conspiracy stuff...   There aren't any judicial orders to shut down Open Source software, are there????

madmax2, 2018-05-27:
@Roxy-3D No I didn't know anything about "SD Card support has been broken out from the particular LCD Panel and needs to be enabled separately."

What do you mean by this comment?
All I did was defined/ enable SD card support.

I am not up to date with conspiracy theories nowadays..
The last big one was about 911.

As you should know, I am more interested in causing mayhem and that other stuff I am not able to talk about. 

**"Now... Back to the conspiracy stuff... There aren't any judicial orders to shut down Open Source software, are there????"**

You better ask Trump about that.

madmax2, 2018-05-27:
@thinkyhead Should title of the issue not include "RepRapDiscount Full Graphic Smart Controller"
in case someone with same problem, does a google or github search of the problem, can find this issue on github more easier.

thinkyhead, 2018-05-27:
Do you know which pin on your board the `SD_DETECT_PIN` from the LCD controller is connected to? The default value in `pins_MKS_SBASE.h` is `P2_11` but apparently this is not necessarily where the LCD cable routes it.
```cpp
/**
 * P0.27 is on EXP2 and the on-board SD card's socket. That means it can't be
 * used as the SD_DETECT for the LCD's SD card.
 *
 * The best solution is to use the custom cable to connect the LCD's SD_DETECT
 * to a pin NOT on EXP2.
 *
 * If you can't find a pin to use for the LCD's SD_DETECT then comment out
 * SD_DETECT_PIN entirely and remove that wire from the the custom cable.
 */
#define SD_DETECT_PIN      P2_11   // J8-5 (moved from EXP2 P0.27)
```
If you can't find the pin, or if there isn't one, just comment out the `SD_DETECT_PIN` and use manual SD init.

madmax2, 2018-05-27:
@thinkyhead Should the title of the issue not include "RepRapDiscount Full Graphic Smart Controller"
in case someone with same problem, does a google or github search of the problem, can find this issue on github more easier.

Roxy-3D, 2018-05-27:
> What do you mean by this comment?
> All I did was defined/ enable SD card support.

It used to be that if you enabled a particular LCD Panel...  That would turn on LCD support.   But now...  The LCD Panel Support and LCD are separate.   You can enable them separately.    But from what you are saying, you have it turned on.   Your LCD Panel is working, right?

madmax2, 2018-05-27:
@Roxy-3D 
yes LCD works fine, except for there is no LCD feedback beep when scrolling through the menu
even though I enable LCD frequency.

Just no SD card detected.

madmax2, 2018-05-27:
@thinkyhead I also notice the SD card works on smoothieware firmware.

So it must be something to do with how marlin has assign the pins.

Roxy-3D, 2018-05-27:
What pins does Smoothieware say the external SD card is on?

madmax2, 2018-05-27:
@Roxy-3D Good idea.

Here's the section with regards to external SD card

panel.lcd                                   reprap_discount_glcd     #
panel.spi_channel                           0                 # SPI channel to use  ; GLCD EXP1 Pins 3,5 (MOSI, SCLK)
panel.spi_cs_pin                            0.16              # SPI chip select     ; GLCD EXP1 Pin 4
panel.encoder_a_pin                         3.25!^            # Encoder pin         ; GLCD EXP2 Pin 3
panel.encoder_b_pin                         3.26!^            # Encoder pin         ; GLCD EXP2 Pin 5
panel.click_button_pin                      1.30!^            # Click button        ; GLCD EXP1 Pin 2
panel.buzz_pin                              1.31              # Pin for buzzer      ; GLCD EXP1 Pin 1
panel.back_button_pin                       2.11!^            # Back button         ; GLCD EXP2 Pin 8

**panel.external_sd                     true              # set to true if there is an extrernal sdcard on the panel
panel.external_sd.spi_channel         1                 # set spi channel the sdcard is on
panel.external_sd.spi_cs_pin          0.28              # set spi chip select for the sdcard (or any spare pin)
panel.external_sd.sdcd_pin            0.27!^            # sd detect signal (set to nc if no sdcard detect) (or any spare pin)**

thinkyhead, 2018-05-27:
If that's what Smoothie uses then try setting `SD_DETECT_PIN` in `pins_MKS_SBASE.h` to `P0_27`. The comment above the pin definition states that it is shared with the card on the board, but maybe that doesn't matter.

madmax2, 2018-05-27:
@thinkyhead Does it matter that in the config it indicates the pin 0.27 is inverted pin? 
ie "^!"

http://smoothieware.org/pin-configuration 
!   | Exclamation mark |   | Invert pin
^ | Caret, Shift+6 on QWERTY keyboards |   | Set pin to pull up (Default on most pins)

Even if it is shared, I don't think you would be accessing both at once?
Or it allows sharing.

I just reflashed back to smoothieware just to test the internal and external SD card.
Yes, I am able to access both the internal microsd card on the board and the external SD card on the LCD.

thinkyhead, 2018-05-27:
The pullup will be set automatically for this pin. If you don't get the right behavior after the first flash, try making this change in `Conditionals_post.h`…
```diff
/**
 * Override here because this is set in Configuration_adv.h
 */
#if ENABLED(ULTIPANEL) && DISABLED(ELB_FULL_GRAPHIC_CONTROLLER)
- #undef SD_DETECT_INVERTED
+ //#undef SD_DETECT_INVERTED
#endif
```

madmax2, 2018-05-28:
@thinkyhead  Okay I just tested both suggestion and it still does not read the SD card.

I don't understand why Marlin can't read the SD card (unless you create some hack cable) as referenced in https://github.com/MarlinFirmware/Marlin/issues/8498

But under smoothieware it is has no problem reading the external or internal SD card from within the LCD screen (without resorting to using a hacked cable).

Do you think I also need to change some other pins like what I posted above for the smoothieware config?

Please investigate this problem further.

thinkyhead, 2018-05-28:
Does Smoothieware detect the card being removed and re-inserted?

thinkyhead, 2018-05-28:
You may want to get in on the discussion at #10400 too.

madmax2, 2018-05-28:
Okay I just did some tests on smoothieware.

When LCD screen is on and no SD card inserted.
I insert the card, I can read the card when I go to "Play> ext" menu item

While the LCD is on...
If I pull out the card and insert it, 
Then go to ext menu item, it will show as empty.
Then I go out and back into in the EXT menu item, it will show me two files on my SD card that is now truncated like DOS filename.

But I cannot print those gcode files and it seems to be invalid files.

So basically it does not update or show the correct contents of the SD card when I remove and reinsert the card while the LCD is on, not matter how many times I do it.

I need to press the reset switch on my LCD, before it will show the correct contents of the SD card again.

madmax2, 2018-05-28:
Yeah I will monitor that thread..
But from reading it, there does not appear to be a solution to this problem right
other than creating a hacked cable...

Do you have any hunch / clue / idea as to why you think smoothieware works right out of the box with external and internal SD card but marlin doesn't?

Is this something that can possibly be fixed as a software solution in marlin or 
we will need to permanently resort to that hacked cable solution if want SD support.

Maybe we can contact the Smoothieware team and ask them how they got SD card support working?

===
Just to understand the underlying problem with regards to marlin..
and why they came up with a hacked cable solution..

Is the problem currently that marlin can't read that SD card detect pin on the EXP2 jumper on the SBASE board?  ie pin 0.27!.

So they had to re route that SD card detect pin to the J pins on the Sbase board?
where it can be read.

And why did they route 3 pins to J pin sections

ie..
used to go to P0.8 to J8-3
"     "  "  " P0.7 to J8-2
"     "  "  " P0.5 to J8-4

So the question is why Marlin can't read those 3 pins to begin with?

Bob-the-Kuhn, 2018-05-28:
The problem is that Marlin uses two different SD card reader software routines for the on-board SD card and the external SD card.  They both assume they have excusive access to the hardware SPI channel used for the SD card.

That's OK on the Re-ARM because it uses different SPI channels for the SD cards.  On the Smoothie and clones the same SPI channel/hardware is used for both.

p3p, 2018-05-28:
As Marlin stands there is no centralised SPI access or a way to add multiple sd-cards, due to this, in early development I decided that the onboard sd-card would be used for internal framework purposes and not exposed to Marlin this allowed me to have the onboard sd-card accessible over usb as a mass storage device for flashing, and usable to store the eeprom data. 

The reason those pins (P0.06, P0.07, P0.08, P0.09) are inaccessible for normal GPIO use is that they are configured as hardware SPI (SSP1 Hardware port) for the onboard card, unfortunately unlike the Re-ARM the extension port on the sbase uses the same spi pins, this is currently unsupported for quite a few reasons, (not something the smoothie devs can help with their architecture is very different), even if the rrdfgsc and its sd-card could be used using hardware spi (it can't atm) there would still be the issue of the concurrent accesses from the usb drive which would need disabled along with the eeprom emulation.

I designed our LPC176x support for the Re-ARM, although this makes it very nearly compatible with the S-base and a pins file was added by someone external peripherals are not supported as well yet, it's something I work on when I have time, unfortunately needing to try and maintain compatibility with Arduino style libraries make things more difficult than they otherwise would be.

thinkyhead, 2018-05-28:
> Do you have any hunch / clue / idea as to why you think Smoothieware works right out of the box with external and internal SD card but Marlin doesn't?

I hope the replies above explain sufficiently. They certainly give more intimate detail on the issue than I can! Smoothieware was written for a single board and a single architecture, so it doesn't bump into itself. Marlin is building on previous methods, and sometimes these methods come into conflict. In the future —once the HALs are more mature— these issues will surely be resolved.

boelle, 2019-02-20:
@madmax2 is this still an issue?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-03-15 by p3p

axelloco, 2019-11-26:
I was trying to fix the detection of the SD card, the problem wasn't the logic of P0_27 pin, but it's close to it.
In pins_MKS_SBASE.h we have the SDCARD_CONNECTION option it says what we already know, the connection is ONBOARD so... in SD_CONNECTION_IS(ONBOARD) we have to change the SS_PIN (CSN) to P0_28 or just only change ONBOARD to LCD.

qulet, 2020-05-08:
I have an MKS SBASE v1.3 disc I compiled Marlin 2.0 I connected the octopus cable according to this description: P0,8 to J8-3
P0.7 to J8-2
P0.5 to J8-4
P0.27 to J8-5
Unfortunately, the LCD menu works, but the SD card does not.
I ask for info, what lines in the merlin code must be uncommented for the SD card to work?[
my configuration files
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4601741/Marlin.zip)
https://drive.google.com/file/d/1La0vZjQUHWgPP9kvDKmyDlTTuEoOf--S/view?usp=sharing

help Me !

thinkyhead, 2020-05-10:
@qulet — 
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

madmax2, 2020-05-10:
@thinkyhead
was this issue solved?

thinkyhead, 2020-05-10:
Definitely solved, according to my notes.