MarlinFirmware/Marlin #10873: New feature - Ram purging

erikcrane, 2018-05-27:
### Description

Hi, last day reading e3d forum I found what they call "ram purging" [https://forum.e3d-online.com/threads/edge-and-scaffold-not-printing-correctly.2736/](url)

This method consist in extrude an amount of filament at high feedrate to clear out heat chamfer and do a very quick retract. The filament doesn't have to melt if you realize these movements quick enough.

I think this is a great feature to include in marlin to change filament. I have done the test and the amount of filament that is need to clear, when filament is changed, is very very low.

I have check the code and I think the best way is to change the M600. I can the job but I'm not an expert and my work should be inspect.

Sorry if something is not clear, but my english level is not very good

Best regards

thinkyhead, 2018-05-27:
Marlin has something similar currently. When doing an unload, with the following settings there will be an initial retract at the `PAUSE_PARK_RETRACT_FEEDRATE`, then there will be a fast purge of `FILAMENT_UNLOAD_PURGE_LENGTH` at the maximum E speed, and then the regular unload of `FILAMENT_CHANGE_UNLOAD_LENGTH` happens.
```cpp
  #define FILAMENT_UNLOAD_RETRACT_LENGTH      13  // (mm) Unload initial retract length.
  #define FILAMENT_UNLOAD_DELAY             5000  // (ms) Delay for the filament to cool after retract.
  #define FILAMENT_UNLOAD_PURGE_LENGTH         8  // (mm) An unretract is done, then this length is purged.
```

erikcrane, 2018-05-27:
Hi thinkhead!

If I have understood correctly, first of all Marlin do a retract and in ram purge first of all you have to realize a extrude to clear the hot chamber and then realize a fast retract.

I don't know if you get the same results, I will continue the research

Thanks for your reply :D!

inventabuild, 2018-12-20:
The manufacturer DisTech Automation of the Prometheus hotend does not recommend an initial retraction before purging the filament.  Here is his code for removing the filament from the hotend for a filament change (from here: https://www.distechautomation.com/pages/pro-sys-guide):

G91 ; relative positioning
G1 E18.0 F540 ; 9mm/s purge
G1 E-20 F3000 ; 50mm/s retract
G4 P2500 ; 2.5s cooling period
G1 E-55 F3000 ; 50mm/s long retract

So I'm planning on setting the FILAMENT_UNLOAD_RETRACT_LENGTH to 0.

Has anyone found any benefit to retracting before purging when the extruder is parked?

inventabuild, 2018-12-20:
thinkyhead, is it possible to add a purge feed rate option instead of forcing it to be max feed rate?  To satisfy the above following line I'm forced to put an M203 E9 T0 before M600 then an M203 E
50 T0 after it:

G1 E18.0 F540 ; 9mm/s purge

Or is this option already available?  Thanks.

inventabuild, 2018-12-30:
I finally had some time to play around with this.

I'm running an e3d Titan Aero 0.4 mm noz 1.5 mm filament and in my experience the mystical thinkyhead's infinite wisdom turns out to be right as follows (in case this might be helpful for someone else):

1.  When I perform a "ram purge" followed by fast retract (pause in between or not) I get a deformed clunky head on the end of the filament which will sometimes pull out (or get stuck at) the ptfe tube at the top of the extruder.

2.  When I perform thinkyhead's 13 mm retract (*9mm/sec) + 5 sec pause + 8 mm extrude (*9mm/sec) + 50 mm fast retract (*50 mm/sec) [*are my custom speeds for now], I get a significantly smaller bulb on the head of the filament that is nearly the same diameter of the filament and always pulls right out of the ptfe tube at the top of the extruder without getting stuck.

e3d's "ram purge" may do what it says which is clear out the melted filament in the heated nozzle, but it deforms the end of the filament.  Perhaps some combination of "ram purge" extrusions + retractions will give the best of both worlds, but my limited testing has not found a good combination yet that is better than thinkyhead's default settings for my application:

  #define FILAMENT_UNLOAD_RETRACT_LENGTH      13  // (mm) Unload initial retract length.
  #define FILAMENT_UNLOAD_DELAY             5000  // (ms) Delay for the filament to cool after retract.
  #define FILAMENT_UNLOAD_PURGE_LENGTH         8  // (mm) An unretract is done, then this length is purged.

nionio6915, 2019-03-04:
Is there any followup on this?  I see the parameters sin 2.0x config files, but I can not find any documentation on anyone using this yet. 

Which of the Marlin code files do these get used in?  I would like to try and dig in to read up on this more.

thinkyhead, 2019-03-05:
[image]