MarlinFirmware/Marlin #10874: [2.0x] Teensy 3.x - encoder does not work

android444, 2018-05-27:
Pressing the button works uncertainly.
Turning the knob completely does not work.
I checked on 16MHz and 120MHz.

Where to look for a problem?

Movie in zip file.
[https://youtu.be/-IdReQPcRng.](url)

JXPA, 2018-05-27:
I think that a Logic Level Converter is necessary.
A similar thing happened when I tested Teensy 3.6 in the past.
The LCD used at that time was the Reprap Full Graphic Smart Discount Controller made in China.
Each terminal of the encoder required 5V.
Directly connecting the encoder and Teensy 3.6 has become like a movie, but using Logic Level Converter worked normally.

android444, 2018-05-27:
PULLUP does not work on 3 encoder inputs.
The voltage is about 0.15V and it should be 3.3V.

JXPA, 2018-05-27:
[image]

I do not know the reason for it.
However, by using a logic level converter like an image and operating the encoder at 5 V, it became to work properly.
This is the logic level converter I used.
[https://www.sparkfun.com/products/12009](https://www.sparkfun.com/products/12009)

android444, 2018-05-27:
I used Teensy 3.5.
An additional board forces pull up.

JXPA, 2018-05-27:
It is not necessary to pull-up each pin.
I think the important thing is that the encoder is running between 0V - 5V.
Teensy 3.5 is 5V tolerant, but the logic level voltage is 3.3V.
At 3.3V the encoder does not work properly.
A logic level converter is necessary to operate the encoder at 5V.

I also did a direct connection at first and did not work properly, just like you.
However, I know that the encoder works correctly on Ramps1.4 running at 5V.
So I tried a logic level converter.
As a result, Encoders now work correctly in Teensy 3.6.
I think that similar problems are occurring in Teensy 3.5 too.

android444, 2018-05-27:
You're wrong.
You do not understand how an encoder works.
It's just mechanical contacts.

After adding two pull-up resistors, it works correctly at 3.3V.

The error is in the file fastio_Teensy.h

JXPA, 2018-05-27:
Ah…
Logic level converter is pull-up at 3.3V and 5V.
It seems that he was working well for that.
Thank you for teaching me.

It was a problem that `_PULLUP` in fastio_Teensy.h was wrong.

TadyTheFish, 2018-05-28:
I compiled for 3.5 and I culd not get the encoder to work ether... But i van't remember what Idid to make it work. If you type M43 I are all encoder inputs zero? Even with external pullups. 
I think there is a problem with the pinMode function.. The HAL of the teensy doesn't have it drfined.. I can check when I get back from work

android444, 2018-05-28:
I checked and the pullup pin voltage is 0.15V. It is not enough.
I can add external pullup resistors, but is it necessary?

TadyTheFish, 2018-05-28:
I will check when I get home.. I also had big problems on getting the encoder to work... 
If you take a look at src/HAL/HAL_TEENSY35_36/fastio_Teensy.h
You can see #define _SET_INPUT(P)

And if you take a look at...
src/lcd/ultralcd.cpp

You can see 
```cpp 
 #if ENABLED(NEWPANEL)
    #if BUTTON_EXISTS(EN1)
      SET_INPUT_PULLUP(BTN_EN1);
    #endif
    #if BUTTON_EXISTS(EN2)
      SET_INPUT_PULLUP(BTN_EN2);
    #endif
    #if BUTTON_EXISTS(ENC)
      SET_INPUT_PULLUP(BTN_ENC);
    #endif```

You can see that the last function calls SET_INPUT_PULLUP
But it is not defined in fastio_Teensy...
I made change somewhere pinMode(ENC1,INPUT_PULLUP) and it worked

android444, 2018-05-28:
> 

#define SET_INPUT(IO)         _SET_INPUT(IO)
#define **SET_INPUT_PULLUP**(IO)  do{ _SET_INPUT(IO); _WRITE(IO,HIGH); }while(0)
#define SET_OUTPUT(IO)        _SET_OUTPUT(IO)

>

TadyTheFish, 2018-05-28:
😂 how did I miss that:) but still after i manually declared pinMode() it worked:) The M43 prints ok.. Before it printed all 0.
Like I said I can chec when I get back from work;)

TadyTheFish, 2018-05-28:
Well i took a look at my code...
Anyway I had the same problem.. 
But just for fun I changed this...
```cpp
 #if ENABLED(NEWPANEL)
    #if BUTTON_EXISTS(EN1)
      SET_INPUT_PULLUP(BTN_EN1);
    #endif
    #if BUTTON_EXISTS(EN2)
      SET_INPUT_PULLUP(BTN_EN2);
    #endif
    #if BUTTON_EXISTS(ENC)
      SET_INPUT_PULLUP(BTN_ENC);
    #endif
```

And it works now. M43 also works.. Before it was all 0 all the time (all three encoder inputs)

TadyTheFish, 2018-05-28:
Does my test hack work for you?

android444, 2018-05-28:
Yes.

TadyTheFish, 2018-05-28:
I'm only in to Arduino IDE i don't know C so I also don't know the solution... This is a good fix but it's limited to Arduino IDE :( maybe someone else can shed some light on the issue.

TadyTheFish, 2018-05-28:
The only other fix I can think of is looking in to the core files of Teensy and see what happens when you call pinMode.

p3p, 2018-05-28:
> #define SET_INPUT_PULLUP(IO) do{ _SET_INPUT(IO); _WRITE(IO,HIGH); }while(0)

That macro is not complete, never actually setting the pin mode.. if you see the another platforms for reference 

```C++
#define SET_INPUT_PULLUP(IO)    do{ _SET_INPUT(IO); _PULLUP(IO, HIGH); }while(0)
```

The person that added the platform must have been having trouble with it as the _PULLUP macro is commented out 

```C++ 
//#define _PULLUP(IO,V)  { pinMode(IO, (v!=LOW ? INPUT_PULLUP : INPUT)); }
```

So it definitely isn't adhering to the fastio interface..

android444, 2018-05-28:
Who did it?

TadyTheFish, 2018-05-28:
@teemuatlut is signed on some of the Teensy stuff

teemuatlut, 2018-05-28:
The original Teensy fork by PaulStoffregen is older than the `SET_INPUT_PULLUP` macro.
The current one @p3p posted is the one used by AVR and I believe it's a bit of a trick but it works on that platform. However the Teensy HAL needs the macro written for it so that it targets the correct register and correct bit(s).

The register that controls pin modes is `PORTx_PCRn`.
Bit 0 selects pullup or pulldown.
Bit 1 enables pulls.

Or we can call `pinMode`...

TadyTheFish, 2018-05-28:
this is from Arduino\hardware\teensy\avr\cores\teensy3\pins_teensy.c
```cpp
void pinMode(uint8_t pin, uint8_t mode)
{
	volatile uint32_t *config;

	if (pin >= CORE_NUM_DIGITAL) return;
	config = portConfigRegister(pin);

	if (mode == OUTPUT || mode == OUTPUT_OPENDRAIN) {
#ifdef KINETISK
		*portModeRegister(pin) = 1;
#else
		*portModeRegister(pin) |= digitalPinToBitMask(pin); // TODO: atomic
#endif
		*config = PORT_PCR_SRE | PORT_PCR_DSE | PORT_PCR_MUX(1);
		if (mode == OUTPUT_OPENDRAIN) {
		    *config |= PORT_PCR_ODE;
		} else {
		    *config &= ~PORT_PCR_ODE;
                }
	} else {
#ifdef KINETISK
		*portModeRegister(pin) = 0;
#else
		*portModeRegister(pin) &= ~digitalPinToBitMask(pin);
#endif
		if (mode == INPUT) {
			*config = PORT_PCR_MUX(1);
		} else if (mode == INPUT_PULLUP) {
			*config = PORT_PCR_MUX(1) | PORT_PCR_PE | PORT_PCR_PS;
		} else if (mode == INPUT_PULLDOWN) {
			*config = PORT_PCR_MUX(1) | PORT_PCR_PE;
		} else { // INPUT_DISABLE
			*config = 0;
		}
	}
}
```

android444, 2018-05-28:
And what's next?

TadyTheFish, 2018-05-28:
in ultralcd.cpp i changed this
```cpp
#if ENABLED(NEWPANEL)
    #if BUTTON_EXISTS(EN1)
      SET_INPUT_PULLUP(BTN_EN1);
    #endif
    #if BUTTON_EXISTS(EN2)
      SET_INPUT_PULLUP(BTN_EN2);
    #endif
    #if BUTTON_EXISTS(ENC)
      SET_INPUT_PULLUP(BTN_ENC);
    #endif
```
and in fastio_Teensy.h i added 
```cpp
#define _SET_INPUT_PULLUP(P) do{ \
  CORE_PIN ## P ## _CONFIG = PORT_PCR_MUX(1) | PORT_PCR_PE | PORT_PCR_PS; \
  GPIO_BITBAND(CORE_PIN ## P ## _DDRREG , CORE_PIN ## P ## _BIT) = 0; \
}while(0)

```
and
```cpp
#define SET_INPUT_PULLUP(IO)  _SET_INPUT_PULLUP(IO)
```

works at my end

TadyTheFish, 2018-05-28:
Sorry. I was messing around with ultralcd.cpp so if you have the original you don't need to change it just fastio_teensy.h

TadyTheFish, 2018-05-28:
I learned alot:))) glad it works for all of us Teensy users!