MarlinFirmware/Marlin #10875: [2.0.X-bugfix] Printing at wrong height after [Pausing] -> [G28] -> [Resuming] a print

AcHub, 2018-05-27:
### Description

I was pausing a print in Octoprint after a few layers and then re-homed XY with G28 X0 Y0. This makes the printhead to raise about 2mm and then home as expected. But when I resume the print, the printhead continues to print at the elevated height, instead of lowering back down to the level where it printed before, making the print useless.

### Steps to Reproduce

1. Start a print with Octoprint
2. Pause the print with Octoprint
3. Home XY with the Command-Tools in Octoprint / Send G28 X0 Y0. The printhead will raise ~2mm and then home.
4. Resume Print. The Printhead resumes printing without lowering ~2mm.
5. Fail to resume at correct height

**Expected behavior:**
When resuming the print, the printhead should be moving to the correct XY position for continuing the print and then be lowered back to the correct height before continuing the print.

**Actual behavior:**
Resuming the Print does not lower back to the print-level.

#### Additional Information

Printer is running on 498a328 (May 26).

[config 27.5.18.zip](https://github.com/MarlinFirmware/Marlin/files/2042461/config.27.5.18.zip)

Terminal log from Octoprint:

Send: N2425 G1 X89.256 Y85.100 E0.0094\*98
Recv: ok
Send: N2426 G1 X103.925 Y99.768 E0.7176\*82
Changing monitoring state from "Printing" to "Pausing"
Recv: ok
Send: N2427 M400\*20
Recv: ok
Send: N2428 M114\*27
Recv: X:103.93 Y:99.77 Z:0.20 E:41.67 Count X:8318 Y:7985 Z:77
Recv: ok
Changing monitoring state from "Pausing" to "Paused"
Send: G91
Recv: ok
Send: G28 X0 Y0
Recv: echo:busy: processing
Recv: X:0.00 Y:0.00 Z:2.00 E:41.67 Count X:0 Y:-394 Z:801
Recv: ok
Send: G90
Recv: ok
Changing monitoring state from "Paused" to "Printing"
Send: N2429 G1 X104.241 Y99.518 E0.0139\*90
Recv: ok
Send: N2430 G1 X89.508 Y84.785 E0.7207\*111
Recv: ok

cdedwards, 2018-05-27:
I can verify this occured to me as well. I did this twice on the same print. The first time, everything worked as expected.. The second time, printing resumed at MAX delta height, causing some issues with my mechanical endstops.

thinkyhead, 2018-05-28:
Marlin doesn't know that a print is being paused and resumed; it's simply interpreting the commands being given by the host. Since typical G-code doesn't contain the Z coordinate in every `G1` there's no way the machine knows that it should move back to some previous Z. There are a couple of solutions:

- Don't home with `G28` but simply use `G1 X0 Y0 F6000`. If a Z raise is desired, it should be done in relative mode with `G91`;`G1 Z2`;`G90` and before resuming the print, a relative Z lower should be done with `G91`;`G1 Z-2`;`G90`.

-or-
- Enable `PARK_HEAD_ON_PAUSE` and use the `M125` command to pause from the host. Make sure `EMERGENCY_PARSER` is enabled so you can resume with `M108`, otherwise you must resume with the LCD controller.

thinkyhead, 2018-05-28:
@MarlinFirmware/host-software-team — Just a note that since `G28 X Y` might do a raise depending on its configured behavior, and some may want to use this command as part of pausing a print, the host should remember the current (last sent) Z position on pause and return to that Z position as part of resuming a print.

foosel, 2018-05-28:
For the record, OctoPrint already records the position of the print head before executing the pause script. If OP had reset X/Y/Z positions in their resume script using that information [as documented](http://docs.octoprint.org/en/master/features/gcode_scripts.html#more-nifty-pause-and-resume) there probably wouldn't have been an issue - the above log doesn't look like this was the case though, explaining the observed behaviour.

cdedwards, 2018-05-28:
I was printing from SD card and therefore used the LCD pause/resume function. my USB serial function is too flaky with the rearm board to print from a host.

AcHub, 2018-05-28:
>For the record, OctoPrint already records the position of the print head before executing the pause script.

Am I getting you right @foosel, Octoprint will resume at the correct position by default? Because I didn't change any settings regarding the pause/resume behaviour of Octoprint. And pausing -> G28 -> resuming within Octoprint did work as expected before (before means like months ago with some older version of a 1.x Marlin firmware)

thinkyhead, 2018-05-28:
In the case of SD Card pause / resume, when the `PARK_HEAD_ON_PAUSE` feature is enabled, pausing from LCD or with `M25` invokes `M125` to move the carriage to the configured park position. Resuming from the LCD or with `M24` will move the carriage back to where it was when resuming, and it should not be bothered by any movement in Z during the pause.

cdedwards, 2018-05-29:
speaking of which.
#define NOZZLE_PARK_POINT { (X_MIN_POS + 10), (Y_MAX_POS - 10), 20 }
this is not a good place to park a delta while printing. I know kossel XL at least has this for a parking position.

cdedwards, 2018-06-04:
I was thinking more the height. 20mm above the bed no matter where it is on the bed is a lousy place to park. If it’s relative it might be ok but could cause issues near max height

thinkyhead, 2018-06-10:
The Z value is relative, and constrained to avoid going past the top.

boelle, 2019-02-20:
@AcHub is this still an issue?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-13 by thinkyhead