MarlinFirmware/Marlin #10877: [2.0.x] Building multiple configurations at once

waddyano, 2018-05-27:
I am sure I could come up with some approach of my own but I wondered if there was a suggested technique of easily building multiple configurations at once e.g. same platformio environment but different Configuration.h?

Bob-the-Kuhn, 2018-05-28:
PlatformIO actually tries to compile every environment in platformio.ini.  It just skips the ones that don't apply to the specified target.

In theory you could set a flag in each environment and then a bunch of #if statements could make the selections.  You'd also need to modify the build folder for each flavor.

With PlatformIO it's like a dog walking on it's hind legs, no pretty or weffiicent but it can be done.

You're better off going with Visual Studio **Community**.  It's designed for what you're wanting to do.

waddyano, 2018-06-01:
Thanks for the reply hadn't had time to comment earlier. Indeed the way I did it so far is to create a new environment with an extra compile switch and #ifdefs. I actually didn't realize that you could use full visual studio - currently using VS Code or often the command line but have just looked at what sort of project it creates. It is just a makefile based project that drives the pio command line so not really that much different. Now I am very used to visual studio for editing - more so than vs code - but  "platformio init --ide" doesn't seem to do a good job of creating a sensible header and source folder structure inside the solution - maybe I will try and write a script to organize the project/solution better.

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-05-16 by thinkyhead