MarlinFirmware/Marlin #10878: Servomotor is shaking with G28

hevilp, 2018-05-27:
### Description

Servomotor is shaking random when homing (G28)

### Steps to Reproduce

1.  I set #define NUM_SERVOS to 1  and connected my Servomotor (HS-225MG )
2. Send G28

All axis goes to home, but the Servormotor is shaking around

If I do not home, the motor works nice with M280

Why does it shakes and struggles 
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/2042880/Configuration.txt)
when I use G28?

GMagician, 2018-05-27:
What's the servo purpose?
No BLTOUCH, SWITCHING_EXTRUDER, SWITCHING_NOZZLE
Sure it's not EMI noise?

GMagician, 2018-05-27:
If you try to move steppers? what happens

hevilp, 2018-05-27:
The x y z moves normal, jittering is still there

GMagician, 2018-05-27:
So issue is when you move axes, not home as a command...emi noise I think

hevilp, 2018-06-08:
I connected the servo now on 5V on the MKS GEN L and got no problems, before they got the power direct from the PSU (lowered to 5V)

GMagician, 2018-06-08:
The problem was differential in potential on GND I suppose