MarlinFirmware/Marlin #10879: Marlin 2.0 running on Teensy 3.5 servo problem?

TadyTheFish, 2018-05-27:
Hello!

Out of curiosity I was playing aaround with the Marlin 2.0 on Teensy 3.5.
When I wanted to test the BLTouch function I came across a problem.
When I enabled it the pin 0 (witch is the X DIR pin on the standard pin map) stated blinking very fast. The motor was not able to turn but it buzzed around. So i hooked up my scope and saw that the frequency is 50Hz. That made me belive that there is something wrong. I was digging around the source files and came across this line...
`void libServo::move(const int value) {
  if (this->attach(0) >= 0) {
    this->write(value);
    delay(SERVO_DELAY);
    #if ENABLED(DEACTIVATE_SERVOS_AFTER_MOVE)
      this->detach();
    #endif
  }
}`
This is inside HAL_Servo_teensy.cpp
This line made no sense to me. But after removing the if sentence the adding... 
`this->attach(SERVO0_PIN);`
The firmware started to work perfectly.. 
I don't know if i broke anything else... But can some one please tell me why is this written like it is...
This is my change....

`void libServo::move(const int value) {
  //if (this->attach(0) >= 0) {
	this->attach(SERVO0_PIN); // Added this line since I only use BLTouch and no other servos
    this->write(value);
    delay(SERVO_DELAY);
    #if ENABLED(DEACTIVATE_SERVOS_AFTER_MOVE)
      this->detach();
    #endif
  //}
}`

Thank you!

TadyTheFish, 2018-05-27:
it seems code tags break the code structure...
original...

void libServo::move(const int value) { 
    if (this->attach(0) >= 0) { 
       this->write(value); 
       delay(SERVO_DELAY); 
       #if ENABLED(DEACTIVATE_SERVOS_AFTER_MOVE) 
          this->detach(); 
      #endif
     }
 }

my edit...
void libServo::move(const int value) { 
       this->attach(SERVO0_PIN);  // Added this line since I only use BLTouch and no other servos
       this->write(value); 
       delay(SERVO_DELAY); 
       #if ENABLED(DEACTIVATE_SERVOS_AFTER_MOVE) 
          this->detach(); 
      #endif
 }

It seems that the original code attaches the SERVO0_PIN to the servo and then detaches (this is normal i saw that) and then when it wats to move the servo it always attaches pin 0 to the servo and writes to that pin...
This has to be a bug

Thank you!

TadyTheFish, 2018-05-27:
Hmmm.. Never looked at other board sources... Teensy uses native servo library.
What is the reason for this function? his->attach(0)

GMagician, 2018-05-27:
In avr, but I think generrally, `attach` is used to start timer and 0 means don't change assigned pin.
Pin is assigned to servo in `servo_init` that resides in servo.cpp

TadyTheFish, 2018-05-27:
So it seemsit doesn't work for the teensy..

Thank you for the explanation!

GMagician, 2018-05-27:
servo_init is called for every board, sure pin is not already used for somerthing else?

GMagician, 2018-05-27:
Enable PINS_DEBUGGING and use M43 to view the pins mapping.

GMagician, 2018-05-27:
@thinkyhead there is a reason why avr uses safe_delay in servo and teensy (and all stm) use delay, without access to indexed value

> I see something different from avr source:
delay(SERVO_DELAY);
instead of:
safe_delay(servo_delay[this->servoIndex]);

TadyTheFish, 2018-05-27:
100% i used different pins.. I added Serial.println inside the native library for the teensy and it printed out strange stuff... When the board boots up it prints the pin I assigned then it prints a detach and then when it tests the probe it prints 0 and then the angle and then 0 again and another angle.. I don't have my computer here but the short story is that at boot up the pin gets attached and detached.. And every time M280 is called it fors attaches to pin 0 and then it writes thecdesired angle

GMagician, 2018-05-27:
servo_init call 'attach' and 'detach'. First to setup associated pin

GMagician, 2018-05-27:
... and everytime you move you have attach(0) to start PWM and detach only when you define DEACTIVATE_SERVOS_AFTER_MOVE

TadyTheFish, 2018-05-27:
I know that.. And the attach prints the correct pin.. Then detach.. I know that.. But every call to M280 P0 Sxxx prints attach to pin 0..
M43 prints the correct pin number (defined by SERVO0_PIN

GMagician, 2018-05-27:
attach will execute this code:
```c++
int8_t Servo::attach(const int pin, const int min, const int max) {

  if (this->servoIndex >= MAX_SERVOS) return -1;

  if (pin > 0) servo_info[this->servoIndex].Pin.nbr = pin;
  pinMode(servo_info[this->servoIndex].Pin.nbr, OUTPUT); // set servo pin to output
```
but pin is used only when > 0

thinkyhead, 2018-05-28:
> is there a reason why avr uses `safe_delay` in servo and teensy (and all stm) use `delay`, without access to indexed value?

I don't know without investigating. Just speculating, it may be that `safe_delay` is not usable with certain platforms until after they get into the main loop.

teemuatlut, 2018-05-28:
A likely reason is that there was no `safe_delay` when I worked on the Teensy HAL and perhaps it just was missed when the change was introduced.

thinkyhead, 2018-05-28:
> it seems code tags break the code structure...

Bracket with three backticks and an optional language extension. ` ```cpp ` . . . ` ``` `

thinkyhead, 2018-05-28:
> A likely reason is that there was no `safe_delay` when I worked on the Teensy HAL…

[image]

🤓

GMagician, 2018-05-28:
> #if IS_32BIT_TEENSY
  #include "HAL_TEENSY35_36/HAL_Servo_Teensy.h" // Teensy HAL uses an inherited library

@TadyTheFish I missed that part... but in this .h you can see:
```c++
// Inherit and expand on the official library
class libServo : public Servo {
[cut]
```

some original method are overridden but I think not the attach one

TadyTheFish, 2018-05-28:
Someone made a fix  few hours ago (added a safe delay) but this does not fix the problem. At boot-up the pin gets assigned to the specified pin and then detaches. every time I issue a M280 the firmware attaches to pin 0 and wtites to it. it is always pin 0. This interferes with X dir pin.

I don't know why this line is in HAL_Servo_teensy.cpp but it should be deleted. If there is a reason for this line in the AVR part of the code it does not work with the native teensy library.
```cpp
if (this->attach(0) >= 0) {
```

teemuatlut, 2018-05-28:
The two attach methods are overloaded from the parent class.

```cpp
if (this->attach(0) >= 0)
```
This does not call the parent method, but the overloaded one.
Basically it does the check `servoIndex >= MAX_SERVOS`, because if this is true then `attach` will return a negative value and we can't write to the servo. Later the parent `Servo::attach` returns an unsigned value and the move can happen.

thinkyhead, 2018-05-29:
> The two attach methods are overloaded from the parent class.

The servo portion of the HALs could all definitely use some cleanup. Much redundancy.

thinkyhead, 2018-05-29:
> @GMagician : In avr, but I think generrally, attach is used to start timer and 0 means don't change assigned pin.

The Teensy Servo library has a bug. We'll have to work around it. Here's how it does `attach`, and as you can see it doesn't bother to check if the pin passed is 0.
```cpp
uint8_t Servo::attach(int pin, int min, int max)
{
  if(this->servoIndex < MAX_SERVOS ) {
    pinMode( pin, OUTPUT) ;                                   // set servo pin to output
    servos[this->servoIndex].Pin.nbr = pin;  
    // todo min/max check: abs(min - MIN_PULSE_WIDTH) /4 < 128 
    this->min  = (MIN_PULSE_WIDTH - min)/4; //resolution of min/max is 4 uS
    this->max  = (MAX_PULSE_WIDTH - max)/4; 
    // initialize the timer if it has not already been initialized 
    timer16_Sequence_t timer = SERVO_INDEX_TO_TIMER(servoIndex);
    if(isTimerActive(timer) == false)
      initISR(timer);    
    servos[this->servoIndex].Pin.isActive = true;  // this must be set after the check for isTimerActive
  } 
  return this->servoIndex ;
}
```

thinkyhead, 2018-05-29:
The following change to `HAL_Servo_Teensy.cpp` should make everything work:
```cpp
uint8_t servoPin[MAX_SERVOS] = { 0 };

int8_t libServo::attach(const int pin) {
  if (this->servoIndex >= MAX_SERVOS) return -1;
  if (pin > 0) servoPin[this->servoIndex] = pin;
  return Servo::attach(servoPin[this->servoIndex]);
}

int8_t libServo::attach(const int pin, const int min, const int max) {
  if (pin > 0) servoPin[this->servoIndex] = pin;
  return Servo::attach(servoPin[this->servoIndex], min, max);
}
```

thinkyhead, 2018-05-29:
Evidently you can't even compile STM32F1R with servos. The HAL code for this seems to be essentially broken. The STM32 platform doesn't even use the same methodology for servos that the other platforms do. Each servo in STM32 is an instance of the `Servo` class, rather than just an element in an array within the `Servo` class.

TadyTheFish, 2018-05-29:
It works now. 
Thank you!

closed 2018-05-29 by TadyTheFish