MarlinFirmware/Marlin #10880: Trigorilla Ramps 1.4 Pin 9 and Pin 44 operate reverse of configuration.

ArtieH, 2018-05-27:
### Description

What I found is that on the Trigorilla Ramps 1.4 board, the firmware swaps Pin 9 and Pin 44 operation and may be the cause of much of the confusion among Trigorilla owners as to fan configuration.

The forum fix I have seen is to swap the fan connections. This does not seem to be a solution since there are now boards defined for Trigorilla 1.3 and 1.4 boards.

To prove the swap issue I defined:

## Step 1 ##
>Hot End cooling fan: E0_AUTO_FAN_PIN 44
>Parts cooling fan: FAN_PIN 9

**pins\_RAMPS.h**
added under "Heaters and Fans"
 >     #ifndef RAMPS_D44_PIN // This added to define E0 Hot End Cooling fan 5/17/18
 >          #define RAMPS_D44_PIN 44
 >     #endif

and a few lines further down

>     #if ENABLED(IS_RAMPS_EFB)                      // Hotend, Fan, Bed
>          #define FAN_PIN       RAMPS_D9_PIN
>          #define E0_AUTO_FAN_PIN  RAMPS_D44_PIN // This added to setup auto fan on E0
>          #define HEATER_BED_PIN RAMPS_D8_PIN

## Step 2 ##

The fans in my printer are connected per the Anycubic Kossel Delta Plus build:
>Hot End cooling fan connected to pin 9 (Fan 0)
>Parts cooling fan connected to pin 44 (Fan 2)

## Step 3 ##

Upload firmware and check configuration
*M43 pin debug lists the following:*
>     PIN:   9   Port: H6  FAN_PIN              protected 
>                          RAMPS_D9_PIN         protected
>     PIN:  44   Port: L5  E0_AUTO_FAN_PIN      protected 
>                          E1_CS_PIN            protected

## Expected Behavior ##
What I expect is that the fans are now defined reversed of their function and will not work correctly.

## Actual behavior: ##
Subsequent printer operation proves this not to be the case. The configuration and debug listing is opposite of the results. The Hot End fan operates correctly and spins up at 50c. The parts cooling fan operates correctly under control of the g-code.

#### Additional Information
I have tried BOARD_TRIGORILLA_13 and BOARD_TRIGORILLA_14 configurations, without the above test definitions. I found that, with the fans connected per the Anycubic build, the Parts cooling Fan operates as the Hot End cooling fan and the Hot End fan does not spin up at all.

[My Configuration_Files.zip](https://github.com/MarlinFirmware/Marlin/files/2042967/My.Configuration_Files.zip)

thinkyhead, 2018-05-28:
Which board do you have, 1.3 or 1.4?

Our current pins are based on [this comment](https://github.com/MarlinFirmware/Marlin/issues/8349#issuecomment-377772053) when we were splitting up 1.3 and 1.4. Would you say that it was incorrect…?

| pin | 1.3 | 1.4 |
|----|----|-----|
|FAN0| D44 | D9 |
|FAN2| D9 | D44|

---
We want to try and allow some flexibility, so the so-called "`RAMPS_D9_PIN`" (the middle MOSFET connector) can be assigned to fan, bed, or extruder based on `IS_RAMPS_EFB`, `IS_RAMPS_EEF`, etc.

Under the current bugfix branches, you can do the following in `pins_TRIGORILLA_13.h` or `pins_TRIGORILLA_14.h` to assign the print-cooling fan to pin 9 and the auto-fan to pin 44:

```cpp
#define RAMPS_D9_PIN         9
#define ORIG_E0_AUTO_FAN_PIN 44

#include "pins_RAMPS_13.h"
```

If this is more correct for either 1.3 or 1.4 (or both), then we can apply the change permanently.

ArtieH, 2018-05-28:
Thank you for your time.

My board is a 1.4. I've never worked with a 1.3 board, so I can't say if the 1.3 pinout you list is correct. The 1.4 pinout in your diagram is correct.

I will test the solution and let you know the results.
But, this solution is a workaround to another problem. Which may be the way you want to go, since it's crunch time on 2.0. My concern is that the underlying issue may be carried forward.

If you look at the M43 output:
- Pin 9 is defined as FAN_PIN, which defines the Trigorilla "E1 cooling fan" pin to be the "Parts cooling fan" pin.
- Pin 44 is defined as E0_AUTO_FAN_PIN, which defines the Trigorilla "Parts cooling fan" pin to be the "E1 cooling fan" pin.

Or to put it another way:

<table>
  <tr>
    <th>Pin</th>
    <th>1.4 Function</th>
    <th>Marlin Config</th>
    <th>M43 Pin Debug</th>
    <th>Actual Oservation</th>
  </tr>
  <tr>
    <td>9</td>
    <td>E1 Cooling</td>
    <td>Parts Cooling</td>
    <td>Parts Cooling</td>
    <td>E1 Cooling</td>
  </tr>
  <tr>
    <td>44</td>
    <td>Parts Cooling</td>
    <td>E1 Cooling</td>
    <td>E1 Cooling</td>
    <td>Parts Cooling</td>
  </tr>
</table>

As can be seen, there is a swapping of defined functions.

ArtieH, 2018-05-28:
## Results of test
I applied the changes you suggested.
- The Parts Cooling Fan operates properly under control of G-Code. (M106/M107)
- The E1 Cooling Fan is non-functional. Does not spin up on E1 temp above 50c.

M43 listing
>    PIN:   9   Port: H6        FAN_PIN                     protected 
>    .                          RAMPS_D9_PIN                protected 
>    
>    PIN:  44   Port: L5        E_MUX2_PIN                  protected 
>    .                          FAN2_PIN                    protected 
>    .                          ORIG_E0_AUTO_FAN_PIN        protected 

I also tried undefining the FAN2_PIN and E_MUX2_PIN but got the same results.

Shouldn't I be seeing, on pin 44, an assignment of E0_AUTO_FAN_PIN, either with ORIG_E0_AUTO_FAN_PIN assignment or without?

I've attached the Configuration.h, Configuration_adv.h, pins_TRIGORILLA_14.h and pins_RAMPS.h
I commented out all my original edits and added your suggested changes.

[Modified Marlin Files.zip](https://github.com/MarlinFirmware/Marlin/files/2045687/Modified.Marlin.Files.zip)

thinkyhead, 2018-05-29:
You have the E0 auto fan pin disabled in your `Confgiuration_adv.h` file.
```cpp
#define E0_AUTO_FAN_PIN -1
```
Comment out that line to enable it on the `ORIG_E0_AUTO_FAN_PIN` pin.

ArtieH, 2018-05-29:
Results # 2:

That worked!!! Both fans are operating correctly now, although they are defined in reverse.

M43 listing
>     PIN:   9   Port: H6        FAN_PIN                     protected 
>     .                          RAMPS_D9_PIN                protected 
>    
>     PIN:  44   Port: L5        E_MUX2_PIN                  protected 
>     .                          E0_AUTO_FAN_PIN             protected 
>     .                          FAN2_PIN                    protected 
>     .                          ORIG_E0_AUTO_FAN_PIN        protected 

This is a workable patch within the configuration schema. But, the underlying issue remains.
Unless you wish to continue to identify the pin swap issue I will consider this closed.
I understand what cruch time is like. Been there many times, many long nights and weekends. Thank you so much for your time.

ArtieH, 2018-05-29:
On the Trigorilla 1.4 board your presumption is correct.
>     #define FAN2_PIN             44
>     #define ORIG_E0_AUTO_FAN_PIN 44
>     
>     #include "pins_RAMPS_13.h"

Fans are working correctly even though their ports are defined in reverse.

Originally, I didn't do changes or testing on the Trigorilla 1.3, since I am not familiar with that board, and my printer has the 1.4 board. I did some searches and it seems that as far as the internet is concerned the 1.3 board doesn't exist. All links that I identified concerning 1.3, point back to the Marlin forums.

Below is the results of some testing I performed:

The oringinal pins_TRIGORILLA_13 config did not operate the fans on my printer. I applied your presumed changes to the file and tried to compile. I ran into the infamous compile error of FAN_PIN and E0_AUTO_FAN_PIN cannot be defined to the same pin. I made a few modifications and what I found is that it is basically the same configuration as pins_TRIGORRILA_14, with one exception; I commented out the FAN2_PIN definition.

*pins_TRIGORILLA_13*
>     #define RAMPS_D9_PIN       44
>     #define FAN_PIN   9            
>     //#define FAN2_PIN            9 
>     #define ORIG_E0_AUTO_FAN_PIN RAMPS_D9_PIN

*pins_RAMPS.h*
>     #ifndef FAN_PIN                         //Added for pins_Trigorilla_13 to prevent reassignment of FAN_PIN
>       #define FAN_PIN         RAMPS_D9_PIN
>     #endif                                  //Added for pins_Trigorilla_13 to prevent reassignment of FAN_PIN

*M34 Listing*
> PIN:   9   Port: H6        FAN_PIN                     protected
>     .                          RAMPS_D9_PIN                protected
>
>     PIN:  44   Port: L5        E_MUX2_PIN [...]

After the changes there is no longer a conflict between FAN_PIN and E0_AUTO_FAN_PIN. Also the config runs on my printer because it is the same as the 1.4 configuration, just implemented differently. Is it possible that 1.3 and 1.4 are actually the same? The confusion may lie in the swapping of pin definitions, either in Marlin or within the Trigorilla board?

There is a mystery afoot.....

ArtieH, 2018-05-31:
I've discovered that the mystery is myself. At some point in time I swapped, in my recollection, the designations of the fan connections on the Trigorilla 1.4 board. I was under the misconception that Fan0 (Pin 9) is the E1 Cooling fan and Fan2 (Pin 44) is the parts cooling fan; I am __incorrect__.

__Fan0 is the parts cooling fan and Fan 2 is the E1 Cooling fan.__

In spite of this we did get the configuration in pins_TRIGORILLA_14 correct. Only the reason it didn't work originally has changed. Please disregard my pin swapped comments. I apologize for adding to the confusion with the Trigorilla boards.

Thank you for all of the assistance.

closed 2018-05-31 by ArtieH

thinkyhead, 2018-06-01:
Since I did end up patching the Trigorilla pins files in response to this issue, can you double-check to see whether they are now correct in the bugfix branches? I believe they should be, as I simply made very basic corrections.

ArtieH, 2018-06-01:
I tested both configurations.

The BOARD_TRIGORILLA_14 configuration operates on my printer correctly.
The E1 Cooling fan spins up at temperature and the Parts cooling fan is under control of gcode.

BOARD_TRIGORILLA_14
> PIN:   9   Port: H6        FAN_PIN                     protected
>     .                          RAMPS_D9_PIN                protected
>
>     PIN:  44   Port: L5 [...]

The BOARD_TRIGORILLA_13 configuration operates on my printer as I would expect from your earlier pin designation diagram.
The E1 Cooling fan is defined to my Parts cooling fan, and vice-versa.

BOARD_TRIGORILLA_13
>     PIN:   9   Port: H6        E0_AUTO_FAN_PIN             protected 
>     .                          FAN2_PIN                    protected 
>     .                          ORIG_E0_AUTO_FAN_PIN        protected 

>     PIN:  44   Port: L5        E_MUX2_PIN                  protected 
>     .                          FAN_PIN                     protected 
>     .                          RAMPS_D9_PIN                protected 

In my opinion the fans are operating per the earlier diagram you posted.

Thank you again.

viktorglekler, 2018-12-11:
Hi ArtieH! Do you know how to change pwm frequency for E0_AUTO_FAN (pin 44) ? I have tried to assign E0_AUTO_FAN_PIN to FAN1 (pin 7) or FAN2 (pin 44) and then using FAN_SOFT_PWM or FAST_PWM_FAN, but it does not work.. In this configuration the fan does not spin at all.