MarlinFirmware/Marlin #10884: Trinamic overcurrent handling and error flags? (trinamic) (question)

AletheianAlex, 2018-05-28:
I suppose tagging @teemuatlut is appropriate for this one, but I am about to run some failure-mode tests, and I was wondering: TMC2130 claims in it's data sheets to have overcurrent protection, but there are no error flags that I can see in the data sheets or the Marlin driver libraries, and no information from Trinamic that I could find that specifically cites the tmc21xx series.  Did I miss some other literature (or some obscure comment in the 84 page data sheet), or reference in the Marlin libs... or  is the operation undocumented/assumed to be the same as drivers from other series?  If it is unknown, I will try bugging Trinamic about it.

(This relates to the shifting layers thread -- as far as eliminating user error and/or bad settings --but I didn't want to cause another hardware divergence in a thread that is ostensibly about hunting down a software issue. I plan to put a scope on the driver current output and plot current vs speed vs temperature under a variety of conditions to determine which conditions incur unexpected temporary current drop, and it would be helpful to know the expected overcurrent protection behavior.  )

I did dig this up in the data sheet that wrt Open Load Flags, but little else :  

`"One or both flags are active, if the current regulation did not succeed in scaling up to the full target current within the last few fullsteps (because no motor is attached or a high velocity exceeds the PWM limit)."`

That suggests that some sort of current sensing is tied to other error flags (I suppose in the way that bad CS wiring flags otpw or gnd fault, unless that is handled on the Marlin end).  Other trinamic chips have overcurrent flags for OCa and OCb phases, as well as bridge high-side, and their overcurrent behavior is:

`"...the PWM cycle is terminated at once and all transistors of the bridge are switched off for the rest of the PWM cycle. The error counter is increased by one. If the error counter reaches 3, the bridge remains switched off for 63 PWM cycles and the error flag is read as “active”...The error counter is cleared, whenever there are more than 63 PWM cycles without overcurrent..."`

Which seems common across several chip series, so maybe it applies?

The idea would be to compare that knowledge to:

`"...for a higher velocity range, variations of the motor back EMF caused by motor acceleration and deceleration may require a quicker regulation. Therefore, PWM_GRAD setting should be optimized for the fastest required acceleration and deceleration ramp...during a deceleration phase the voltage PWM amplitude must be adapted in order to keep the motor coil current constant. When the upper acceleration and the upper deceleration used in the application are identical, the value determined for the acceleration phase will already be optimum for both."`

With regard to `tmc2130_init` function's `st.stealth_gradient()` and `st.stealth_amplitude()` when switching over the hybrid mode spread/stealth threshold, OR just from an abnormality stemming from a poor setup / motor matching / layer change / odd movements / etc. 

Thank you.

teemuatlut, 2018-05-28:
The error flags are in `DRV_STATUS` register. See bits 24..30.
A more general flag is present in `GSTAT` register.
Some of the bits from these registers are mirrored in `SPI_STATUS` response nibble.
A brief description of the diagnostic flags is in section 12 of the TMC2130 data sheet.

Your "other" TMC chips seem to be variants that I've not dealt with (TMC249?) and the first two quotes do not apply to the ones we use.

A few week back I talked with Trinamic about the default parameters we use for the drivers. One of the settings we will be tweaking is the `stealth amplitude`, going from full 255 to 180. The reason given to me was "to avoid over current upon switching back [to stealthChop from spreadCycle]".
`stealth gradient = 5` apparently should be good, but I suspect this might also be a value that is somewhat dependent on the used stepper motor characteristics as well as supply voltage. You can try tweaking the parameter but it might not make noticeable difference.

AletheianAlex, 2018-05-29:
I got an answer from Trinamic:

"the high-side MOSFETs are protected against overcurrent by measuring voltage drop during ON-state. There is no difference to the short detection - it triggers the S2G-Flags."

Any least the clears up one mystery.

boelle, 2019-06-22:
if there are not more hidden "gems" in this one :-D

closed 2019-07-09 by shitcreek