MarlinFirmware/Marlin #10886: Firmware retract w/z-hop doesn't update Z position, firmware recover doesn't restore correct Z.

zarthcode, 2018-05-28:
### Description

Firmware retract w/z-hop doesn't update Z position, firmware recover doesn't restore position.

### Steps to Reproduce

M207 S6 F9000 Z1  ; G10 Retract settings
M208 S-0.05 F900 ; G11 Recover settings

G0 Z0; Move nozzle to bed
G10; Retract - Zhop occurs,
M114; Z is registered as still 0
G11; Recover occurs, but Z doesn't move back down!

**Expected behavior:**

G10 should update position
G11 should recover position

**Actual behavior:**

Non-existent layer adhesion. ;-)

#### Additional Information

Performed on latest bugfix, 5/28

thinkyhead, 2018-05-28:
Duplicate of #10200

thinkyhead, 2018-05-28:
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as requested in the Issue template) and drop them into your next reply. We'll check them over and see if anything is amiss.

thinkyhead, 2018-05-30:
As the patch totally fixes your issue I'll close this. If it crops up again we'll hit it with a bigger stick!

closed 2018-05-30 by thinkyhead