MarlinFirmware/Marlin #10888: Marlin 2.0, Arduino Due, RR Discount Full Graphic Smart Controller blank screen on power up.

juhjar, 2018-05-28:
Hi, I have set up Marlin 2.0, RRD Full Graphic Smart Controller running on Arduino DUE. Works ok.. _almost_ Downloaded Marlin 2.0 at 28.05.2018.

**Expected behavior:** 
- Startup logo of Marlin and normal operation.

**Actual behavior:** 
- After power up, blank screen with back light on
- Encoder button does not cause beep.
- When 'Reset'-button of DUE is pressed display wakes up and works ok.

Steps to Reproduce:

1. Power on, only back light on display.
2. Press Reset on DUE, display starts to work ok.

#### Additional Information
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2045759/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/2045760/Configuration_adv.zip)

+5v is supplied to display.
Check video about issue:  [https://youtu.be/1aL8L20r0wE](url)

ejtagle, 2018-05-28:
I have exactly the same configuration, and it works for me. But, long time ago, we found some issues on the reliable reset of the Due. Maybe the problem the hw related problem we suffered for a long time...

juhjar, 2018-05-28:
So while testing more, found out that when display is connected to 3,3v supply, display initializes ok and encoder push button makes beep. But display is very faint, barely visible texts, contrast adjustment is not enough.

When display supply is +5v, display does not initialize on power up, need to press reset button of DUE after that it initializes...

ejtagle, 2018-05-28:
That could be noise picked up byt the LCD cabling. I have an ST7920 display powered by 5v hooked to the DUE, and i have no issues (yes, 3.3v is not enough)

juhjar, 2018-05-29:
Solved!! Need to solder 10k resistor on board... Now DUE starts every time.

[http://forum.arduino.cc/index.php?topic=256771.msg2512504#msg2512504](url)

closed 2018-05-29 by juhjar