MarlinFirmware/Marlin #10889: Sensorless Homing CoreXY issue

smoki3, 2018-05-28:
Hi,
I have a problem with sensorless homing on my coreXY. 

After a reboot of the printer or firmware reset, on the first homing the printer crashes at the end positions and the motors are loosing a few steps. After a few seconds the homing will success anyway.

After the first homing is finished, it is working as expected (no crashes).

Using TMC2130 on XY Axis on my RADDS board. 

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/2045957/Config.zip)

th33xitus, 2018-05-29:
I have some issues with Sensorless Homing again as well.
Right now i am using commit `cbcb284` and didn't changed anything on my setup so far.
I get quite a decent amount of false positives and setting the Bump Sensitivity to a higher value doesn't affect anything but i can't really explain why the issue occurs.

When i do an initial homing of all axes, everything is fine, the nozzle parks at Z0 in the middle of the print bed. For testing purposes i then tried to home X alone. So i send a G28 X to the machine, Z raises a little bit and then nothing. X doesn't move at all. When i send a M122 i get this output:
```
Send: M122
Recv: 		X	Y	Z	E1
Recv: Enabled		false	false	false	false
Recv: Set current	670	670	670	670
Recv: RMS current	642	642	642	642
Recv: MAX current	905	905	905	905
Recv: Run current	20/31	20/31	20/31	20/31
Recv: Hold current	10/31	10/31	10/31	10/31
Recv: CS actual		20/31	10/31	10/31	10/31
Recv: PWM scale	117	68	67	63
Recv: vsense		1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true
Recv: msteps		16	16	16	16
Recv: tstep		1048575	1048575	1048575	1048575
Recv: pwm
Recv: threshold		79	79	658	276
Recv: [mm/s]		100.08	100.08	3.00	30.03
Recv: OT prewarn	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false
Recv: off time		5	5	5	5
Recv: blank time	24	24	24	24
Recv: hysteresis
Recv: -end		2	2	2	2
Recv: -start		3	3	3	3
Recv: Stallguard thrs	35	26	0	0
Recv: DRVSTATUS	X	Y	Z	E1
Recv: stallguard				X
Recv: sg_result		810	0	0	0
Recv: fsactive
Recv: stst		X	X	X	X
Recv: olb
Recv: ola				X
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:	X = 0x80:14:03:2A
Recv: 	Y = 0x80:0A:00:00
Recv: 	Z = 0xA1:0A:00:00
Recv: 	E1 = 0x80:0A:00:00
```
I guess the `sg_result` of 810 for X is the indicator that stallguard kicked in in this case? Sometimes i get a value of only 4, sometimes of 860. So the value isn't always the same. As you can see i already cranked up the Bump Sensitivity to 35 (was 26 before - same as Y and Y is working flawless - X worked flawless before too).

When i send a M18 and disable the steppers, then i can home X alone and it works fine.
But as soon as i home all axes again and then try to home X alone, it stucks again.
Something there is triggering something :D 
My mechanics are fine, since it worked before, so that's not the issue (i even freshly oiled the bearings and the X-Rods to make it glide even smoother again).

I also activated the `ENDSTOP_INTERRUPTS_FEATURE` and tried again, no change. Still the same issue.

I hope this is the correct place for my issue? Maybe my issue is correlating to the initial issue?

smoki3, 2018-05-29:
https://photos.app.goo.gl/WrK6mUcAgC4QPuRR2

I made a short video to describe the issue. The first homing is the initial, the second after the initial

thinkyhead, 2018-05-29:
It could be that the endstop hit state isn’t getting cleared after the initial hit, though we did attempt to patch that possibility just a couple days ago. I’ll take a closer look at the corexy endstop testing code-path and see if I can discern a logical flaw.

Give it a test both with and without quick-home to see if there’s any noticeable difference.

smoki3, 2018-05-29:
So I just enabled quick home. Still the crashing

https://photos.app.goo.gl/cTgA6vD6AeKDwf3G3

thinkyhead, 2018-05-29:
Sure, that was expected. I’ll scrutinize the new video to see if there are any clues about the issue I can discern in the behavior.

thinkyhead, 2018-05-29:
In the second video the initial quick-home XY move is stopping way too soon. There must be a false trigger happening during that first XY move. That might indicate that the sensorless homing sensitivity is set too sensitive. The fact that it doesn’t trigger right away is telling.

But then the first video shows that non-quick-home works okay after the first funky homing, which is harder to make sense of.

As in the first video, does quick-home also work okay on a second homing?

Note that Marlin is responding this way because it sees endstops as triggered. Whether that’s due to an old hit-state bit not getting cleared, due to a floating pin, or due to oversensitive sensorless homing is still up in the air.

thinkyhead, 2018-05-29:
Another thing you can test is to set `HOME_Y_BEFORE_X` and see what effect, if any, that has on the behavior.

smoki3, 2018-05-29:
> As in the first video, does quick-home also work okay on a second homing?
Yes the second homing is fine again. But also this wired move on the Y axis. 
For me it doesn't looks like that the homing sensitivity is related to this because after is stop on the Y axis, the y axis is homing fine anyway

smoki3, 2018-05-29:
I will try tomorrow to increase the sensitivity.
Also will do an test with Hom_Y_Before_X. 

But anyway it's wired that it's working after the first homing

thinkyhead, 2018-05-29:
Let’s get a closer look…

- Enable `DEBUG_LEVELING_FEATURE` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Repeat.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Try this with quick-home first, since it has that “interesting” short XY move.

smoki3, 2018-06-04:
Sorry for my late replay. I don't had time to test it. I will do it this week!

But I found out that when I home my X and Y axis separately (G28 X0 and G28 Y0) than everything is fine.

smoki3, 2018-06-04:
this with 
 `Recv: >>> G28
Recv: Machine Type: Core
Recv: Probe: BLTOUCH
Recv: Probe Offset X:-6 Y:-45 Z:-2.56 (Left-Front & Below Nozzle)
Recv: Auto Bed Leveling: UBL (enabled)
Recv: UBL Adjustment Z raw get_z_correction(-25.00,0.00) = -0.377553 >>>---> -0.377553
Recv: -0.378
Recv:  raw get_z_correction(-25.00,0.00) = -0.377553 >>>---> -0.377553
Recv:   current_position=(-25.00, 0.00, 5.00) : bracket_probe_move
Recv: > endstops.enable(true)
Recv:   current_position=(0.00, 0.00, 5.00) : sync_plan_position
Recv: >>> do_blocking_move_to(-375.00, -285.00, 5.00)
Recv: <<< do_blocking_move_to
Recv: >>> homeaxis(X)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(X, -375.00, 0.00 [50.00])
Recv:   current_position=(0.00, 0.00, 5.00) : sync_plan_position
Recv: <<< do_homing_move(X)
Recv: Move Away:
Recv: >>> do_homing_move(X, 5.00, 0.00 [50.00])
Recv:   current_position=(0.00, 0.00, 5.00) : sync_plan_position
Recv: <<< do_homing_move(X)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(X, -10.00, 25.00 [25.00])
Recv:   current_position=(0.00, 0.00, 5.00) : sync_plan_position
Recv: <<< do_homing_move(X)
Recv: >>> set_axis_is_at_home(X)
Recv: For X axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv:  soft_endstop_min = -25.00
Recv:  soft_endstop_max = 225.00
Recv: > home_offset[X] = 0.00
Recv:   current_position=(-25.00, 0.00, 5.00) :
Recv: <<< set_axis_is_at_home(X)
Recv:   current_position=(-25.00, 0.00, 5.00) : sync_plan_position
Recv:   current_position=(-25.00, 0.00, 5.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(X)
Recv: >>> homeaxis(Y)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(Y, -285.00, 0.00 [50.00])
Recv:   current_position=(-25.00, 0.00, 5.00) : sync_plan_position
Recv:  T:27.84 /0.00 B:27.09 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: <<< do_homing_move(Y)
Recv: Move Away:
Recv: >>> do_homing_move(Y, 5.00, 0.00 [50.00])
Recv:   current_position=(-25.00, 0.00, 5.00) : sync_plan_position
Recv: <<< do_homing_move(Y)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(Y, -10.00, 25.00 [25.00])
Recv:   current_position=(-25.00, 0.00, 5.00) : sync_plan_position
Recv: <<< do_homing_move(Y)
Recv: >>> set_axis_is_at_home(Y)
Recv: For Y axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv:  soft_endstop_min = 0.00
Recv:  soft_endstop_max = 190.00
Recv: > home_offset[Y] = 0.00
Recv:   current_position=(-25.00, 0.00, 5.00) :
Recv: <<< set_axis_is_at_home(Y)
Recv:   current_position=(-25.00, 0.00, 5.00) : sync_plan_position
Recv:   current_position=(-25.00, 0.00, 5.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(Y)
Recv: Z_SAFE_HOMING >>>
Recv:   current_position=(-25.00, 0.00, 5.00) : sync_plan_position
Recv:   destination=(111.00, 140.00, 5.00) : Z_SAFE_HOMING
Recv: >>> do_blocking_move_to(111.00, 140.00, 5.00)
Recv: <<< do_blocking_move_to
Recv:  T:27.81 /0.00 B:27.12 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: >>> homeaxis(Z)
Recv:   current_position=(111.00, 140.00, 5.00) : set_probe_deployed
Recv: deploy: 1
Recv: do_probe_raise(10.00)
Recv: >>> do_blocking_move_to(111.00, 140.00, 12.56)
Recv: <<< do_blocking_move_to
Recv:  T:27.78 /0.00 B:27.06 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: >>> do_blocking_move_to(111.00, 140.00, 12.56)
Recv: <<< do_blocking_move_to
Recv: Home 1 Fast:
Recv: >>> do_homing_move(Z, -300.00, 0.00 [8.00])
Recv: set_bltouch_deployed(1)
Recv:   current_position=(111.00, 140.00, 0.00) : sync_plan_position
Recv:  T:27.72 /0.00 B:27.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:27.69 /0.00 B:26.97 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: set_bltouch_deployed(0)
Recv: <<< do_homing_move(Z)
Recv: Move Away:
Recv: >>> do_homing_move(Z, 5.00, 8.00 [8.00])
Recv:   current_position=(111.00, 140.00, 0.00) : sync_plan_position
Recv:  T:27.78 /0.00 B:26.81 /0.00 @:0 B@:0
Recv: <<< do_homing_move(Z)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(Z, -10.00, 2.00 [2.00])
Recv: set_bltouch_deployed(1)
Recv:   current_position=(111.00, 140.00, 0.00) : sync_plan_position
Recv: echo:busy: processing
Recv:  T:27.81 /0.00 B:27.16 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: set_bltouch_deployed(0)
Recv: <<< do_homing_move(Z)
Recv: >>> set_axis_is_at_home(Z)
Recv: For Z axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv:  soft_endstop_min = 0.00
Recv:  soft_endstop_max = 200.00
Recv: *** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
Recv: > zprobe_zoffset = -2.56
Recv: > home_offset[Z] = 0.00
Recv:   current_position=(111.00, 140.00, 2.56) :
Recv: <<< set_axis_is_at_home(Z)
Recv:   current_position=(111.00, 140.00, 2.56) : sync_plan_position
Recv:   current_position=(111.00, 140.00, 2.56) : > AFTER set_axis_is_at_home
Recv:   current_position=(111.00, 140.00, 2.56) : set_probe_deployed
Recv: deploy: 0
Recv: do_probe_raise(10.00)
Recv: >>> do_blocking_move_to(111.00, 140.00, 12.56)
Recv: <<< do_blocking_move_to
Recv:  T:27.72 /0.00 B:26.97 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: >>> do_blocking_move_to(111.00, 140.00, 12.56)
Recv: <<< do_blocking_move_to
Recv: <<< homeaxis(Z)
Recv: <<< Z_SAFE_HOMING
Recv: >>> do_blocking_move_to(111.00, 140.00, 5.00)
Recv: <<< do_blocking_move_to
Recv:  T:27.75 /0.00 B:27.16 /0.00 @:0 B@:0
Recv:   current_position=(111.00, 140.00, 5.00) : sync_plan_position
Recv:  raw get_z_correction(111.00,140.00) = 0.050654 >>>---> 0.050654
Recv:   current_position=(111.00, 140.00, 4.95) : bracket_probe_move
Recv: X:111.00 Y:140.00 Z:4.95 E:0.00 Count A:40160 B:-4640 Z:8000
Recv: <<< G28`

smoki3, 2018-06-04:
this when I separatly home X and Y axis:
https://pastebin.com/yLSDk4tq

smoki3, 2018-06-08:
@thinkyhead  I tried to figure out I don't find what's going wrong

thinkyhead, 2018-06-13:
We've patched an issue with sensorless homing that might help with this. Please test the latest `bugfix-1.1.x` or `bugfix-2.0.x` branch to see if it's working any better.

smoki3, 2018-06-13:
Yeah it's working now

closed 2018-06-13 by smoki3