MarlinFirmware/Marlin #10891: [FR] K-factor calibration generator w/ fwretract enabled not setting retraction values

agoulah, 2018-05-28:
Hi there,

I was just trying out the firmware retraction feature and was using the k-factor calibration pattern generator on the Marlin website but it doesn't appear to add the gcode for setting the retraction distance to the desired value, and is just relying on the firmware hard-coded values.

Would it be possible to update the generator page so it adds in the M207 and M208 into the gcode produced when firmware retraction is enabled?  Much appreciated.

thinkyhead, 2018-05-29:
This should really be posted at https://github.com/MarlinFirmware/MarlinDocumentation/issues

boelle, 2019-02-20:
@agoulah 

did you post where thinky suggested?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-13 by thinkyhead