MarlinFirmware/Marlin #10894: Feature Discussion: Smart Homing

dot-bob, 2018-05-29:
I run repeated prints and I am finding it redundant to home the printer or auto level at the beginning of a print if the printer has already been homed and been leveled.

So I am thinking adding an option to both G28 and G29.  With G28 before homing to check and see if axis_known_position[] is set, and if so skip homing.  And with G29 check planner.leveling_active to to see if we need to perform bed leveling.

For instance if the option for this type of smart homing was O, You could place the G28 O, and G29 O in your startup script.  On the first run it would home and auto level as usual.  On the second run if the steppers never went inactive it would skip homing and auto leveling and continue with with the print as usual.

I know the commands G28 and G29 are sacred and adding additional options requires much discussion.  At this point what are the options to use for alpha characters?  Or do we use non alpha like ~ | & ^?  Or invent a new G or M command that basically conditionally duplicates the function of G28 and G29?  It would be nice to use the same option for both G28 and G29 as it has similar meaning with both commands.

Roxy-3D, 2018-05-29:
> I run repeated prints and I am finding it redundant to home the printer or auto level at the beginning of a print if the printer has already been homed and been leveled.

Yes.  If your printer is perfect and you are sure you will never lose a step on your motors...  Homing is redundant.

And...   I would be supportive of adding a #define option in Configuration_adv.h to say "Be smart about homing and speed it up".

> So I am thinking adding an option to both G28 and G29. With G28 before homing to check and see if axis_known_position[] is set, and if so skip homing. And with G29 check planner.leveling_active to to see if we need to perform bed leveling.

Please do!   

> I know the commands G28 and G29 are sacred and adding additional options requires much discussion. 

They are not 'sacred'.   But messing with them is problematic.  So let's be very careful!  

> At this point what are the options to use for alpha characters? Or do we use non alpha like ~ | & ^? Or invent a new G or M command that basically conditionally duplicates the function of G28 and G29? 

I'm all for this.   However...  There is going to be lots of resistance.   We need some GCode syntax that lets us break out of the vanilla old school format.   In standard GCode we are not even allowed to use G, M, N or T as a parameter.   That seems pretty lame.  And really, we need full words for parameters.\\

I'm wondering if it makes sense to use a delimiter to flip into a more sophisticated mode.   Usually a space is placed after the GCode command to seperate it from the parameters.   But what if we allowed either a space or a ~ to delimit the command?   Something like:

G1 X100    or    G1~X100   ????

In this case, both are the same.   But if we could agree to recognize both forms, then it is a short hop to allowing something like G1~X_Axis 100    or     G29~Full_Probe

At some point, the GCode needs to  be expanded.  It is too limited.   It is too confusing.   Something has to give.

>  Or invent a new G or M command that basically conditionally duplicates the function of G28 and G29?

No....  Let's see if we can evolve the GCode parsing to make more sense.

thinkyhead, 2018-05-29:
There are valid reasons to use `G28` when the machine "thinks" it knows where it is. For example, after the axis gets knocked out of position. Likewise, you might not like the previous `G29` result and decide to repeat it.

I personally favor keeping things simplified and standardized, so that you can be assured that random printer X at the local library will behave the same as random Chinese printer Q that you order from AliExpress — that they not all have their own variants of common commands with customized behaviors. I prefer to know I can go up to any machine and know that `G28` will re-home every time. I don't want to have to ask the person who configured and flashed the machine what's going on, as they may be in a coma or deceased.

closed 2018-06-01 by dot-bob