MarlinFirmware/Marlin #10895: [Bugfix 2.0] UBL: Issue with Mesh Edit

Bergerac56, 2018-05-29:
Is it me or 16f92dca445ced48219344bfb4a0dece48941dc5 introduced a problem with MESH EDIT ?

Normally, when used, this function (graphic lcd)  shows a visual map on LCD and park the extruder in a corner of the bed (0,0)
Then, using the encoder, we can travel to each point of the mesh and edit it when needed.

After  the introduction of PR 16f92dca445ced48219344bfb4a0dece48941dc5 (in fact the chain of PRs between this one and 59f7861bcb7431fcaac1c85ce3ac6e8d589c19db), the MESH EDIT function does not work anymore. It shows the visual mesh on screen and immediately after disables the motors and stop. No move to (0,0) and no action from the encoder. Screen is frozen. Long press emits a beep but no action.Emergency stop works.

It is still possible to send gcodes via host: for ex: a G28 will home the printer and LCD screen will then come back to Info screen.

Roxy-3D, 2018-05-29:
It would be valuable to try a version of the firmware just before this commit.   If that version works for you, this is a serious problem.    I'm not sure what the best path to find the root cause of the problem is going to be.    That commit is doing very low level interrupt service stuff.

What is your build environment?   What type of processor are you using?

bill-orange, 2018-05-29:
I believe I experienced the same thing.  Also, I can not get a mesh report. G29 S0 does not generate a report.

bill-orange, 2018-05-29:
> This is important information because we want to test the environment that fails.

I am running 1.1.8 on an Anet A8. I compiled with the Arduino IDE. Manual bilinear leveling is turned on. The A8 is using the stock board.

thinkyhead, 2018-05-29:
ABL (Bilinear) uses the command `M420 V` (common to all leveling systems) to generate a mesh report, not `G29 S0`. OTOH If you’re using MESH_BED_LEVELING then `G29 S0` should say at least _something_.

Bergerac56, 2018-05-30:
@Roxy-3D . I am using ATOM with platformio (everything up to date). Build for RE-ARM as I use only RE-ARM for the moment. I could re-mount an AVR if really needed ;)

> It would be valuable to try a version of the firmware just before this commit

The "build" based on commit 59f7861bcb7431fcaac1c85ce3ac6e8d589c19db is working fine. The build based on commit 16f92dca445ced48219344bfb4a0dece48941dc5 does not.

Between those 2 commits, there are 2 commits from Scott and 5 from etagle. They concern the refactor and, in my mind, constitute a module.

I will try to build intermediary versions and see if I can refine the search.

Bergerac56, 2018-05-30:
I refine a bit the search:
Commit 0566badcefda3146fbccebade4a9131cde5e6f20 works OK
Commit e0ca627033504dd7bf8d9b87ce9ec526ee792276 does NOT work
This means that the problem was introduced by the commit a11eb50a3eab6d58d595a67e526fb51190018db3

Unfortunately, I cannot compile with this commit: Marlin\src\module\planner.cpp:105:19: error: conflicting declaration 'uint16_t Planner::cleaning_buffer_counter'

Roxy-3D, 2018-05-30:
Thank You @Bergerac56  !!!

bill-orange, 2018-05-30:
@thinkyhead Yes,  M420 V does work, however, G29 S0 does nothing on the LCD display or terminal.  Something does happen, however, the X Axis stepper starts 'shivering' without noticeably changing position.

thinkyhead, 2018-05-30:
Again, if you’re using `AUTO_BED_LEVELING_BILINEAR`, there is no `G29 S0`.

Bergerac56, 2018-07-09:
@Roxy-3D Hello Roxy. A very discreet commit (603e136e87e3b26177eff8d24d5df776c055fcb6) solved the issue of edition of meshes. THANK YOU Roxy. It makes my day easier.

There is still a small issue with the behavior of the full graphic LCD. The "dot" indicating the position of the head is not consistent with the real position when we turn quickly the encoder to reach a position far from the position we are in.

Tannoo, 2018-10-06:
I believe the "dot" was supposed to represent the place to go to. Not the the current machine location.

closed 2018-11-07 by thinkyhead