MarlinFirmware/Marlin #10899: BLTouch Not Deploying

osdcommand, 2018-05-30:
I'm having an issue with my BLTouch sensor, and getting it to run on Marlin. I have a ramps 1.4 board, with a 550W ATX power supply. I can't get my BLTouch to deploy, be it using the manual commands, or the auto leveling feature. I've set up the firmware as specified in the manual. I've triple checked the connections to the servo and z endstop pins, but that doesn't seem to be the issue. Whenever I try to deploy or store the pin, the LED goes from a dim blue to a slightly darker blue, and won't "trigger", even when I pull the pin out manually. What am I doing wrong?

thinkyhead, 2018-05-30:
Do you have the servo 12V jumper installed on the board?

osdcommand, 2018-05-31:
I'm not entirely sure what that is, so I'd have to assume not. What is that, and what does it do? I'm a bit new to working with auto leveling.

thinkyhead, 2018-05-31:
There's a pair of pins usually, right next to the servo connectors. It requires a jumper to be installed on it or there will be no power supplied to the servo connectors.

osdcommand, 2018-05-31:
I don't see one on my ramps 1.4, and I can't really find any diagrams online for it either. Do you have a picture of yours, by any chance?

kAdonis, 2018-05-31:
this jumper supplies 5V to the servo connectors:
[image]

osdcommand, 2018-05-31:
It works now, you guys are the best. Thank you again!

closed 2018-05-31 by osdcommand