MarlinFirmware/Marlin #10900: Incorrect hotend temperature readings after enabling LIN_ADVANCE.

Virindi-AC, 2018-05-30:
This may be related to https://github.com/MarlinFirmware/Marlin/issues/10735

Using the latest bugfix-1.1.x, if I enable LIN_ADVANCE it seems to affect hotend temperature readings. A test print always produces "cold extrusion prevented" messages during the first couple layers. Sometimes, temperatures in Pronterface appear below 170c. When LIN_ADVANCE is disabled, printing works normally (with identical gcode, and identical build except disabling LIN_ADVANCE).

Note that I am also using CoreXY mechanics as in the above issue.

I have directly attached a scope and multimeter to the thermistor pin on the arduino and those show that there is nothing electrically different between the two builds. When "cold extrusion prevented" appears the actual temperature measured by the thermistor is far in excess of the safety threshold, and further, no noise on the scope is outside of the safe extruding range.

Confs: http://www.virindi.net/junk/confs.zip
Note that I am using a custom board with a Mega 2560. The hotend thermistor is on A13 the same as on RAMPS. The pins definition for my board is included in the zip.

Virindi-AC, 2018-05-31:
Could this be related to the fact that I am using a 20:1 geared extruder, so my E steps/mm is 4111? Perhaps that is messing with LIN_ADVANCE somehow, like it is taking too long stepping?

thinkyhead, 2018-05-31:
It could be. Try reducing your E micro-steps by half and see if it helps.

Sebastianv650, 2018-05-31:
With an e jerk of 5 and a whopping 4111 steps/mm LA can try to advance the pressure at a rate of 20555steps per second, or 20.5kHz. This will most likely not work on an AVR, I'm surprised you only see temperature issues.

Is there a special reason why you are trying to run such a high resolution on your extruder? Up to now, I have rarely seen a printer beating my somewhat above 800steps/mm setting.

Virindi-AC, 2018-05-31:
Thanks. I have reduced the E steps to 1028 (/4), the microstepping to 4 (/4), and the E jerk to 2.0. The results look really good so far, no errors, temperature reading is solid and the print looks nice.

The reason for so many steps is merely that I used a 20:1 gear ratio to be able to utilize a tiny motor in direct drive configuration. I have also been experimenting with small nozzles. I didn't think about the pulse rate, in retrospect it was silly for me to leave the microsteps at the default value. Sorry for the trouble.

Perhaps there should be some kind of warning at compile time, or detection that the system cannot keep up with interrupts?

Sebastianv650, 2018-06-01:
> Perhaps there should be some kind of warning at compile time, or detection that the system cannot keep up with interrupts?

Marlin has so many options to choose from, each increasing or decreasing the performance, that I see no way to give valid limits. This would also change during development, as optimizations are made and new features are added.

VanessaE, 2018-06-10:
Mine's a 5.222:1 geared extruder (driven by an A4988 at 16x microstepping, with a MK8 drive gear, and set for 763 steps/mm), and I also see these hotend temperature dropouts.  I can't say if they're `LA`-related, but they only started recently, they cause "cold extrusion prevented", and have killed one print once because the dropout lasted long enough to confuse the temperature watchdog.  I've done everything in my power to ensure there are no problems with electronics, power, etc., and it's definitely not a bad thermistor either.

It always drops to the same temperature, about 170, same as the OP.  For me, the dropouts are brief enough (a few seconds at a time) that I only get to see them in Pronterface's graph history, where exact temperatures can't be displayed, so 170 is a rough guess.

Wouldn't this suggest a corrupted or mis-referenced variable somewhere, rather than CPU overload?  170 is the default setting for `EXTRUDE_MINTEMP`, after all.

thinkyhead, 2018-06-10:
> It always drops to the same temperature, about 180, similar to the OP.

Loose thermistor?

VanessaE, 2018-06-10:
In my case, definitely not.  I've checked and re-checked, stuffed some aluminum foil in to tighten its contact with the hotend (wires have PTFE insulation, there's no chance of a short), and re-soldered the wires just to be sure.  The entire hotend is tightly wrapped in ceramic cotton and kapton tape.  That, plus it being an axial thermistor leaves no way on earth for it to work loose.  Motherboard connector is tight.  wires are zip-tied at both ends such that neither the thermistor nor the motherboard connector can wiggle appreciably from normal machine motions.

I caught another drop and grabbed a screenshot so that I could look more closely.  I'm wrong about the 180 - it's closer to 195, while the print was going at 220.  Here are two of them from different prints:  

[image]

As you can see, the dropouts are both the same relative amount, some 25 degrees.

thinkyhead, 2018-06-11:
This is probably another side-effect of all the recent _improvements_.

VanessaE, 2018-06-18:
I think it's the thermistor watchdog (or whatever it's called) not getting enough CPU time - just driving retractions too high (even if you're still under the theoretical limit, such as here:  45 mm/s actual, 50 mm/s limit assuming 40000 steps/sec divided by 763 E steps/mm) is enough to cause these little dropouts to appear.

thinkyhead, 2018-06-30:
I agree with your explanation. Previously we'd be more likely to get a watchdog reset, but with recent changes it just starts to encroach on the temperature ISR.

boelle, 2019-02-20:
@Virindi-AC is this still an issue?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

Virindi-AC, 2019-03-16:
I have just been using lower microsteps for higher ratio extruders, and the issue has not recurred.

closed 2019-03-16 by Virindi-AC