MarlinFirmware/Marlin #10903: Shift in VIKI2 Display

plloppii, 2018-05-30:
Bug report
### Description
So I am using the latest version of bugfix 1.1.8 with a VIKI2 display and Azteeg 1.3 Pro board. I did some research and saw that people had this issue in the past. (Issue #9459)
However, it seems like this bug is back. My screen is shifted to the right by 4 pixels.

 I have attached my config, config_adv, and viki display files.
[image]

If anyone has any input on fixing this issue, I'd really appreciate it! 

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2054954/Marlin.zip)

thinkyhead, 2018-05-30:
For the previous issues relating to this problem, how was it resolved in those cases?

plloppii, 2018-05-31:
So I am quoting Iamturbo1978, and he said 
"I think I may have explained what I did wrong. The column address is split into (2) 4-bit nibbles, a high nibble, and a low nibble. I first changed the high nibble from x10 to x14, and my display did not like that. I changed it back, and then I changed the low nibble from x00 to x04. That is what corrected my display. The display would see the value of x14, but as x10 for the high, and x04 as the low. Hope that make sense."
(Regards to ultralcd_st7565_u8glib_VIKI.h)

So I found where the column address is edited:
`     ST7565_WRITE_BYTE(0x010);         /* set upper 4 bit of the col adr to 0x10 */
      ST7565_WRITE_BYTE(0x004);         /* set lower 4 bit of the col adr to 0x00. Changed for DisplayTech 64128N */
                                        /* end of sequence */`
Since it already has 0x10 for the high byte, and 0x04 for the low, I was thinking maybe putting in 0x08 for the low byte to maybe shift the display over 4 pixels. 

Ill test it out tomorrow!

Thanks
Noah

plloppii, 2018-05-31:
hmmm, didnt work as I hoped to. 
@Iamturbo1978 have you encountered this issue again? 

Any other suggestions how I could go about fixing this bug?

Bob-the-Kuhn, 2018-06-01:
You were partially correct.  The lower column address bits should be zero.

The other thing that needs to be done is to go to **Conditionals_lcd.h** and enable line 83.  You can delete 82 if you want.   Because that define wasn't active the file we were all playing with wasn't being used.

plloppii, 2018-06-01:
Haha well, that's great to know. Thank you @Bob-the-Kuhn 
Thing is, the display is rotated. I tried commenting out #define LCD_SCREEN_ROT_180 in ramps.h but that didn't work. Any ideas on how to flip it?

plloppii, 2018-06-01:
Fixed the Issue! So for future reference, I adjusted the ADC Select and Common Output to flip the display horizontally and vertically. Then if your display is shifted again, you need to adjust the upper and lower col adr.

closed 2018-06-01 by plloppii

thinkyhead, 2018-06-02:
I note some small differences between ST7565 and ST7565R, such as the voltage resistor ratio referring to V5 on ST7565 and V0 on ST7565R. Which one of these interfaces are we targeting for the VIKI? I assume it's actually the ST7565R.

In either case, this appears to be incorrect:
```cpp
ST7565_WRITE_BYTE(0x10);          // Set V0 Voltage Resistor ratio. Setting for controlling brightness of Displaytech 64128N
```
The correct command to set the resistor ratio is `0x20 + ratio`. The `0x10` command would simply set the high nybble of the column address to 0 … unless its meaning is changed in the init sequence.