MarlinFirmware/Marlin #10904: 25kHz pwm frequency for 4pin fan control - not an issue!

italocjs, 2018-05-30:
Is it possible to assign a faster speed for a pwm pin on the marlin code? 

I want to control two 4-pins fans, with internal speed control through PWM (those fans used above processors), i've looked at the datasheet and the switching frequency needs to be between 21khz and 28khz, the current drawn is only about 2ma at 5v, so it could be directly connected to the analog output

The idea is to replace the noisy fan on the power supply, and add a second fan to cool down the ramps and drivers,  i've bough these fans because they are extremely silent (and today, the thing making more noise on this printer are the fans), 

I'm planning to sketch a code where the fan would be disabled if no heater or motor are on, run at 30% if the nozzle is on, and between 70-100% if the target temp.

thinkyhead, 2018-05-30:
It depends on which timer is used, and the 12V PWM outputs must use a particular timer. If that timer is used for other purposes, as I believe it is, then we must keep it at 1kHz. To get a different rate, you would need to connect a 5V pin, such as 44, to an external MOSFET and use Timer 5.

ejtagle, 2018-05-31:
Why not using one of these? https://www.ebay.com/itm/12V-PWM-PC-CPU-Fan-Temperature-Control-Speed-Controller-Module-High-Temp-Alarm/112174066548?hash=item1a1e186b74:g:TZ4AAOSwrslafAFF .. At 2.5us$ it is probably cheaper that buying a single mosfet. You can place the NTC in the power supply heatsink and get proper temperature regulation ...

italocjs, 2018-06-10:
Update, i've chosen to use the digital output from the arduino mega, i'm turning it on by gcodes for now, when the pin is off the speed is about 25% of the max, and with it on, goes to 100%.  
So far i've been monitoring the temperatures at the psu transistors and they are not getting over 40C with 25% fan, so i'll probably just keep it slow by hardware and not mess with the firmware.   

I'm using one from intel (old pc fan), it is incredibly silent,  now i'm only getting about 40-42 db coming out of the 3dprinter

italocjs, 2018-06-10:
@ejtagle, Thanks, ive ordered two of those, i didnt even knew these existed, its really useful and i'll use as proposed, with the ntc at the heatsink of the PSU and the second on the bed mosfet heatsink

closed 2018-06-10 by italocjs