MarlinFirmware/Marlin #10918: Thermistor not working, any solutions? (pins.h marks pin -1)

Ivomola, 2018-06-02:
Hi. I have A Tevo Tarantula, and while I was changing my nozzle, the thermistor stopped working and marked a constant 90°, so the printer gave me an error, I checked it and it was broken in half. Literally. So I changed it for another I had, exactly the same as the other and it marked the same temperature, 90° constant. I tried plugging it in the bed thermistor plug and it worked, and when I held it in my hands it raised temperature, so as I thought it was the connector that was broken, I searched if I could swap the thermistors pins and use the 2 extruder ones instead of the others, but people always say "change pins.h" the only problem is that in my firmware (I'm using MarlinTarantula-EasyConfig-1.1.x) it says "#define TEMP_1_PIN -1" and "#define TEMP_2_PIN -1", and so on with the mayority of the options there, so how can I change it? The printer worked fine until 30 minutes ago when this happened

Ivomola, 2018-06-02:
Thanks for your response. My board is the MKS Gen 1.3, and I alredy tried before hand in the pins_MKS_gen_13.h and nothing related to the thermistor seems to appear

Bob-the-Kuhn, 2018-06-02:
That's because **pins_MKS_gen_13.h** pulls in **pins_RAMPS.h** where most of the pin assignments are done.

Attached is a modified **pins_RAMPS.h**.  It has the extruder thermocouple inputs swapped.

[pins_RAMPS.zip](https://github.com/MarlinFirmware/Marlin/files/2065366/pins_RAMPS.zip)

Ivomola, 2018-06-02:
IT WORKS! Thank you very much, I didn't know the pins file of the mks was referenced to the ramps file, but now that I tkink about it it makes sense because it's easier to program 1 main file and 3 smaller ones than 4 that are exactly the same but with little variations. But again, thanks!

Bob-the-Kuhn, 2018-06-02:
Glad to be of help.

closed 2018-06-02 by Bob-the-Kuhn

VerrroS, 2020-04-01:
Hey, I had the same problem with my Prusa i3 pro w. In Marlin 2.0x bug fix I changed my pins_GT2560_REV_A.h file. But now my hotend showes +60C more than my hotbed.. I have no Idea what could be the problen, since I already tried out the thermometer and it is working normally..
Any ideas?