MarlinFirmware/Marlin #10926: Einsy Rambo MMU Question

Sloppsta, 2018-06-03:
Hey guys, Sorry to bug you, but I have the Einsy Rambo and just preordered the MMU 2.0 upgrade to go on a hypercube style printer I am building. My question is this.
Looking at pins_einsy.h i see that E_MUX2_PIN is pin 78, but looking at the einsy rambos schematics, pin 78 is not connected.
[image]

Am I just looking at this wrong?

Thanks

// Průša i3 MK2 Multiplexer Support
//
#define E_MUX0_PIN         17
#define E_MUX1_PIN         16
#define E_MUX2_PIN         78   // 84 in MK2 Firmware, with BEEPER as 78

3d-gussner, 2018-06-06:
@Sloppsta Made a pull request https://github.com/prusa3d/Prusa-Firmware/pull/182 in the past for the Prusa MK2 branch this and other PRs never made it into the MK3 branch.
The E_MUX2_PIN stayed from the early MMUv1 super switch stage.
As the MMUv2 has a own ATmega2560 controller board with 3x TMC2130 drivers i expect that it will be serial communication between the EINSY (P3 pin 3/5 TX2/RX2) and MMUv2 board. With 5 filaments in MMUv2 it will be hard to squeeze that in 2-bit (MMUv1) ...maybe they are doing something else ... we will see.
Hope that helps

amoose136, 2018-06-07:
I have one on the way that should arrive sometime late this month or next and will be in the same boat as far as getting it working with Marlin goes.

Sloppsta, 2018-06-08:
I will probably not recieve mine until October.  Maybe someone smarter than myself can help bring the new mmu into marlin!

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-13 by thinkyhead