MarlinFirmware/Marlin #10933: Z axe issue when printing close to Xmax.

FabioMC, 2018-06-04:
I recently update my 3D printing to 1.1.8 version of marlin firmware and since that I am facing a very strange issue related to Z axe. Every time that I try to print a big element or a group of itens and those parts/elements is close to the Xmax, my printer starts to increase Z stoping the movements of X and Y axes. The printer continues increasing Z until achieves the bottom part of the printer, but don't stop the motor, forcing the bed against printer structure. I need to kill the printer to stop it completely. It only happens on X max. Never happen on any other region of the printer area. The funny thing is that appears to be always on to the same spot independent the piece I am tying to print.

I thought that could be a problem in the gcode file but I checked it and nothing suspect is in it. My guess is there are something wrong with my configuration files or a very specific bug in a very strict situation.

I appreciate any hint or help in trying to fix such behavior. Attached are my configuration files.

Thanks

Fábio
[Configuration Files.zip](https://github.com/MarlinFirmware/Marlin/files/2069377/Configuration.Files.zip)

GMagician, 2018-06-07:
try to use bugfix branch, if you are lucky maybe has been already solved

boelle, 2019-02-20:
@FabioMC did you try bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-03-17 by p3p