MarlinFirmware/Marlin #10935: Corexy Auto bed leveling reversed

TheLukey21, 2018-06-04:
So i have an issue with the auto bed leveling feature were it seems to compensate in the wrong direction, this isn't the first time i have had this issue and i thought it was solved when i changed to a new firmware version, im now on 1.1.8 i was using bilinear ABL but i am now using linear and its no different, its homes in the back left corner (both min) and prints in the same orientation as repetier host displays and it also moves in the right direction when testing using pronterface etc.

Please see pictures attached.
[image]
[image]
[image]

Roxy-3D, 2018-06-05:
Where ever the home position is...   That is (X,Y) = (0,0)  right?    If it is....  Do you have a 'Left Handed' coordinate system?   Most of the time when this type of problem comes up, the user has configured a 'Right Handed' coordinate system.     When you look at your homed position....   Does it look like:

mathworld.wolfram.com/Left-HandedCoordinateSystem.html

If it doesn't, we need to flip either your X or Y to move in the opposite direction.

TheLukey21, 2018-06-05:
It's right handed i would say, when i home it goes to the same point as shown in repetier (the back left corner) and then when i go positive on the y axis it goes forwards,positive on the x axis moves it right and positive on the z goes up. i can't see a reason why it would be inverted.

thinkyhead, 2018-06-08:
Viewed from the top, the nozzle orientation should be one of these…
```
Y positive
^
|
|
|
|
|
0,0-------> X positive
```
…or…
```
X positive <-------0,0
                    |
                    |
                    |
                    |
                    |
                    v
               Y positive
```
…or you can set it up to home X or Y to MAX, then you get something like…
```
0,200-------> X positive
|
|
|
|
|
v
Y negative
```
```
                Y positive
                     ^
                     |
                     |
                     |
                     |
                     |
X negative <-------200,0
```
```
X negative <-------200,200
                      |
                      |
                      |
                      |
                      |
                      v
                 Y negative
```

TheLukey21, 2018-06-09:
Its the last one out of those, it seems to compensate correctly in the x axis now after changing the homing to the max for both axis and reversing both motors so it still homes in the same corner. But i noticed that my piece of aluminium is slightly warped in the centre and when the glass is clamped onto it there is a high spot in the middle and maybe i have my probe set to coordinates that miss the high spot so i am going to change to UBL bed leveling and make sure it probes the high spot and see how it goes.

TheLukey21, 2018-06-09:
Can't quite figure UBL out so im using bilinear and it still doesn't compensate correctly even though it moves as it should, i can't figure out why.

thinkyhead, 2018-06-10:
What does your mesh (`M420 V`) look like?

TheLukey21, 2018-06-10:
Bilinear Leveling Grid:
      0      1      2      3      4      5      6      7
 0 +0.319 +0.195 +0.123 +0.062 +0.034 +0.031 +0.054 +0.105
 1 +0.280 +0.186 +0.121 +0.054 +0.032 -0.012 +0.004 +0.041
 2 +0.308 +0.190 +0.105 +0.039 -0.037 -0.059 -0.089 -0.056
 3 +0.269 +0.145 +0.086 -0.025 -0.068 -0.101 -0.123 -0.132
 4 +0.226 +0.128 +0.048 -0.036 -0.098 -0.173 -0.217 -0.218
 5 +0.130 +0.023 -0.042 -0.165 -0.221 -0.279 -0.318 -0.346
 6 +0.044 -0.060 -0.141 -0.222 -0.279 -0.358 -0.422 -0.462
 7 -0.104 -0.176 -0.239 -0.353 -0.414 -0.489 -0.538 -0.601
Subdivided with CATMULL ROM Leveling Grid:
        0        1        2        3        4        5        6        7        8        9       10       11       12       13       14
 0 +0.31889 +0.25371 +0.19500 +0.15495 +0.12278 +0.08958 +0.06194 +0.04448 +0.03417 +0.02917 +0.03056 +0.03880 +0.05389 +0.07771 +0.10500
 1 +0.29509 +0.23979 +0.18941 +0.15363 +0.12302 +0.08806 +0.05872 +0.04567 +0.03720 +0.02070 +0.00960 +0.01601 +0.03137 +0.05196 +0.07531
 2 +0.27972 +0.23076 +0.18556 +0.15177 +0.12139 +0.08531 +0.05445 +0.04175 +0.03194 +0.00762 -0.01194 -0.00927 +0.00361 +0.02111 +0.04139
 3 +0.29387 +0.24018 +0.19000 +0.14966 +0.11427 +0.08057 +0.05002 +0.02292 -0.00052 -0.02071 -0.03557 -0.04403 -0.04349 -0.02785 -0.00661
 4 +0.30806 +0.24696 +0.19000 +0.14427 +0.10500 +0.07120 +0.03861 -0.00177 -0.03667 -0.05078 -0.05917 -0.07733 -0.08861 -0.07623 -0.05611
 5 +0.29307 +0.22770 +0.16887 +0.13088 +0.09710 +0.05123 +0.00663 -0.02830 -0.05491 -0.06899 -0.07875 -0.09440 -0.10569 -0.10252 -0.09458
 6 +0.26917 +0.20299 +0.14500 +0.11488 +0.08639 +0.02990 -0.02472 -0.05141 -0.06833 -0.08623 -0.10139 -0.11375 -0.12305 -0.12818 -0.13167
 7 +0.25123 +0.19292 +0.13995 +0.10481 +0.07141 +0.02164 -0.02602 -0.05447 -0.07767 -0.10663 -0.13325 -0.15283 -0.16614 -0.17057 -0.17156

thinkyhead, 2018-06-10:
There's a lot of deviation in the mesh, making the bed look more warped than it may actually be. You might want to try setting `MULTIPLE_PROBING` to 3 or 4 and see if the mesh looks any better. Of course, also be sure to test with `bugfix-1.1.x` or `bugfix-2.0.x` so we're all on the same page.

TheLukey21, 2018-06-13:
changing the probing amount didn't seem to change much if anything, its like i just need to swap the X axis  direction but keep the home position the same, it isn't possible to swap just one axis without affecting the other on a corexy is it? i'll try the latest bugfix next or maybe an earlier stable release to see if it does the same thing.

thinkyhead, 2018-06-13:
For best results getting help with configuration and troubleshooting, we recommend the following resources:
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)

boelle, 2019-02-20:
@thinkyhead think we can close this one

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead

TAOGde, 2019-05-23:
I have the same problem with inverse UBL Mesh.

My cordinate are 0,0 left front. The bed is in real world on the Y axis front lower than on the back. in my UBL mesh the graph show the front higher than the back.
how can I switch the compensation to make the front also lower in the mesh?