MarlinFirmware/Marlin #10937: FR: Please re-enable babystepping when NOT printing

joris57, 2018-06-05:
In the past, babystepping was enabled also when not printing. In combination with 'BABYSTEP_ZPROBE_OFFSET', this allowed to easily finetune the Z-probe offset (M851) with a shim from the display. I can't immediately think of workarounds.

insanity67, 2018-06-05:
I do my calibration on first layer skirt most of the time, remember that value and adjust z-offset + save eeprom.

Roxy-3D, 2018-06-05:
I think there are very legitimate reasons for things being the way they are right now with regard to BabyStepping.    

But...   we should have a discussion about all of the concerns.     One solution might be a configuration option to make  BabyStepping work even if the machine is not homed, and the printer is not printing.  If we have that discussion here, I suspect we are all going to come to the conclusion that if we included an option to make that happen....   It should be off by default.

It is important for people to voice their thoughts and concerns on this if they have strong feelings one way or the other.

forkoz, 2018-06-06:
I too liked tweaking the Z offset when not printing or right before. Afaik the head wasn't supposed to move unless you were printing. When I saved with M500, the new z-offset would be saved. But if you add features to bugfix 2.0 only then I can't use it since its 32bit only.

ManuelMcLure, 2018-06-06:
> But if you add features to bugfix 2.0 only then I can't use it since its 32bit only.

Not true. bugfix-2.0 _supports_ 32 bit boards, but should also work on any 8-bit board supported by 1.1.x.

forkoz, 2018-06-06:
If I had known that I would have tried it already.  It was always presented as marlin rewritten for 32 bit boards. Also, what is the reason for curves in 1.1?

Roxy-3D, 2018-06-06:
The history of bugfix-2.0.0 is we didn't want to destabilize bugfix-1.1.x with 32-bit activity.   The work to get things working on 32-bit platforms (especially combined with the hierarchical file layout) was expected to cause instability.     We didn't want to destabilize the existing, 8-bit AVR based users.

But bugfix_2.0.0 is far enough along, any bug in it (for 8-bit AVR users) is probably also in the bugfix_1.1.x branch.   The two branches are pretty much in lock step and if you are on an 8-bit platform, it shouldn't matter which branch you use.

My question about whether there really was a need to support it in both branches is just because we will never be able to freeze bugfix_1.1.x if we keep adding features to it.    So...   I was hoping everybody could agree it doesn't need to be both places.

thinkyhead, 2018-06-07:
> In the past, babystepping was enabled also when not printing.

No, it never was. It was accidentally possible to go into the babystepping screen with a double-click outside of a print-job, but then babystepping didn't do anything. So that was fixed.

> make BabyStepping fully active regardless of the machine's state

I see no reason why it couldn't be, except that we would need to add code to ensure the steppers are turned on. Babystepping was originally designed to be used during the first layer, at the beginning of a print-job, so you could see that the first layer was improving. Without anything printing, babystepping would move the print head (with no limits) without any feedback in terms of its effect on the first layer.

> It was always presented as marlin rewritten for 32 bit boards

We've tried to be pretty clear that it's universal in all our discussions and documentation. But some users people have been confused about that and have spread the misconception.

ManuelMcLure, 2018-06-08:
> We've tried to be pretty clear that it's universal in all our discussions and documentation. But some users people have been confused about that and have spread the misconception.

I think the biggest confusion comes from the download page which says
> Marlin 2.0 (32-bit) development (bugfix-2.0.x.zip — Bleeding-edge code! Caution!)

The main github page for the bugfix-2.0.x branch does specify that it's for all boards but I think people may never get there because of the download page description.

Roxy-3D, 2018-06-08:
That will be removed soon enough.   But for right now...   People that need stability probably are best off using bugfix_1.1.x.   When bugfix_1.1.x gets frozen...    that warning definitely will be removed.

thinkyhead, 2018-06-08:
Well, yes, if you have a 32-bit board then you MUST use 2.0.x…

But we'll just change that to:
```
Marlin 2.0 (32-bit ... but not ONLY 32-bit) development (bugfix-2.0.x.zip — Bleeding-edge code! Caution!)
```

Roxy-3D, 2018-06-08:
I'm of the opinion that bugfix_2.0.0 isn't really "bleeding edge" edge any more.   There still is some pain in using for 32-bit platforms.  But it is pretty solid.    

My vote is we soften the "bleeding edge" term a little bit to describe it.

forkoz, 2018-06-09:
Your "bleeding edge" is bloody good. I've seen real bleeding edge that just keels over. The only danger here is when there are new commits and a bug isn't caught for a few hours. So I always keep the last download to roll back to when moving the configs.

My only use for non-printing baby stepping was: I think I need to move the next print first layer up or down a bit, I don't like how the last one turned out. Lets do it while the extruder is heating or after this print ends.

boelle, 2019-02-20:
@thinkyhead maybe add label to this one?

boelle, 2019-06-22:
missing label

InsanityAutomation, 2019-06-22:
This is completed with the BABYSTEPPING_ALWAYS_AVAILABLE and tweaks completed in that area by both Roxy and Myself. This can be closed.

closed 2019-06-22 by Roxy-3D