MarlinFirmware/Marlin #10938: Hard Reset on long extrusions when commanded by a Gcode file verses manual request.

cobramike41, 2018-06-05:
### Description

I updated the advanced configeration tab to allow for longer extrusions due to my color switching bowden setup having to pull/push 500mm of filament to change colors. I can manually extrude 500 mm with no issues using octoprint and repeater. EX: M83, G1 E500 F3000. However when printing if the GCODE tries to do this EX: G1 X15 E490 F1562, the printer hard resets and the print is lost.

### Steps to Reproduce

1. Using Slic3R 1.4 Beta PE, slice the model with the Pruge tower on using at least 2 colors.
2. Send to OCTOPRINT
3. Print

**Expected behavior:**
Printer follows the below commands
G1 X# E20 F1256
G1 X# E490 F1563
G1 X# E20 F1256
G1 X# E10 F200

The feed rates are variable based on whatever math Slic3r uses to do a purge.

**Actual behavior:** [What actually happens]
Printer Runs
G1 X# E20 F1256
Printer starts
G1 X# E490 F1563
then hard resets at about 60% of that extrusion.

#### Additional Information

I can get around this by post-processing the GCode to replace G1 X# E490 F1563 with the following commands G1 X# E100 F3000, G1 E100 F3000, G1 E100 F3000, G1 E100 F3000, G1 E190 F3000.

This solution was confermed on a 2-color print which switched colors aproximatly 15 times, however I would like to not have to post process.

thinkyhead, 2018-06-08:
> the printer hard resets and the print is lost

This is usually caused by too much load on the processor. Keep reducing your maximum feedrate for the E axis until you no longer get a reboot. If you're using 32x micro-stepping for E, reduce it to 16x.

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-13 by thinkyhead