MarlinFirmware/Marlin #10939: [FR, HAL, 2.0.x] Need generic "spiSend" for arbitrarily sized buffer in HAL

marcio-ao, 2018-06-05:
I notice that in "src/HAL/HAL_SPI.h" there is the following function:

```
/** Read from SPI into buffer */
void spiRead(uint8_t* buf, uint16_t nbyte);
```

However, there is no corresponding "spiRead" that reads an arbitrary number of bytes into a buffer. There is a "spiSendBlock", but it always tacks on a token byte and uses a 512 block size. I understand `spiSendBlock` is tailored to the SDCard stuff, but I am implementing drivers for SPI based displays and it would be nice to have more general functions in the HAL. 

Writing one byte at a time using "spiSend" works, but is slow. An ideal function would be something which compliments "spiRead", like this:

```
/** Write from buffer to SPI */
void spiWrite(const uint8_t* buf, uint16_t nbyte);
```

Thank you,

-- Marcio

ejtagle, 2018-06-05:
i completely agree with this

thinkyhead, 2018-06-08:
If you find gaps in Marlin while paying attention to certain regions of the code, please feel free to contribute whatever is needed to fill them in. The SPI code for the HALs is all very much in need of improvement and better unification.

marcio-ao, 2018-06-08:
@thinkyhead: I fiddled around a bit trying to add the function, but was unsuccessful. I don't understand enough about the architecture of the HAL to be able to really enhance it yet.

> The same is true for servos, serial, etc.

Interestingly, I am using Marlin 2.0 with servos. It was my understanding that the HAL for it had not been written yet, but strangely enough when I enabled servos they work. We are using it for one of our prototype toolheads.

thinkyhead, 2018-06-09:
Good to know servos are working for you. The code was added at some point, and then patched up a bit recently, but it needs even more cleanup before I'd consider it "perfect."

boelle, 2019-02-20:
is this one still relavant?

boelle, 2019-06-22:
missing a label

sjasonsmith, 2020-11-10:
@rhapsodyv is this supported in your new SPI classes? I understand they aren't available for all platforms yet, but they could provide a path forward.