MarlinFirmware/Marlin #10941: [FR] Developer Tools - EditorConfig

Squid116, 2018-06-06:
I have seen quite a few PR's lately where there are simple things like tabs and spaces and brackets being bought up to the [coding standards](http://marlinfw.org/docs/development/coding_standards.html).

I'd suggest that most people are editing using a tool that either natively or via a plugin supports [editor config](https://editorconfig.org/).

Editors which support it (and I am aware of being used in Marlin Dev):

- Vim
- Emacs
- Sublime
- Atom
- VS Code
- Notepad ++
- Github

The one noteable exception is the Arduino IDE.

I don't think it is a great use of the core developers time to be applying coding standards, and in my experience this is one very effective way to limit the amount of time spent doing that.

I'm happy to create the base .editorconfig file based on what is documented in the coding standards and raise a PR, just wanted to check if others agree that this is valuable before proceeding.

cc @thinkyhead @Roxy-3D

ejtagle, 2018-06-06:
I am guilty of using Notepad++ to do patches and software development... ⭕️ I don´t like platformio, as my inet connection is not reliable, and i simply can´t depend on it to be always available.
That leaves Arduino, or VS Code, or something based on Eclipse (for most of my other embedded projects, it is Eclipse...)

Roxy-3D, 2018-06-06:
> I don't think it is a great use of the core developers time to be applying coding standards, and in my experience this is one very effective way to limit the amount of time spent doing that.

I agree.   My thinking is it would be very easy to add a small C program that could kill trailing spaces and get the indentation exactly correct.    Any time something is checked in, it would clean it up.     And if that could be figured out, the next step would be to get it have the proper spacing on expressions.  Especially within if( ) statements.

I suspect this can be added to the github check in procedure.   But I don't know how to do it.

Squid116, 2018-06-06:
@ejtagle this is just a nicety to have - if you edit in an unsupported editor/ide, it will just ignore the config.

Squid116, 2018-06-06:
@Roxy-3D - why re-invent the wheel, there are plenty of linters out there that will cover these of that could be included in the build process.

Squid116, 2018-06-06:
C++ is not my area of expertise, but I'l do some research on automatic linters with GH integration.

thinkyhead, 2018-06-08:
Just DO NOT remove any trailing spaces from the `.patch` file in `buildroot/share/fonts`.

alexxy, 2018-06-08:
There is a greate tool called **uncrustify** that can apply coding standarts (and its already used by some projects for years). It also can check chengeset for coding style

Roxy-3D, 2018-06-08:
My belief is that an automatic tool to apply the coding standards would be great.  But it has to also allow exceptions too.    Ideally, it could be inserted into the build process such that it is transparent to any developer that generates a Pull Request.      But I suspect that is hard to do because we don't even know what operating system the developer is using.  

It might be enough if we could just get it to format the code appropriately on the user's GitHub account.    That way, when they check in the code, they at least see the way it will ultimately appear.

alexxy, 2018-06-09:
For projects like GROMACS it's done by post-recive git hook (it checks if format is good, and accept if it is)

Squid116, 2018-06-10:
I'm not a fan of changing the code someone has commited, to me it is about giving people the tools to get it right rather than trying to fix it for them.

boelle, 2019-06-22:
missing a label

closed 2019-07-11 by thinkyhead