MarlinFirmware/Marlin #10942: [Marlin-bugfix-2.0.x] Maybe a bug? Re-bump not working on Z

boelle, 2018-06-06:
As per subject re-bump works on X and Y when homing but not on Z

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/2074361/Configurations.zip)

is there something in the config that prevents it?

ejtagle, 2018-06-06:
Try enabling ENDSTOP_NOISE_FILTER and see if that makes a difference...

boelle, 2018-06-06:
that did not help

but my probe has disconnected at the plug

the strange thing is that i dont use the probe for homing but rather i have dual z drivers and dual z endstops

should it not disregard the probe during homing?

ejtagle, 2018-06-06:
During homing the probe should be ignored...

boelle, 2018-06-06:
it is not

if i have the probe connected and i home i get the rebump

if i remvoe the probe i dont get it

and as you can see in my config i use max endstops and when i home the Z goes up

i does not make sense to me

GMagician, 2018-06-06:
I also have dual Z, dual switches and bltouch and it works for me, but I have `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
` disabled

GMagician, 2018-06-06:
Yes but I think that when this is enabled probe is used to home (I have to check code)

boelle, 2018-06-06:
i can home my machine no problem

its the rebump that is missing if the probe is not connected

remember my machine is set to use the max endstop and not the probe

rebumb is where it touches the endstop and then move away and then goes towards the endstop again at slow speed.

GMagician, 2018-06-06:
> y to y min

USE_YMIN_PLUG is disabled

boelle, 2018-06-06:
oh yes.... forgot to tell that i have swapped pin numbers in the ramps pin file

but to make it clear x and y are working 100% as they should

GMagician, 2018-06-06:
you also have:
#define Z2_USE_ENDSTOP _XMIN_

boelle, 2018-06-06:
yes... as i said i swapped some pin numbers in the pin file

will attach that one, it will make more sense. But i did not want to change the plugs on the board so i just changed the pin file

GMagician, 2018-06-06:
ok I wrote message before explorer refresh

boelle, 2018-06-06:
[pins_RAMPS .zip](https://github.com/MarlinFirmware/Marlin/files/2078111/pins_RAMPS.zip)

rename file to pins_RAMPS.h

ITS not a zip file

boelle, 2018-06-06:
reason why i'm lazy is that all my wires are secured with zip ties and i have to waste about 5-10 zip ties... so it was more easy to change arroun pin numbers

GMagician, 2018-06-06:
well I have to check code to go further..

boelle, 2018-06-06:
yeah, its a funny one

everything works 100% as it should if z probe is connected. what is interesting is that it should not care about the probe

UBL also works fine.... 

its only when i disconnect the probe it does not home Z correctly...

GMagician, 2018-06-07:
Sure yourt config is updated?
`#define Z2_USE_ENDSTOP XMIN` should be `#define Z2_USE_ENDSTOP _XMIN_`

boelle, 2018-06-07:
let me double check... but it was prob one of those that might have slipped past my nose when i updated the config

boelle, 2018-06-07:
nope its correct as you would have seen in the configs attached to first post

```
#define Z_DUAL_STEPPER_DRIVERS
#if ENABLED(Z_DUAL_STEPPER_DRIVERS)
  #define Z_DUAL_ENDSTOPS
  #if ENABLED(Z_DUAL_ENDSTOPS)
    #define Z2_USE_ENDSTOP _XMIN_
    #define Z_DUAL_ENDSTOPS_ADJUSTMENT  0
  #endif
#endif
```

boelle, 2018-06-07:
hmm... will check the attached config to make sure you got the right data

GMagician, 2018-06-07:
ok I think I understood. I opened it in a linux mint SO under virtualbox. It display underscores as space...

GMagician, 2018-06-07:
In code I see something like:
`#define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 && ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN))`

I really think that this define will do something "not wanted" when defined. I think you can try to remark it and define Z_MIN_PROBE

boelle, 2018-06-07:
yeah, i think you found it

to me is looks strange that just because i have a probe and z home dir  is some value and  the probe uses z min endstop pin then homing with probe is enabled autoamtic

but when i do a home is uses the microswitch endstops

but it kind of halfway make sense why it will not rebump

not sure if this is a bug.... but its not correct either.

boelle, 2018-06-07:
forgot... will test this tomorrow, had visitors today and have to take printer off and store it

boelle, 2018-06-07:
> The recommended course right now is to go back to an earlier version of Marlin.

I mange by keeping sure the Z probe does not disconnect. 

There is only one plug not in use, and remember i did swap the pin numbers in the pins file so i would not have to waste 10 zip ties, i did attach the pins file a few post back. Remember its not a zip file but i just renamed the extension so i could attach it

thinkyhead, 2018-06-07:
> i dont have dual axis but could it still have some influence?

If you're not using dual _**endstops**_ (correction) then disable this:
```cpp
  #define Z_DUAL_ENDSTOPS
```

boelle, 2018-06-07:
i meant i dont have dual x axis or dual y axis, but i do have dual Z endstops

boelle, 2018-06-07:
and i just re-read your post and see that you wrote dual endstops..... me getting tired here :-(

thinkyhead, 2018-06-07:
Right, and you're also using dual Z steppers (aka "dual Z axis"), so each one is connected to a separate stepper driver, Z and Z2. So if someone asks in future if you're using a dual axis, the answer is "yes."

boelle, 2018-06-11:
forgot this one, is it a bug when it does not do rebump because i disconnect the probe. which i think it should ignore since the probe is not used for homing

or should it be closed? i was just thinking about labels

thinkyhead, 2018-06-12:
> is it a bug when it does not do rebump because i disconnect the probe

It might be, but you should use `M119` to see if connecting and disconnecting the probe has any effect on endstop readings.

boelle, 2018-06-12:
output from octoprint terminal

before disconnect:
```
Send: N901 M119*34
Recv: Reporting endstop status
Recv: x_max: open
Recv: y_max: open
Recv: z_min: open
Recv: z_max: open
Recv: z2_max: open
Recv: filament: TRIGGERED
Recv: ok
```

after disconnect:
```
Send: N902 M119*33
Recv: Reporting endstop status
Recv: x_max: open
Recv: y_max: open
Recv: z_min: TRIGGERED
Recv: z_max: open
Recv: z2_max: open
Recv: filament: TRIGGERED
Recv: ok
```

zmin is the probe and that is the only one i would expect to change status

thinkyhead, 2018-06-13:
We'll keep tweaking the endstop code, and hopefully eliminate all unwanted false-positives.

boelle, 2018-07-23:
closing this one as it seems its not regarded as a bug

closed 2018-07-23 by boelle