MarlinFirmware/Marlin #10948: [FR] Documentation on Marlin 2.0 for Developers

kcheeeung, 2018-06-06:
Hello everyone once again!

I would like to request documentation for the overall development and logic on how the Marlin was written. The primary purpose for asking for this is so that others, like me, could gain a better and deeper understanding of the software engineering behind the Marlin firmware. I believe having such documentation would be beneficial in a general sense because it will attract new developers to easily and quickly understand the code to make major contributions rather than needing to spend a lot of time trying to decode the logic of how it was written, as Marlin has grown very large. As I see it now, only those that know Marlin very well can develop Marlin properly.

Similarly, the folder structure hierarchy must carry some meaning for the Marlin firmware in terms as a whole. Previously with Marlin 1.x, it feels like the files were littered everywhere in a single folder and retracing the organization was difficult, thus leading to Marlin 2.0 with a new folder system. In Sublime Text, I use "Ctrl + P" (Quick-open files by name) to open the necessary files so the folder structures hierarchy incurs not much meaning, as I do not understand why a file is in a particular folder.

Would this be feasible/beneficial to do such a task as to explain the software engineering behind the Marlin? What kind of items would be critical to understanding?

Thanks for reading guys.

thinkyhead, 2018-06-07:
Sounds like a job for me, and I would like to make a video about this. But I'm too busy chasing down issues right now to get into it.

DerAndere1, 2019-01-31:
Recently, I reverse-engineered the overal code structure of Marlin2.0 (simplified/incomplete) for myself and made a diagram. I attach it as a PDF as well as a zip-compressed LibreOffice Impress file as a starting point. If someone reviewed it carefully, something similar could be added to the Marlin code repository or to http://marlinfw.org/meta/development/ 
[Marlin_code_structure.pdf](https://github.com/MarlinFirmware/Marlin/files/2815608/Marlin_code_structure.pdf)
[Marlin_code_structure.zip](https://github.com/MarlinFirmware/Marlin/files/2815616/Marlin_code_structure.zip)

DerAndere1, 2019-01-31:
Unfortunatelly the diagrams above don't follow any documentation standard (don' know anything about UML etc.). Arrows show mainly information flow within the program. Page 7 was certainly nonsense. Below is a diagram where arrows represent roughly the depency of files with mainly pre-processor directives
[Marlin_code_structure_preprocessor.pdf](https://github.com/MarlinFirmware/Marlin/files/2815906/Marlin_code_structure_preprocessor.pdf)

boelle, 2019-02-20:
@thinkyhead could we make use of the pdf's?

or should this one be put to sleep? or is it just the lack of time that gets in the way?
to me its more a feature request than a question

thinkyhead, 2019-02-20:
Sounds like a job for a specialist. I have an assistant now, and one of his jobs is helping with documentation and the website. User-level documentation will be first, of course, but I will be doing code documentation as well. In the meantime, I have made some videos on how to implement Marlin features.

https://www.youtube.com/playlist?list=PL_pyRRyve8dzcGJ-3rlPSMOd0e1QzlBj7

DerAndere1, 2019-02-21:
I will add updated diagrams to my  fork at https://github.com/DerAndere1/Marlin/tree/Marlin2ForPipetBot/docs so I can improve them continuously . they might be useful for some users that want to customize things and need a very rough first overview as long as the good stuff by you guys is pending.

boelle, 2019-06-22:
missing label

DerAndere1, 2020-01-08:
duplicate of issue https://github.com/MarlinFirmware/Marlin/issues/5768

thisiskeithb, 2022-03-13:
Duplicate of #5768

closed 2022-03-13 by thisiskeithb