MarlinFirmware/Marlin #10950: TMC2130 on Azteeg X3 Pro

plloppii, 2018-06-06:
Hi all,

I am having some trouble configuring the tmc2130s from watterrott with the azteeg x3 pro board.
Things I have done:
-Installed three tmc drivers for X, Y, and Y2
-Wired all of the SD0 together, SD1s and SDK. I connected them to the corresponding SPI pins on the Azteeg. I attached a picture of my board as well as the schematic. 
-Enable tmc2130 and changed X_CS_PIN, Y_CS_PIN, and Y2_CS_PIN to open I/O pins on my board.

When I have #define MONITOR_DRIVER_STATUS enabled, I get Driver Error Reset Printer.
Here is my epprom: 
READ: 		X	Y
READ: Enabled		false	false
READ: Set current	600	600
READ: RMS current	581	581
READ: MAX current	819	819
READ: Run current	18/31	18/31
READ: Hold current	9/31	9/31
READ: CS actual		31/31	31/31
READ: PWM scale	255	255
READ: vsense		1=.18	1=.18
READ: stealthChop	true	true
READ: msteps		0	0
READ: tstep		4294967295	4294967295
READ: pwm
READ: threshold		0	0
READ: [mm/s]		-	-
READ: OT prewarn	true	true
READ: OT prewarn has
READ: been triggered	false	false
READ: off time		15	15
READ: blank time	54	54
READ: hysteresis
READ: -end		12	12
READ: -start		8	8
READ: Stallguard thrs	0	0
READ: DRVSTATUS	X	Y
READ: stallguard		X	X
READ: sg_result		1023	1023
READ: fsactive		X	X
READ: stst		X	X
READ: olb		X	X
READ: ola		X	X
READ: s2gb		X	X
READ: s2ga		X	X
READ: otpw		X	X
READ: ot		X	X
READ: Driver registers:	X = 0xFF:FF:FF:FF
READ: 	Y = 0xFF:FF:FF:FF

When I disable MONITOR_DRIVER_STATUS, I don't get the error on the lcd screen, and I can move the axis fine and they are very silent. However, when I sent M122, I get the same spitback from above. 

I doublechecked my wiring and I am pretty sure it is correct. 
Also, I took a thermal viewer to the board and it seemed like the tmc2130s were 8-10 degrees hotter when ever I turn the machine on

[image]

[image]

[x3pro_wiring.pdf](https://github.com/MarlinFirmware/Marlin/files/2077080/x3pro_wiring.pdf)

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2077087/Marlin.zip)

.

If there is any feedback to help me get these running I would really appreciate it.

plloppii, 2018-06-07:
So I actually did get them running, but my dual seems like its not moving the correct steps per mm. 

So I am getting alittle confused. Should the microstepping jumpers below the drivers be removed for the tmc2130s? Would these two lines of code be affecting my issue?

// Microstep setting (Only functional when stepper driver microstep pins are connected to MCU.
#define MICROSTEP_MODES {16,16,16,16,16} // [1,2,4,8,16]

// Actual motor currents in Amps. The number of entries must match DIGIPOT_I2C_NUM_CHANNELS.
// These correspond to the physical drivers, so be mindful if the order is changed.
#define DIGIPOT_I2C_MOTOR_CURRENTS { 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0 }  //  AZTEEG_X3_PRO

plloppii, 2018-06-07:
Also, I just burnt out two of my steppers on my dual y axis, by trying to increase the micro stepping to 64. I do not know what I did wrong... but I am upset two drivers are fried now.

teemuatlut, 2018-06-07:
You can remove the jumpers because if you soldered the pins to the topside, they're not even connected anymore at that point.

`MICROSTEP_MODES` and `DIGIPOT_I2C_MOTOR_CURRENTS` are not used for TMC drivers.

Setting microstepping to 64 does not burn your drivers.

plloppii, 2018-06-07:
Update: I blew out three drivers

Yeah, I am actually out of ideas, I had it running before, now I cant even get any axis running with my drivers. 

So all I can think of right now is that I changed the microstepping in the software and it is super weird that my drivers were fried. When I tried to move the axis, I got a high pitch noise and smelled magic smoke the next minute. 
The pin that I fried was pin 16, the motor supply voltage pin. 

[image]
[image]
[image]

plloppii, 2018-06-08:
@Iamturbo1978 Would you be able to attach one of your marlin files for me to look at? I saw that you posted in Issue #9673 that you were dealing with an Azteeg X3 pro board and tmc2130s. 
Thank you for the help

Iamturbo1978, 2018-06-08:
@plloppii what files were you looking for?

plloppii, 2018-06-08:
@lamturbo1978 could I see your config.h, config_adv, and pin out files please?

thinkyhead, 2018-06-10:
@plloppii — https://github.com/MarlinFirmware/Marlin/files/2077087/Marlin.zip

boelle, 2019-02-20:
@plloppii problem solved?

closed 2019-02-20 by thinkyhead