MarlinFirmware/Marlin #10951: Feature Request: Toolchange in M600 command

hevilp, 2018-06-06:
Hi,

I have a dual extruder single nozzle single (direct) drive.

I have a filamentsensor and instead of manual changing the filament, I would need the T1 gcode command. My programming skills are not enough to put it myself in the M600 code.

Its possible to create a new command M6XX with the ability to do it?
And: to define the feedrate in the load and unload filament scene would be nice, too.

Kind regards

comps, 2018-06-07:
See `Configuration.h`, you cannot easily change what `M600` does, but you can change what gcodes run when the filament runs out:
```
//#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  ...
  #define FILAMENT_RUNOUT_SCRIPT "M600"
#endif
```

hevilp, 2018-06-07:
Where is your comment?
`See Configuration.h, you cannot easily change what M600 does, but you can change what gcodes run when the filament runs out:

//#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  ...
  #define FILAMENT_RUNOUT_SCRIPT "M600"
#endif
—
You are receiving this because you authored the thread.`

thinkyhead, 2018-06-07:
> [Is it] possible to run a few gcodes step by step? [What] would the code look like?

```cpp
#define FILAMENT_RUNOUT_SCRIPT "M117 Changing Tools\nT1\nM300 S800 P200"
```

hevilp, 2018-06-08:
Thank you very much!

InsanityAutomation, 2021-01-01:
This feature, though hard to use and needing more work, is implemented. See TOOLCHANGE_MIGRATION_FEATURE.

closed 2021-01-01 by InsanityAutomation