MarlinFirmware/Marlin #10952: BLTouch Crashes Into Bed

WillShawx, 2018-06-06:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

The Bed Autolevels On The 9 Points

**Actual behavior:** [What actually happens]

The X/Y Axis Autohome and then the Z Attempts to go to -70 instead of autoleveling. 

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

WillShawx, 2018-06-06:
[10952.zip](https://github.com/MarlinFirmware/Marlin/files/2077765/10952.zip)

smoki3, 2018-06-06:
There are many fake BLTouch there which are not working reliable. So have it worked before?

Roxy-3D, 2018-06-06:
Also...   One more question:   Did the EEPROM configuration version change immediately before this happened?  Did you upgrade the firmware, and then have this happen?     I don't know the reason, but I do believe there is an issue with BL-Touch probes and having the EEPROM configuration version change.

WillShawx, 2018-06-06:
@Smoki3 yes it is a fake from China that has over 1000 orders and 4.8 stars.

WillShawx, 2018-06-06:
@Roxy3D I did not change the marlin 1.1.8 version or the EEPROM Version. 

Thanks

smoki3, 2018-06-07:
So I also had this issue with the fake black touch :(

With the original everything is stock

WillShawx, 2018-06-07:
Does your arm contract / deploy with the g code. 

Mine does not.

boelle, 2019-02-20:
@WillShawx still an issue?

closed 2019-02-20 by thinkyhead