MarlinFirmware/Marlin #10953: DUE + RAMPS-FD v2.1 + TMC2130 Spi communication does not work when TEMP_SENSOR_BED is defined

tig33r, 2018-06-06:
Any value other than 0 for TEMP_SENSOR_BED causing halt and restart board with ramps-fd+due combo+tmc2130. Newest bugfix 2.0. Below log.

log:

`start
echo: Watchdog Reset
Marlin bugfix-2.0.x
echo: Last Updated: 2018-01-20 | Author: (none, default config)
echo:Compiled: Jun  6 2018
echo: Free Memory: 83536  PlannerBufferBytes: 1408
echo:Hardcoded Default Settings Loaded
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X160.00 Y160.00 Z6400.00 E3564.00
echo:Maximum feedrates (units/s):
echo:  M203 X300.00 Y300.00 Z5.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.30 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P30.25 I1.75 D130.71
echo:  M304 P200.00 I13.00 D775.40
echo:Stepper driver current:
echo:  M906 X1000
X driver error detected:
overtemperature
short to ground (coil A)
		X
Enabled		false
Set current	1000
RMS current	550
MAX current	776
Run current	17/31
Hold current	8/31
CS actual		31/31
PWM scale	255
vsense		1=.18
stealthChop	true
msteps		0
tstep		4294967295
pwm
threshold		0
[mm/s]		-
OT prewarn	false
OT prewarn has
been triggered	false
off time		15
blank time	54
hysteresis
-end		12
-start		8
Stallguard thrs	0
DRVSTATUS	X
stallguard		X
sg_result		1023
fsactive		X
stst		X
olb		X
ola		X
s2gb		X
s2ga		X
otpw		X
ot		X
Driver registers:	X = 0x00:00:00:00
Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!`

tig33r, 2018-06-07:
Looks like TEMP_SENSOR_BED is not connected with this. My configuration files in attachment.
[marlin due.zip](https://github.com/MarlinFirmware/Marlin/files/2081678/marlin.due.zip)
[image]

JXPA, 2018-06-08:
I think the pin assignment of TMC_SW_SCK and X_CS_PIN is incorrect.
It is a pinout of RAMPS-FD V2.1 AUX-2.
D15   D14
D69   D68
D61   D60
D59   D58
GND  VCC

tig33r, 2018-06-08:
Probably this is root cause... I will check your advice evening.

Can you share complete pinout for ramps-fd v2.1?

Thanks a lot anyway.

JXPA, 2018-06-08:
I'm using RAMPS-FD V2.2 and FYSETC TMC2130.
I checked the schematic and Due pinout image every time I needed it, so I didn't make RAMPS-FD pinout image.

Just to be on the safe side, Check logic voltage.
In my case, there was a problem of increasing the logic voltage when installing TMC motor driver.
It was caused by 5V supplied from the buffer IC to a EN pin of each driver.
There was no change if only one was attached.
Raised to about 3.5V when two were attached.
Raised to about 3.7V when three were attached.

There was no problem with A4988 or DVR 8825, but voltage rise occurred at TMC2100 and TMC2130. (All my driver modules are made in China)
I don't know if there is a problem with Chinese driver module or whether there is a problem with the Trinamic driver IC, but I'd recommend checking it.

tig33r, 2018-06-08:
I have two wattertot modules but i will check it. For rest axes i will use lv8729 or tmc2208 which also I will check regarding your advice.

Phil1988, 2018-06-13:
@JXPA What is a RAMPS-FD V2.2?
I thought the development was canceled at V2.A and the only boards on the market are v1.2...
I am interessted in the RAMPS-FD you have because I want a DUE with TMC2130 too and will otherwise modify a Ramps "1.6"

JXPA, 2018-06-13:
@Phil1988 
This is V2.2.
[RAMPS-FD V2.2](https://github.com/bobc/RAMPS-FD/tree/dev)
V2.2 is a development version, if you want this you have to make it yourself.

Phil1988, 2018-06-14:
@JXPA Thanks dude - looks like I will mod the Ramps "1.6" instead :)

tig33r, 2018-06-30:
Seems like my issue is solved. Thank you @JXPA for help.

closed 2018-06-30 by tig33r

bruce356, 2018-09-13:
@tig33r hi, where did you purchase your ramps v2.1a board
thanks - bruce

Miran81, 2019-03-20:
@JXPA @tig33r  I have due+ramps-fd v1+tmc2130 . I did all of things but tmc2130 not working. I m using x cs pin 55 and mıso,mosı,sck is servo pins. 
[image]

Bob-the-Kuhn, 2019-03-21:
I have used the TMC2130s on the same hardware.

Please open a new issue & we'll work through this.

When you open the new issue please include the following:
- A ZIP file containing the files **configuration.h**, **configuration_adv.h** and **pins_RAMPS_FD_V1.h**
- Details on your TMC2130 setup such as how many and what slave select pins you are using.

I think I may know what is wrong.  I made errors on the AUX 2 logical pins table in that diagram.  Try using 60 instead of 55.

Also, you'll need one slave select for each TMC2130.  They can share the same MISO, MOS and SCK.

Miran81, 2019-03-21:
@Bob-the-kuhn #13418