MarlinFirmware/Marlin #10954: Servo arm moves during homing / leveling with optical BTouch z probe

Morgy746, 2018-06-07:
Hi all
I got an Anet A8 3d printer with the standard heat bed, but I buyed a Ultrabase glas printing bed.
Bad luck, my old inductive sensor will not recognize the glas bed, so I tried to build this setup

https://www.thingiverse.com/thing:2805327

But I have a big problem with it, I'm trying to be as accurate as possible to tell what I did.

I printed all parts and installed it in my Anet A8, everything fine.
Wiring was no problem because I cut the flex cable to the display at line 3 and soldered the servo pwm signal to it, so I didn't had to solder to the PCB. Also I checked the IR sensor, it will bring 5V if nothing between the sensor and about 0,1V when blocked.
I put the servo arm down, pin head laying on the support, sent M119 and got an open message. Then I push the rod 2mm up, positioned the flag well, so M119 sends a triggered, so far so well. Also the pin is lower than the nozzle by about 4mm so it will trigger when the nozzle is still 2mm over the bed, so hardware should be all right.

So I went to modify the config file.
I started with the Anet A8 autolevel stock anet sensor config file and changed it like this
*changed Z_MIN_ENDSTOP_INVERTING to false
*changed Z_MIN_PROBE_ENDSTOP_INVERTING to false
*checked that Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN is not commented out
*commented out FIX_MOUNTED_PROBE
*activated #define BLTOUCH
*activated #define Z_ENDSTOP_SERVO_NR 0
*activated #define Z_SERVO_ANGLES {115,10}  (and changed the standard values)
*changed X_PROBE_OFFSET_FROM_EXTRUDER to -27 and y_PROBE_OFFSET_FROM_EXTRUDER to 17 and set Z_PROBE_OFFSET_FROM_EXTRUDER to -3 as a default value (fine tuning will come later with m851 command)
*changed Z_CLEARANCE_DEPLOY_PROBE to 5 and checked Z_CLEARANCE_BETWEEN_PROBES to be activated
*checked AUTO_BED_LEVELING_LINEAR is active
*changed the PROBE_BED_POSITION to left 20 right 200 back 200 front 20 because the standard values combined with new nozzle probe offset will bring a compiling error message (cannot reach position)
*checked Z_SAFE_HOMING is active
*changed Z_SAFE_HOMING_X_POINT to 10 and Z_SAFE_HOMING_Y_POINT to 20
*changed NUM_SERVOS to 1
*checked SERVO_DELAY is set to 300 and active
*activated DEACTIVATE_SERVOS_AFTER_MOVE

So far as described in the tutorial, I compiled and uploaded it, but my servo did not move with M280 command.
So I added

#define SERVO0_PIN 27

down at the * R/C SERVO support chapter.

When uploaded, I noticed my extruder now turns backwards, regardless if I click extrude or retract.
So way up in the config file, I changed SWITCHING_EXTRUDER_SERVO_NR from 0 to 1, and now the extruder runs correct.
Also the M280 command will work correctly, sending 
M280 P0 S0 
will move the arm way up, I sent 
M280 P0 S10
and positioned the servo arm on the servo gear so that the pin will be higher than the nozzle so it won't catch to the print, and locked the screw.

So for the final testing, I turned the servo down by
M280 P0 S10
and checked the sensor with M119, will reply open.
So I manually lowered z to about 2mm between nozzle and bed, M119 deliveress triggered.
Setting the servo arm up by
M280 P0 S10 
pulles the rod up (and of course higher as the nozzle) so M119 sends triggered.

Now I wanted to check the homing by G28, but before I put the servo arm down again by M280 P0 S115 (or restarting the printer) and hit G28.

NOW COMES THE PAIN
In the example Youtube Video, the servo arm rotates down and the printer head drived down on the Z axis until the sensor triggeres, drives up until the sensor is open again, drives down until triggered, drives up, moves to next position, just what I would expect, normal z probing behaviour.
But my printer, while driving down the z axis, the servo arm rises!!!
This causes the sensor to trigger up in the air, far away from the bed, then driving up the z axis and at the same time the servo arm rotates down, the sensor is open, now driving down the z axis again (slower) but again the servo arm rises, the sensor triggeres up in the air again.
This happens when homing by G28 and also by leveling with G29.
The G29 command is handled pretty fast because it just jumps from x/y position to the next x/y position and waving the servo arm :)

When the servo arm would stop this behaviour, I would be pretty sure it would drive down to the bed and trigger just as intended, so I could measure my Z offset and go to work, but this way my z axis is homed up in the air.

Why does the servo arm move while homing/leveling?

Currently I am printing with the servo arm turned so it won't trigger when up and after the leveling, I add a code
M280 P0 S
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2078712/Configuration.zip)

0, but I had to move the flag so it will trigger not 2mm between nozzle and bed but with about 0,4mm so the M280 P0 S0 will pull just a tiny bit higher than the nozzle but this is pretty annoying, also when the printed filament will warp only a tiny bit the pin will run into the printing.

Also I tried to change the Z_SERVO_ANGLES {115,10} to other values like Z_SERVO_ANGLES {60,50} so the servo arm has not such a big influence to the rod and won't trigger the sensor when wrongly driving up during homing/leveling, but it seems entering different values has absolutely no effect, the servo arm still moves just like with the (115,10).
Also I commented out the Z_SERVO_ANGLES {115,10} and manually entered #define BLTOUCH_DEPLOY 70 and #define BLTOUCH_STOW 50 but still no change in the rotation of the arm.

How can I check why the servo arm is moving when driving down the z axis while leveling/homing and stop it or, if nothing helps, change the angles of the servo arm so I can add a M280 command in my printer settings after the G28 and G29?

in the attachment, my configuration file.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2078715/Configuration.zip)

thinkyhead, 2018-06-07:
The `BLTOUCH` probe is a "servo," so if you want to also use a switching extruder, it must be attached to the second servo port (`SERVO1_PIN`) and `NUM_SERVOS` must be set to 2. Your switching extruder's angles should be configured here…
```cpp
#define SWITCHING_EXTRUDER
#if ENABLED(SWITCHING_EXTRUDER)
  #define SWITCHING_EXTRUDER_SERVO_NR 1
  #define SWITCHING_EXTRUDER_SERVO_ANGLES { 0, 90 } // Angles for E0, E1[, E2, E3]
  #if EXTRUDERS > 3
    #define SWITCHING_EXTRUDER_E23_SERVO_NR 1
  #endif
#endif
```
You can then use `T0` and `T1` to switch extruder steppers. If you want to change its angles with `M280` then you would need to specify the second servo: `M280 P1 S<angle>`.

Morgy746, 2018-06-08:
No you got me wrong, I don't have a switching extruder.
My problem is that the servo motor rotates during the homing and leveling, pulling the probe and triggering the sensor without touching the heat bed.
It is most probably the #define BLTOUCH line's fault.
The author of the probe named is bltouch like z probe, so I assumed to activate BLTouch.
I had no idea what a BLTouch sensor even is, so I googled and saw that its probe goes in when triggered and jumps out again at the next position, which might be ideal for the BLTouch sensor but not the selfmade IR sensor solution. This also would explain why I can't change the deploy and stow angles.
I would like to test it but my printer is currently running a job, so I will test another z probe setting instead of bltouch and update the thread.

Morgy746, 2018-06-11:
Commenting out the line with bltouch just did it, now everything works great :D

closed 2018-06-11 by Morgy746