MarlinFirmware/Marlin #10955: [bugfix-1.1.x]Dual Extruders, TMC2208, Motor stops in middle of print.

doskoi, 2018-06-07:
I've 5 TMC2208 for all motors on ramps 1.5. E0 and E1 is 0.9a rest is around 0.7a.

The TMC2208 works when I using single extruder, but since I change to dual extruders.
E0 or E1 motor is randomlly stop working in first 10-20 layers. (no matter 2 extruders or just 1 extruder is working)

1. I have checked extruder, I'm sure the motor is didn't spin.
2. When the motor is stop working, I've pause or cancelled print, Use G-code
`G91; G1 E<Positive or nagetive number> F60; ` The motor still not move.
Even press reset button on ramps board. after that, motor may still not response.
But after I power off then power on again. it's returned.

3. I've check TMC2208 on board by multimeter, v-ref is correctly, and it's cooling.
4. I change TMC2208 to DRV8825, motor is working.
5. I've enable/disable HAVE_TMC2208 in configuration file, not working.

How to check or fix this?

### Steps to Reproduce

1. Slice STL file by S3d
2. Export Gcode to octoprint or direct print via USB by S3d
3. After few minute, only x,y,z is movement. e0 or e1 is stoppted.

**Expected behavior:** Extruders motor works as gcode.

**Actual behavior:** E0 or E1 motor has shutdown.

#### Additional Information

My configuration file is here: https://github.com/doskoi/Marlin/tree/bugfix-1.1.x/Marlin

brisma, 2018-06-07:
Have you activated Linear Advance? Try to disable it, TMC2208 is not happy with this feature :)
Or, if you wanna try, enable Junction Devation and try different settings with Linear Advance activated.

There are some interesting information about this problem here:
https://github.com/KevinOConnor/klipper/issues/196

doskoi, 2018-06-07:
Yes, I do have enabled Linear Advance.

I'm gonna disable it to have test it, after confirm the issue is due to LA, I will gonna try to deal with Junction Devation, I don't familiar this yet.

Thank you.

teemuatlut, 2018-06-07:
If you're using serial to control the drivers, then the potentiometer currents don't apply.
If you have not wired them in and are using them in standalone mode, then it doesn't matter whether or not you've enabled them in `config_adv`.

The driver is stopping because it is trying to protect itself from something harmful and can be restarted by a full power cycle (like you've noticed) or in software by reading a specific register to clear the condition.

crzcrz, 2018-06-07:
My extruder motor was stalling randomly with TMC2208 in legacy mode (which runs in StealthChop2 if I understand correctly), regardless of the current. Connecting it with UART and enabling hybrid mode solved that, but of course the stepper is now whining during retracts. LA is enabled.

There's a reply from Trinamic in the related issue that is linked a bit above, which suggests that StealthChop2 mode does not like abrupt changes in velocity (exactly what LA may be doing). This also means that reducing the jerk value for the extruder axis _may_ help (and explains why activating the hybrid mode appears to solves the issue).

I'm going to play with the jerk a bit in the evening and see if I can reliably drive the extruder stepper with StealthChop2.

doskoi, 2018-06-07:
Thanks for your information.

I've disabled Linear Advance, and TMC2008 is working well.

Now I'm trying to enable Junction Devation instead Jerk setting.

doskoi, 2018-06-08:
I've enabled Junction Devation with default 0.02 value.
For few test prints, TMC2208 works with Linear Advance.

closed 2018-06-09 by doskoi

doskoi, 2018-06-09:
That i print a large part with Junction Devation 0.02, TMC2208 still shutdown in mid print. but work duration much longer than use Jerk.

jonasuea, 2020-02-13:
> Obrigado pela sua informação.
> 
> Desativei o Linear Advance e o TMC2008 está funcionando bem.
> 
> Agora, estou tentando ativar a configuração de junção em vez de Jerk.

where is this option?