MarlinFirmware/Marlin #10961: Z-Offset and G33

Floppynator, 2018-06-08:
### Description

After calibrating with G33 and "probe z-offset" over the Delta autocalibration menu it seems to be the printer ignore the selected z-offset. 

you have manually increase or decrease the Delta_height to set the correct offset. Maybe a bug? 
I can manually move to z0 and adjust the z-offset by my self and store it with G33 P0, that also not work. The value was ignored. 

### Steps to Reproduce

1. Do a G33 Autocalibration over Menu
2. Do a Set Z-Offset over Delta Calibration Menu 
3. try to print. z-offset will be ignored
4. increse or decrease the z-offset amount from the delta-height
5. try to print successfully 

**Expected behavior:** [What you expect to happen]
after the Set Z-Offset the offset will work directly

LVD-AC, 2018-06-08:
This is a known issue for quite a while already. You could give PR #9310 a try; that should solve it. But for the moment z-offset changes only take effect in the print after adjusting the delta height accordingly. You could use G33 **P1** after each Z-offset change for that purpose.

app-js, 2018-06-10:
Does it make a difference if I run G33 P1 or use M665 Hx.x?

Floppynator, 2018-06-10:
Not for me. Still in- or decrase the delta_height do it for me :(

boelle, 2019-02-20:
@Floppynator still an issue when using latest bugfix 2.0?

closed 2019-02-20 by thinkyhead