MarlinFirmware/Marlin #10962: BUGFIX-2.0 AZSMZ_12864 AZSMZ MINI Blank Display

xamxamxam, 2018-06-08:
Hello I am using Marlin with azsmz_12864 and AZSMZ MINI board,
I compiled in april and wa ok
Now with the same conf I compiled today 08 of June but the Dislplay is blank.
The SD is working 
The firmware is working

Can someone help me?
I reloaded old firmware and all is ok.....

2012BenCormier, 2018-06-08:
Hello, this happend to me too, it seem that marlin set the contrast to 0 in the update. I just set it with
M250 [C<contrast>] to 200. Hope this is what happend to you...

alexyu132, 2018-06-09:
A workaround that fixed the issue for me was to revert Marlin/src/lcd/dogm/u8g_dev_st7565_64128n_HAL.cpp to the version in commit d3c0241.

thinkyhead, 2018-06-09:
Try making this change in `Marlin/src/lcd/dogm/u8g_dev_st7565_64128n_HAL.cpp`:

```diff
- #define ST7565_V0_RATIO(N)       (0x20 | ((N) & 0x7))
+ #define ST7565_V0_RATIO(N)       (0x10 | ((N) & 0x7))
```

thinkyhead, 2018-06-09:
Try some or all of these changes to `Marlin/src/lcd/dogm/u8g_dev_st7565_64128n_HAL.cpp` until it works…
```diff
- //ST7565_POWER_CONTROL(0x4),  // power control: turn on voltage converter
- //U8G_ESC_DLY(50),            // delay 50 ms

- //ST7565_POWER_CONTROL(0x6)   // power control: turn on voltage regulator
- //U8G_ESC_DLY(50),            // delay 50 ms

+ ST7565_POWER_CONTROL(0x4),  // power control: turn on voltage converter
+ U8G_ESC_DLY(50),            // delay 50 ms

+ ST7565_POWER_CONTROL(0x6)   // power control: turn on voltage regulator
+ U8G_ESC_DLY(50),            // delay 50 ms

  ST7565_POWER_CONTROL(0x7),  // power control: turn on voltage follower
  U8G_ESC_DLY(50),            // delay 50 ms

- ST7565_V0_RATIO(0),         // Set V0 voltage resistor ratio. Setting for controlling brightness of Displaytech 64128N
+ 0x10,                       // ???
+ //ST7565_V0_RATIO(0),         // Set V0 voltage resistor ratio. Setting for controlling brightness of Displaytech 64128N
```
We'd be curious to know the minimum change required to bring it back up.

cdedwards, 2018-06-09:
I have the same issue with my viki2 display. blank display. smoothieware, display is fine. I've tried setting contrast but that did nothing. I will see what I can do tomorrow since I'm in the beginning of a 15 hour print

xamxamxam, 2018-06-09:
dear thinkyhead , I tried your suggestion
it worked!
both changes are necessary to make display visible.
please consider to patch it!
thank you all for help

massimo

thinkyhead, 2018-06-09:
Interesting! I can't find any definitive document that explains that `0x10`, but the rest seems to be pretty standard. I'll patch it shortly.

dot-bob, 2018-06-12:
@xamxamxam  Can you post your configuration?  I have the same display and mainboard on two printers and haven't seen this issue on bugfix-2.0.x as of last night.

For reference the firmware for my two printers: [https://github.com/dot-bob/Marlin/tree/ZCORE](https://github.com/dot-bob/Marlin/tree/ZCORE)  and [https://github.com/dot-bob/Marlin/tree/fcube-2.0.x](https://github.com/dot-bob/Marlin/tree/fcube-2.0.x)

thinkyhead, 2018-06-12:
Well, it was patched 2 days ago, so that might be why it's not appearing.

boelle, 2019-02-20:
@xamxamxam problem solved in latest bugfix 2.0?

closed 2019-02-20 by thinkyhead