MarlinFirmware/Marlin #10963: UBL issue

modem7, 2018-06-08:
### Description

When doing step by step UBL, getting an issue where the points aren't being printed across the whole bed (inset set to 10mm), and once it gets to a certain point, the z moves up to infinity and crashes into the frame.

### Steps to Reproduce

1. Step by Step UBL
2. Verify mesh PLA

[2018-05-31 17-27-37.zip](https://github.com/MarlinFirmware/Marlin/files/2085437/2018-05-31.17-27-37.zip)
[image]

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

gloomyandy, 2018-06-08:
What version of Marlin are you using? You should probably try a recent bugfix version as many fixes and changes have been made to UBL in recent months.

modem7, 2018-06-08:
Hey @gloomyandy - 1.1.8, whichever is the latest version

gloomyandy, 2018-06-08:
Have you tried either bugfix-1.1.x or bugfix-2.0.x? If not then you really should, many changes in these versions and you are unlikely to get anything fixed in 1.1.8.

modem7, 2018-06-08:
Will check that now!

modem7, 2018-06-08:
OK, with the latest bugfix version (https://github.com/MarlinFirmware/Marlin/archive/bugfix-1.1.x.zip) I get the head trying to crash into the bed in infinity, and it's still not doing the mesh test across the entire bed. 

21:40:33.212 : Bed Topography Report:
21:40:33.243 : (0,9)                                                                   (9,9)
21:40:33.259 : (34,150)                                                                (190,150)
21:40:33.493 : 0.543  [ 0.287]   0.170    0.018   -0.114   -0.314   -0.475   -0.627   -0.929
21:40:33.493 : 
21:40:33.664 : 0.493    0.406    0.214    0.152   -0.010   -0.179   -0.244   -0.446   -0.767   -1.013
21:40:33.664 : 
21:40:33.820 : 0.548    0.279    0.232    0.140   -0.010   -0.094   -0.321   -0.461   -0.655   -0.944
21:40:33.820 : 
21:40:33.976 : 0.498    0.309    0.224    0.080    0.003   -0.206   -0.246   -0.685   -0.829   -1.028
21:40:33.976 : 
21:40:34.132 : 0.543    0.274    0.182    0.062    0.048   -0.254   -0.299   -0.533   -0.772   -0.976
21:40:34.132 : 
21:40:34.288 : 
21:40:34.288 : 
21:40:34.444 : 0.481    0.456    0.262    0.085    0.010   -0.214   -0.423   -0.503   -0.700   -0.976
21:40:34.444 : 
21:40:34.600 : 0.628    0.386    0.356    0.145    0.023   -0.199   -0.411   -0.523   -0.819   -0.971
21:40:34.600 : 
21:40:34.756 : 0.580    0.416    0.351    0.085    0.030   -0.102   -0.461   -0.515   -0.819   -1.013
21:40:34.756 : 
21:40:34.912 : 0.580    0.369    0.219    0.155   -0.030   -0.229   -0.461   -0.680   -0.792   -0.988
21:40:34.943 : (34,10)                                                                    (190,10)
21:40:34.959 : (0,0)                                                                     (9,0)
21:40:37.455 : echo:enqueueing "G28"

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2086051/Marlin.zip)

[image]

thinkyhead, 2018-06-09:
You may also want to test with `AUTO_BED_LEVELING_BILINEAR` so we can determine whether it's a UBL issue or a more general issue with `G26`.

Note that with `AUTO_BED_LEVELING_UBL` points outside of the mesh are not handled, so the nozzle will raise up to the height set in `UBL_Z_RAISE_WHEN_OFF_MESH` for anyplace outside the mesh. Thus it's important with UBL to make the mesh itself cover the whole bed (`#define MESH_INSET 0`), allow it to probe the points that _can_ be reached by the probe first with `G29 P1`, then use `G29 P2` to fill in the points that _cannot_ be reached by the probe, and then save the mesh to EEPROM.

modem7, 2018-06-09:
I'll give the mesh_inset 0 a try tomorrow, and then work my way back!

Appreciate the heads up!

Roxy-3D, 2018-06-09:
Yes!  Just walk through the 'Cheat Sheet' here:  https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_features/unified_bed_leveling.md
```
;------------------------------------------
;--- Setup and initial probing commands ---
;------------------------------------------
M502            ; Reset settings to configuration defaults...
M500            ; ...and Save to EEPROM. Use this on a new install.
M501            ; Read back in the saved EEPROM.  

M190 S65        ; Not required, but having the printer at temperature helps accuracy
M104 S210       ; Not required, but having the printer at temperature helps accuracy

G28             ; Home XYZ.
G29 P1          ; Do automated probing of the bed.
G29 P2 B T      ; Manual probing of locations USUALLY NOT NEEDED!!!!
G29 P3 T        ; Repeat until all mesh points are filled in.

G29 T           ; View the Z compensation values.
G29 S1          ; Save UBL mesh points to EEPROM.
G29 F 10.0      ; Set Fade Height for correction at 10.0 mm.
G29 A           ; Activate the UBL System.
M500            ; Save current setup. WARNING: UBL will be active at power up, before any `G28`.
;---------------------------------------------
;--- Fine Tuning of the mesh happens below ---
;---------------------------------------------
G26 C P5.0 F3.0 ; Produce mesh validation pattern with primed nozzle (5mm) and filament diameter 3mm 
                ; PLA temperatures are assumed unless you specify, e.g., B 105 H 225 for ABS Plastic
G29 P4 T        ; Move nozzle to 'bad' areas and fine tune the values if needed
                ; Repeat G26 and G29 P4 T  commands as needed.

G29 S1          ; Save UBL mesh values to EEPROM.
M500            ; Resave UBL's state information.
```

modem7, 2018-06-09:
This is the result I get with mesh inset set to 0

[image]

modem7, 2018-06-09:
Same issue after clearing EEPROM as per the cheat sheet. Seems it's a bug of some sort.

Cleared EEPROM
Inset set to 0
Config and adv in previous post.

thinkyhead, 2018-06-10:
Here are some things you can try:

- Check the accuracy of your probe using `M48`.
- Try subtracting 0.15 from your Z probe Z offset, then re-probe and re-test `G26`.
- Adjust the tension of your Y belt.

modem7, 2018-06-10:
@thinkyhead If I go back to 1.1.4 everything works, the issue only presents on 1.1.8 latest bugfix. This isn't a hardware issue, it's software, either settings have changed on 1.1.8 which are conflicting, or it's an unusual bug I'm triggering.

Roxy-3D, 2018-06-10:
The circle drawing was changed to use G2 arcs a while ago...   Please turn off ARC_SUPPORT.   It is very likely G26 will start working again when you do that.    If that is the case, we will need to look closer at the Arc code and how that is implemented for UBL.

modem7, 2018-06-10:
ARC_SUPPORT disabled but still not getting full bed being printed in this case. 

[image]

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2088055/Marlin.zip)

Roxy-3D, 2018-06-11:
Yeah...   one of the developers is going to need to step through and debug things.    The problem is, I'm all setup for 32-bit work putting an RTOS under Marlin.      I might take a time out to do this...  but it would be better if somebody else figured out what is going wrong.

What ever is wrong, it isn't much.   It will probably be just a line or two of code to make everything right again.

gloomyandy, 2018-06-11:
It might hekp if you could post a short video clip showing what is happening. That way anyone trying to reproduce the issue can be sure they are seeing the same problem. I've been following this, but I'm still not sure I understand exactly what the issue is.

modem7, 2018-06-11:
One video coming up!

https://www.dropbox.com/s/s1wwlmer8g30nnn/20180611_125424_1.mp4?dl=0

So as we can see, the mesh isn't being printed on the entire bed, and a lovely crunch as it goes down to infinity

gloomyandy, 2018-06-11:
Ouch! That's not nice! Could you provide a G29 T output for the mesh that was used when making the video? I assume it is different to the one above (I think you changed the MESH_INSET).

gloomyandy, 2018-06-11:
Looking at the video up to the point of the nozzle crashing into the bed the test pattern seems to be complete and looks ok (perhaps a little squashed?). Are there problems that are not obvious in the video?

modem7, 2018-06-11:
Aye, I think the Z offset may be a little off, haven't really dialled it in completely until I sort this out (constant reflashing firmware n all).

But apart from the two UBL issues above, haven't noticed anything further. But it certainly looks like it's not seeing a 200x200 bed, but something smaller. Which is odd because the probe homes exactly as intended (200x200/2).

Will edit this with the G29 T

gloomyandy, 2018-06-11:
Are there some values missing from that G29 T? There is something odd going on if not. Look at the 4th and 10th line from the top they only have 4 values instead of 10. It also seems a little odd that your mesh starts at 34, 5 and runs to 195, 150, your bed is supposed to be 200x200. Can you post and up to date copy of your configuration files. Make sure you include both.

modem7, 2018-06-11:
I didn't do any smart fill in for this, just in case something like this came up and it was obscured. 

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2090773/Marlin.zip)

Could it be related to the settings in the above coment? (https://github.com/MarlinFirmware/Marlin/issues/10963#issuecomment-396064433)

Roxy-3D, 2018-06-11:
You should have your mesh fully populated with values.

modem7, 2018-06-11:
Normally I run the G29 under the step by step menu (result above), then smart fill in a couple of times, and only then do I print the mesh (the video was after smart fill in). 

Would you like me to post the values after smart fill in?

gloomyandy, 2018-06-11:
Yes please post the mesh as it is when you actually use it. I can't see any reason from your configuration why the mesh would start at (34,5) or end at (195, 150). I did notice that you have a MESH_INSET of 5 set. Personally I think that would be better as 0 so that the mesh covers all of the printable area.

modem7, 2018-06-11:
@gloomyandy all done for you sire: https://github.com/MarlinFirmware/Marlin/issues/10963#issuecomment-396319655

The second entry (after smart fill in) is what I'd use normally.

gloomyandy, 2018-06-11:
There is something very odd going on with those posted mesh listings. Do they actually look like that or is there some problem with the way you are posting them? There seem to be values missing (in the case of the second one entire lines, you only seem to have 9 rather than 10!). Perhaps it is some problem with my browser, do you see a full mesh posted above?

modem7, 2018-06-11:
Just in case there are, here is the notepad file, directly copied from repetier:

[ubl.txt](https://github.com/MarlinFirmware/Marlin/files/2091099/ubl.txt)

gloomyandy, 2018-06-11:
I've just noticed that your X and Y probe offsets are 34 and -50 which is exactly the same as the odd offsets in your mesh (starts at X = 34 and has a Y size that is 50 smaller then the bed). I have never seen UBL construct a mesh this way. @Roxy-3D  isn't a UBL mesh normally the same size as the bed minus any MESH_INSET?

gloomyandy, 2018-06-11:
Yep still has missing data, could you try G29 T1 that may have a more compact format.

modem7, 2018-06-11:
Since I started using the BLTouch, I've used this mount + their settings:

https://www.thingiverse.com/thing:2394428

> My Marlin firmware settings for this are as follows:
> 
> These settings under the Z probe options:
> 
> define X_PROBE_OFFSET_FROM_EXTRUDER 34 // X offset: -left +right [of the nozzle]
> define Y_PROBE_OFFSET_FROM_EXTRUDER -50 // Y offset: -front +behind [the nozzle]
> define Z_PROBE_OFFSET_FROM_EXTRUDER -1.8 // Z offset: -below +above [the nozzle]
> Auto bed leveling (Bi-linear):
> 
> // Set the boundaries for probing (where the probe can reach).
> 
> define LEFT_PROBE_BED_POSITION 35
> define RIGHT_PROBE_BED_POSITION 194
> define FRONT_PROBE_BED_POSITION 10
> define BACK_PROBE_BED_POSITION 150

gloomyandy, 2018-06-11:
That still seems to have data missing from it. I wonder if your system is dropping characters for large output from Marlin.

I'm at a lost as to why your mesh has those start and end points. Are you sure the configuration files you have posted are what is actually being used to compile your firmware (the mesh looks more like the result of using MESH_BED_LEVELLING not UBL. Have you made any modifications to any of the other files from the download? I notice that your configutation.h file contains a version number 010107 the current bugfix is 010109. You may be better off starting again with your config file to make sure that everything is up to date. 

Another thought, are you sure the firmware on your machine is the bugfix-1.1.x version? Older versions of Marlin used to configure the mesh as you are seeing it when using UBL.

thinkyhead, 2018-06-11:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Copy the log output into a **.TXT file** and attach it to your next reply.

modem7, 2018-06-11:
OK, I've redownloaded the latest bugfix, redid the config and adv from scratch and enabled debug as well.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2091765/Marlin.zip)

gloomyandy, 2018-06-11:
The two configuration files you have uploaded are not from the latest bugfix release (the configuration.h file still shows the version as 010107, the latest bugfix is 010109). Did you attach the correct files? Also the log file seems to indicate that the firmware used to generate the log was version 1.1.8 not the bugfix version. What did you do to download/configure/build/install the latest bugfix?

modem7, 2018-06-11:
Hmmmm - I sincerely apologise. I will see what the heck is going on my end, because things should be uploaded as intended. I will re-do everything tomorrow, double checking things are 010109.

Thank you so far guys, sorry for the idiocy on this end!

Roxy-3D, 2018-06-11:
The bugfix version of UBL is more stable than the release version....   It would be nice to get the release version caught up.

lrpirlet, 2018-06-12:
@modem7 @Gloom

modem7, 2018-06-12:
OK, just uploading these as a matter of course.

As usual, it looks like human stupidity and a pursuit of shortcuts *may* have been the cause here. 

Give me 30 minutes to see if everything is working and I'll confirm the process I used and how I was a moron so that if further muppets come along, we can make sure they're not being as silly as I am (just need to verify to see if head crashes or rises, then we'll go on from there)

[debug.txt](https://github.com/MarlinFirmware/Marlin/files/2095332/debug.txt)
[bed levelling.txt](https://github.com/MarlinFirmware/Marlin/files/2095333/bed.levelling.txt)

modem7, 2018-06-12:
Confirmed. ID10T error.

I was using https://github.com/akaJes/marlin-config to setup and transfer my configs over. 

This obviously didn't do a very good job, and messed a few things up, I suspect that it was trying to mix 1.1.8 with bugfix files which really messed things up. 

Once I downloaded the bugfix + modified the files manually, everything seems to work as expected. 

Apologies all who helped me, and thank you so much for spending your time to do so! You've been an immense help. Sorry for the idiocy!

closed 2018-06-12 by modem7

thinkyhead, 2018-06-12:
Thanks for the update! It's good for us to know where a configuration can cause weird things to occur. We might be able to flag or catch it early.