MarlinFirmware/Marlin #10964: My E Stepper always go in same direction TMC2130

obor00, 2018-06-08:
### Description
When doing a G1 E10 or G1 E-10, the E-stepper always move  in the same direction.
the "-" seems to be igonred

### Steps to Reproduce

Configuration:
- last version Marlin-bugfix-1.1.x from 04/06/18
- board Rumba
- 5 steppers TMC2130, X and Y use sensor less
- Stealthhchop mode 
- coreXY 
- Dual Z motors

To reproduce:
1. Power on
2. Turn on the heater 0
3 wait temperature ok
4 execute a G1 E10 and check direction
5 Execute a G1 E-10 and see same direction

**Expected behavior:** 
G1 E10 move in one direction
G1 E-10 move in opposite direction

**Actual behavior:**

#### Additional Information
G1 E10 move in one direction
G1 E-10 move in same direection than G1 E10

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2085519/Configuration.zip)

thinkyhead, 2018-06-09:
Are you using Linear Advance?

@teemuatlut, @Sebastianv650 — What did we determine will cause this? The only issues I could find that mentioned a steppers going the wrong way were #9816, #10335, and #10605.

Sebastianv650, 2018-06-09:
@thinkyhead according to his config, he isn't using LA. Reading the two last linked issues, I would think it's something TMC2130 related.

teemuatlut, 2018-06-09:
If you can, please test with another type of driver to narrow down the cause. Leave the configuration the same.

There are only two ways to determine the direction of rotation for the TMC drives. One is through software by setting the `shaft_dir` bit and another one is toggling the DIR pin which is sampled by the driver.
Marlin does not use the former and testing with another driver would help with the DIR pin cause.
The stepper library does set pin modes in `begin`, so that's another possibility as well.

obor00, 2018-06-09:
Hi guys, thanks for your hot support ! :)
I replaced the E-stepper with a classic one TMC2110 without SPI, and comment out E in  the config file:

 //#define E0_IS_TMC2130

Now, both Direction works fine, it can extrude and retract.

So there may be something wrong with my driver TMC2130.  I'll test it on another axis to check if it works

obor00, 2018-06-10:
Confirmed: the  TMC2130 initially on extruder  had a problem  and works only one way. 

So I confirm it was hardware problem,  and not a bug

closed 2018-06-10 by obor00