MarlinFirmware/Marlin #10966: 1.1.x BugFix - Step loss

houseofbugs, 2018-06-08:
I have been testing the latest bigfixes on 2 machines here. One is a Tornado running the stock board and the other is a custom machine with a RAMPS board in it. Since switching the custom one over to bugfix I am getting step loss about 75% of the time in a print. Tried lowering accel, jerk, and print speeds. This printer works perfectly on 1.1.8.

Firmware: https://content.timothyhoogland.com/files/ShareXUpload/Marlin-bugfix-1.1.x_SMARTT.zip

Running DRV8825 Drivers on the RAMPS board in 1/32 step mode.

thinkyhead, 2018-06-08:
The planner and stepper code in Marlin 1.1.8 was the result of years of refinement and testing, and although it's far from "perfect" it does seem to handle jerk and acceleration in a more gentle way than more recent code. I would love to be able to figure out the fundamental differences between that code and more recent code to understand why it doesn't exhibit the same problem.

At some point after 1.1.8 (reports are all over the place about when) there seems to be an increase in layer shifting. We're been completely unable to narrow down the cause or even the point in time when it started, though we have a big catalog of reports and theories.

I had hoped that going forward with refining the planner and stepper code would lead to a solution for the issue, but so far we have not been able to get back to the apparent reliability of 1.1.8, in spite of a lot of careful analysis.

There are a lot of things you can try at this point with the latest bugfix code. For example, you can try enabling `JUNCTION_DEVIATION` or `S_CURVE_ACCELERATION`, which each have strong effects on the way jerk and acceleration are handled. Junction deviation might eventually replace the previous jerk limiting code, but for now it's optional. And for that reason I think we do still need to make sure the customary jerk code is working correctly.

Do you have a small object that you can print which always exhibits the error in a pretty consistent way? That would help with testing. Because if the issue only pops up once every few prints, it becomes very hard to tell whether the issue is really fixed, or if it's just "getting lucky" and not exhibiting the issue.

Do you hear a "thunk!" or other obvious noise when the layer shifting happens?

To see if the problem is related to our latest changes, can you try the bugfix code from a few weeks ago —around May 19— before we merged the planner/stepper refactor code, and see if it's any better? That will at least tell us whether we need to worry more about the recent changes, or more about something that came earlier.

I'm still reserving the possibility that the issue might have nothing to do with the planner / stepper code, but could be due to something else getting in the way of the regular movement. I've done a lot of audits of the planner/stepper changes from around the time that most people claim the error seems to have started, and so far it has been very inconclusive. I even made a branch reverting the planner / stepper changes back to 1.1.8 and had some users test that to see if it helped, and it did not seem to make a difference. So… at this point it is still very mysterious.

houseofbugs, 2018-06-08:
I have S_CURVE_ACCELERATION enabled. I will try with the junction option next.

thinkyhead, 2018-06-08:
Duplicate of #10446

GMagician, 2018-06-09:
I read about some of these...I have an idea, maybe crazy, is not possible that old marlin use less resource, these machines are at "cpu" limits  but work ok, new marlin uses more resources and sometime it reach limits and lose steps?

forkoz, 2018-06-09:
J-Dev and S_Curve work on 8 bit boards? Previous config said S-Curve was 32bit only. Will it raise CPU load? I am barely holding on in 1/32 with 20T pulleys and all. I wish I could have marlin features with klipper like offloading of calculations to PC since I print over USB anyway. Klipper requires raspi ($40) and 32bit requires a new board ($40) so no way to have smooth + fast 1/32 or 1/64 without paying that price it seems.

Also DRV8825 has stuttering issues which may have just manifested itself. I have not had any step loss yet keeping to 62mm/s achievable speed. Print quality only got better with newer marlin.

FiCacador, 2018-06-09:
I too dream of a Klipper with Marlin features... or even better, a Marlin branch that uses the Klipper approach.
On one printer I'm using Klipper from an old notebook instead of a raspi and Pronterface to control, I need to print a bunch of large prints fast, no fancy features required, and it's doing great.

Did anyone had (not TMC driver reletad) layer shifting issues on 32 bit boards?

thinkyhead, 2018-06-09:
> J-Dev and S_Curve work on 8 bit boards?

Yes, they work just fine on AVR.

houseofbugs, 2018-06-10:
Tested with S_Curve off and Junction on and off in combination with s_curve. Still exhibited step loss. I even swapped out for A4988 drivers and it still had step loss at random. I reinstalled the DRV8825s with the smoothers that were fitted as well. Rolled back to the 1.1.8 build and ran the same GCode. No step loss. Succeeded on the first print.

houseofbugs, 2018-06-10:
@thinkyhead I haven't dug around in the code but is there an easy way to see the CPU load? I am going to disable the LCD and retry to see if I get step loss. If I don't that could be a CPU issue?

FWIW the Tornado is running A4982 drivers with a Titan extruder and 1/16 stepping on all axises with a RRD LCD. No step loss has happened on that machine. Both are running bilinear ABL on them.

houseofbugs, 2018-06-10:
Closing out and will comment on https://github.com/MarlinFirmware/Marlin/issues/10446

closed 2018-06-10 by houseofbugs

thinkyhead, 2018-06-10:
More patches to planner / stepper code upcoming. When the changes have been merged to `bugfix-1.1.x` I'll post about it in #10446 for testing.