MarlinFirmware/Marlin #10967: UBL Marlin 1.1.8 RAMPS 1.4 FGC

1liminal1, 2018-06-09:
Hey guys, 

Little confused. I have UBL enabled in the firmware. I have a menu option for it under Prepare. 

I go into the Step by Step menu and click step one "build cold mesh" Its starts out ok. But at some point it tries to go to a point just off the bed. I then updated the settings to say my bed was 5mm smaller that it actually is. But it still does the same thing. 

Can anyone advise what the issue could be?

1liminal1, 2018-06-09:
Update,

Have changed the X min and Y min thinking they may be the culprits. Have it spot on now and its still going off the bed at the same place

modem7, 2018-06-09:
Probably worthwhile posting your config and adv settings so peeps can have a gander

1liminal1, 2018-06-10:
Thanks for that. 

These configs are pretty long lol. Is there an area that's pertinent?

I was trying to reason out, that it could be the bed size or the min x/y that could effect it going off the bed. But it seems to ignore those settings, as I said, making the bed 5mm smaller each side, had no effect. 

What other parts do you want? :)

1liminal1, 2018-06-10:
Also this is my 

DEFAULT_AXIS_STEPS_PER_UNIT: {100,100,400,100}

Its a AM8 (converted from a anet a8) runnings RAMPS 1.4

modem7, 2018-06-10:
Your best bet is zip em up and upload it (drag and drop onto the post)

1liminal1, 2018-06-10:
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/2087511/Configuration_adv.zip)

Sure, here you go!

Thanks

thinkyhead, 2018-06-10:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

1liminal1, 2018-06-11:
Well bugfix seems to have broken it even more. 

Had to create a config.h from scratch, think I updated the pertinent parts. 

Try to auto home and it just crashes. Screen freezes, have to do a hard reset.

thinkyhead, 2018-06-12:
We seem to have a mysterious crashing issue with `bugfix-1.1.x` right now, so please test with `bugfix-2.0.x` in the meantime.

1liminal1, 2018-06-12:
Hey, 

Thanks for that, ive gone back to non-bugfix, and just using bilinear. I might try this weekend.

boelle, 2019-02-20:
@1liminal1 did you try it?

closed 2019-02-20 by thinkyhead