MarlinFirmware/Marlin #10969: SINGLENOZZLE  does not work

lavato, 2018-06-09:
I am using Marlin 2.0.x/2137 with SINGLENOZZLE setup but it does not seam to work.
Setup: Custom I3\RAMPS 1.4\Cyclops\2 extrudes

When printing something that uses only my second extruder, printer starts printing immediately without heating the nozzle.

GCODE has only statements like M104 S230 T1

cdedwards, 2018-06-09:
You need to check over your slicer setup. I have a dual extruder single nozzle setup and mine works. T1 is Tool number 1 which would translate to the second extruder. Tool 0 is the first. M104 does not wait for heatup. M109 will wailt.

lavato, 2018-06-10:
I have check and can also find M109  - so this must be something else.

See below the gcode snippet:
```
M190 S80 ; set bed temperature and wait for it to be reached
M104 S230 ; set temperature
;G29 L1
;G29 A
G28
;G29 J
G1 Z5 F5000 ; lift nozzle
M109 S238 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
```
etc.

Note that this works if I use both extrudes. It is only an issue if I use T1 only.

lavato, 2018-06-10:
I checked and can find M109  - so this must be something else.

See below the gcode snippet, generated by Slic3r:
```
M190 S80 ; set bed temperature and wait for it to be reached
M104 S238 T1 ; set temperature
;G29 L1
;G29 A
G28
;G29 J
G1 Z5 F5000 ; lift nozzle
M109 S238 T1 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
```
etc.

Updated: 
The only time this works is if you have a statement like:  "M104 S238"
If you specify only T1 like "M104 S238 T1", the hot end does not heat up.

**This is tagged incorrectly, in my opinion, as this is a bug.**

cdedwards, 2018-06-10:
you are correct, this is a bug. M109 T1 S238 is not doing anything. something in M109 T1 is causing this command to return immediately instead of setting temp and then waiting.

thinkyhead, 2018-06-10:
The `T` parameter to `M104`/`M109` is not the "tool" index, but the hotend index. If there's only 1 hotend, then `M109 T1` has nothing to do.

cdedwards, 2018-06-11:
so if this is a hotend index. how do we specify a extruder motor. I need to be able to switch between two motors not two hotends.

cdedwards, 2018-06-11:
when I tell simplify3d to switch to my other extruder, it sends a T[number] command to switch. Selecting another tool index will cause this as well

skasti, 2018-06-11:
As far as I understand, when you are in SINGLENOZZLE mode, the printer should ignore any M109/M104 that specifies a T other than the currently active tool.

A normal tool change from cura looks something like this:
`T1
M104 T0 S190
M109 S220`

`M104 T0 S190` should be ignored, as the previous T1 command sets tool 1 active.

Note that I am not 100% sure on this, so correct me if I'm wrong :)

thinkyhead, 2018-06-12:
> As far as I understand, when you are in `SINGLENOZZLE` mode, the printer should ignore any `M109`/`M104` that specifies a T other than the currently active tool.

Just reviewing the code now, it appears you are correct!

```cpp
void GcodeSuite::M109() {

  if (get_target_extruder_from_command()) return;
  if (DEBUGGING(DRYRUN)) return;

  #if ENABLED(SINGLENOZZLE)
    if (target_extruder != active_extruder) return;
  #endif
  . . .
```

cdedwards, 2018-06-12:
and therein lies the problem. in a single nozzle system, do not ignore a heatup command. We should not care what hotend is specified by the T command as we only have one. We are using the T command in our case to set a extruder motor instead of a hotend, so what does it matter if the target_extruder != active_extruder.

thinkyhead, 2018-06-12:
I'm not sure why it's always been this way. The original coder must have had some justification. I'll see if I can track them down and find out why.

skasti, 2018-06-13:
I understand why it is this way: When changing extruder part-way through a print, the G-code sequence is this:

```
G92 E0
G22
T1
G92 E0
G23
M109 S220
M104 T0 S190
```

If you don't ignore the M104 T0 190, you will cool the hotend and get a clog

lavato, 2018-06-14:
@cdedwards 
Yes  you are correct, the active tool is 0.

BTW - for some reason github.com was complacently inaccessible for me for couple of day!?!

thinkyhead, 2018-06-14:
- Is the issue that the slicer is trying to turn off or cool down the inactive hotend between tool-changes?
- Is there a setting in your slicer that will simply leave the hotend(s) alone between tool-changes, instead of trying to cool down the inactive one?

lavato, 2018-06-14:
@thinkyhead 

1) No
2) This is not an issue.
3) Agree 

Marlin in the SINGLENOZZLE config,  should understand that it has to activate the only hotend it has, when it executes  e.g. `M104/9 S238 T0/1`.

Currently it only works if the gcode has no T at the end `M104/9 S238`.

If not fixed, when in SINGLENOZZLE  mode and using only the second extruder, Marlin air prints.
I.e. Printer prints nothing,  however the gantry moves.

thinkyhead, 2018-06-14:
> the hot end temp target is 0.

How does the target temperature get set to 0? An `M104`/`M109` that targets a non-existent hot-end should just exit without changing the target temperature.

lavato, 2018-06-15:
When `M1049 S238 T1` it seams never to be set at anything. 

The way I see it, when using extrude 2 only  (T1), Slic3r gcode unchanged:
```
M190 S80 ;         >>>>>>>>EXECUTED
M104 S238 T1 ;  >>>>>>>>IGNORED
G1 Z5 F5000 ; >>>>>>>>EXECUTED
M109 S238 T1 ; >>>>>>>>IGNORED
G92 E0 >>>>>>>>EXECUTED
etc
```

When using extruder 1 (T0) (manually added T0  to gcode, as when using T0 only Slic3r removes T0) :
```
M190 S80 ;         >>>>>>>>EXECUTED
M104 S238 ;  >>>>>>>>EXECUTED
G1 Z5 F5000 ; >>>>>>>>EXECUTED
M109 S238 ; >>>>>>>>EXECUTED
G92 E0 >>>>>>>>EXECUTED
etc
```

When using extruder 1 (T0), Slic3r gcode unchanged
```
M190 S80 ;         >>>>>>>>EXECUTED
M104 S238 ;  >>>>>>>>EXECUTED
G1 Z5 F5000 ; >>>>>>>>EXECUTED
M109 S238 ; >>>>>>>>EXECUTED
G92 E0 >>>>>>>>EXECUTED
etc
```

thinkyhead, 2018-06-15:
> The way I see it, when using extruder 2 only (T1) - Slic3r gcode unchanged:

If the command `T1` were sent first, then everything would work.

lavato, 2018-06-16:
> If the command T1 were sent first, then everything would work.

Not sure what the question or the statement means?
T1 is sent first?
Do you mean if I reverse execution on below, what would happen?

```
M190 S80 ;         >>>>>>>>EXECUTED
M104 S238 T1 ;  >>>>>>>
```>IGNORED

thinkyhead, 2018-06-30:
> Do you mean if I reverse execution on below…?

No, I mean, if `T1` was ahead of the commands…

```gcode
T1
M190 S80
M104 S238 T1
```

boelle, 2019-02-20:
@lavato problem solved in bugfix 2.0?

closed 2019-02-20 by thinkyhead

InsanityAutomation, 2019-02-20:
To elaborate on this being closed, SINGLENOZZLE got quite a bit of attention around the holidays so its pretty much a totally different animal at this point...