MarlinFirmware/Marlin #10970: Manual Bed temperature adjustment resets Max Bed temperature to an undefined  number

lavato, 2018-06-09:
I am using Marlin 2.0.x/2137 with flowing setup: Custom I3\RAMPS 1.4\Cyclops\2 extrudes.

When manually adjusting Bed temperature using menus (LCD), max temperature value is limited to an undefined value.

Steps to reproduce the issue:
1) Prepare -> PreheatABS -> PreheatABS >>>>>>>> Sets Bed Temp to 80
2) Control -> Temperature -> Bed  >>>>>>>> Sets Bed Temp to 75??? - where did that come from?

GMagician, 2018-06-09:
> Margin for what?

Maybe to detect overtemp?
> Also, I cannot see which line of code is this coming from - do you?

ultralcd.cpp line 3488 for example:
`      MENU_MULTIPLIER_ITEM_EDIT_CALLBACK(int3, MSG_BED, &thermalManager.target_temperature_bed, 0, BED_MAXTEMP - 15, watch_temp_callback_bed);
`

lavato, 2018-06-09:
Thank you for pointing out the code snippet!
To me this behaviour does not make scene. 
E.g. 
If I set BED_MAXTEMP to 80, I expect that any "M190 S80" would work, including ability to set it via LCD menu.

Does this behaviour make sense to you?

GMagician, 2018-06-09:
Yes it make sense. Marlin can't know what is maximum measurable temp, you specify that.
Then if you ask me if 15 as margin is a good thing then, that's another story.

In my cnf I set (260+15) for nozzles just to remind me this.

AnHardt, 2018-06-09:
Just make a test and remove the "- 15". Then heat per menu to MAXTEMP,
You'll probably get an MAXTEMP error - because of the overshot.

thinkyhead, 2018-06-10:
I'm not sure of the historical reason for having a 15° margin on the allowable temperature. From the comment by @AnHardt I assume the reason is that the `BED_MAXTEMP` value is supposed to be set higher than the highest temperature you would ever use to allow for overshoot during initial heating. So, if the highest temp you ever want to use is 110° then set `BED_MAXTEMP` to 125. Then if the temperature ever goes over 125 Marlin will panic to prevent a fire.

closed 2018-06-10 by thinkyhead

lavato, 2018-06-10:
OK, this makes sense - thank you all for explaining!

AnHardt, 2018-06-10:
MAXTEMP is not the maximum temperature you want to print with.
MAXTEMP is the maximum temperature your hardware can stand without suffering too much.