MarlinFirmware/Marlin #10972: Boards.h compiling issues

gian1312, 2018-06-09:
Dear all

While verifying my Marlin sketch (fresh cloned from GitHub) following error appears while choosing boards.  <br/> 
`#ifndef MOTHERBOARD  

  #define BOARD_ULTIMAKER  

#endif`  

following Error appears (independent which board is used)  
`boards.h:53: error: operator '==' has no right operand  

 #define MB(board) (MOTHERBOARD==BOARD_##board)

                                              ^

sketch\pins.h:1402:5: note: in expansion of macro 'MB'

 #if MB(ULTIMAKER)

     ^

In file included from sketch\Marlin.h:23:0,

                 from sketch\thermistortables.h:4,

                 from sketch\Configuration.h:800,

                 from C:\Users\USER\Documents\Marlin\Marlin\Marlin.ino:33:

pins.h:2962: error: #error Unknown MOTHERBOARD value in configuration.h

 #error Unknown MOTHERBOARD value in configuration.h

  ^

exit status 1
operator '==' has no right operand`<br/>

I am really sorry for the format of the code.

gloomyandy, 2018-06-09:
You MOTHERBOARD definition should look something like this in configuration.h
#define MOTHERBOARD BOARD_MKS_GEN_13
In your case it should probably be
#define MOTHERBOARD BOARD_ULTIMAKER

closed 2018-06-10 by thinkyhead

raccoonhillcreator, 2020-01-10:
I found this thread tonight in a google search, I am having a similar problem ........... this is the very first ever attempt at firmware creation , its one of the most difficult things I have ever done .
I have installed a Einsy Rambo board in a printer that did not have this board before.
Pardon my complete lack of knowledge here
------------------------------------------------------------------------------------------------------

Arduino: 1.8.10 (Windows 10), Board: "RAMBo"

In file included from sketch\MarlinConfig.h:26:0,

                 from sketch\Marlin.h:35,

                 from sketch\Marlin_main.cpp:258:

boards.h:159:72: error: operator '==' has no right operand

 #define MB(board) (defined(BOARD_##board) && MOTHERBOARD==BOARD_##board)

                                                                        ^

sketch\pins.h:168:7: note: in expansion of macro 'MB'

 #elif MB(EINSY_RAMBO)

       ^~

In file included from sketch\MarlinConfig.h:40:0,

                 from sketch\Marlin.h:35,

                 from sketch\Marlin_main.cpp:258:

pins.h:268:4: error: #error "Unknown MOTHERBOARD value set in Configuration.h"

   #error "Unknown MOTHERBOARD value set in Configuration.h"

    ^~~~~

In file included from sketch\MarlinConfig.h:42:0,

                 from sketch\Marlin.h:35,

                 from sketch\Marlin_main.cpp:258:

SanityCheck.h:58:4: error: #error "MOTHERBOARD is required. Please update your configuration."

   #error "MOTHERBOARD is required. Please update your configuration."

    ^~~~~

SanityCheck.h:1127:4: error: #error "HEATER_0_PIN not defined for this board."

   #error "HEATER_0_PIN not defined for this board."

    ^~~~~

SanityCheck.h:1288:6: error: #error "Enable USE_ZMIN_PLUG when homing Z to MIN."

     #error "Enable USE_ZMIN_PLUG when homing Z to MIN."

      ^~~~~

SanityCheck.h:1492:4: error: #error "X_CS_PIN is required for TMC2130. Define X_CS_PIN in Configuration_adv.h."

   #error "X_CS_PIN is required for TMC2130. Define X_CS_PIN in Configuration_adv.h."

    ^~~~~

In file included from sketch\stepper.h:194:0,

                 from sketch\Marlin_main.cpp:262:

stepper_indirection.h:57:12: error: TMC2130Stepper.h: No such file or directory

   #include <TMC2130Stepper.h>

            ^~~~~~~~~~~~~~~~~~

compilation terminated.

exit status 1
operator '==' has no right operand

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Vertabreak, 2020-01-10:
this issue has been closed you may want to open a new issue and when doing so post copy's of the config files with the issue report so we can assist you.