MarlinFirmware/Marlin #10973: [bugfix-2.0.x] cardreader lsDive

simon-jouet, 2018-06-09:
### Description

In 2168d6ac lsDive was modified to pass NULL instead of an empty string however in https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/sd/cardreader.cpp#L113 prepend is used without checking if it's NULL or not. I changed back from NULL to "" and I can list the files fine.

### Steps to Reproduce

1. Have an sdcard with a folder (on windows "System Volume Information" will be created automatically)
1. M21 to init the SD card
2. M20 to list the files -- crash

**Expected behavior:** list the files

**Actual behavior:** crash

#### Additional Information

GMagician, 2018-06-09:
Thanks for this report

thinkyhead, 2018-06-10:
Patched!

PR #10978 (2018-06-10): [2.0.x] Fix null check in lsDive
Add null check, fix #10973
closed 2018-06-10 by thinkyhead