MarlinFirmware/Marlin #10974: Feature request: Lift between manual mesh probes

thehans, 2018-06-09:
### Steps to Reproduce
1. Perform manual mesh leveling on a bed where later probed positions are higher than previous probed points.

**Expected behavior:** [What you expect to happen]
The nozzle lifts in between manual mesh probes by some configurable amount, which allows the user to drop down gently into position instead of raising up from crashed nozzle half the time.

**Actual behavior:** [What actually happens]
Nozzle grinds into bed when moving from lower to higher bed points, because there is no lift in between.

thinkyhead, 2018-06-11:
There should already be a raise between points, depending on your configuration. Are you using `MESH_BED_LEVELING`? What version of Marlin are you using? Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as requested in the Issue template) and drop them into your next reply. We'll check them over and see if anything is amiss.

thehans, 2018-06-12:
Yes I am using `MESH_BED_LEVELING` and `PROBE_MANUALLY`
I am running Marlin `bugfix-1.1.x` that was recent as of a couple weeks ago.
[MarlinConfig.zip](https://github.com/MarlinFirmware/Marlin/files/2095320/MarlinConfig.zip)

thinkyhead, 2018-06-13:
When using `MESH_BED_LEVELING` the nozzle will raises up to `Z_CLEARANCE_BETWEEN_PROBES` or `Z_HOMING_HEIGHT` between probe points as you can see below:

(The larger of these two settings is assigned to `MANUAL_PROBE_HEIGHT`.)

```cpp
void _manual_goto_xy(const float &rx, const float &ry) {

  #if MANUAL_PROBE_HEIGHT > 0
    const float prev_z = current_position[Z_AXIS];
    do_blocking_move_to(rx, ry, MANUAL_PROBE_HEIGHT);
    do_blocking_move_to_z(prev_z);
  #else
    do_blocking_move_to_xy(rx, ry);
  #endif

  current_position[X_AXIS] = rx;
  current_position[Y_AXIS] = ry;

  #if ENABLED(LCD_BED_LEVELING)
    lcd_wait_for_move = false;
  #endif
}
```

> I am running Marlin bugfix-1.1.x that was recent as of a couple weeks ago.

Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch…

boelle, 2019-02-20:
@thehans is the problem solved in latest bugfix 2.0?

closed 2019-02-20 by thinkyhead